ChristianTremblay/BAC0 #121: How to get the log info in console into log file of our path?

sandhiya88, 2018-11-27:
Hi Im trying to create a log of my script wherein I want all the Log.info/debug/warning in my log file. currently, I can see in home/.BAC0/BAC0.log which does not give INFO. also few errors like value error:'invalid property for object type' breaks the code execution and does not log them in the BAC0.log file too. Hence, let me know how to get this done.

Safrone, 2018-11-27:
For getting INFO logging, I would try:

```
import logging
logger = logging.getLogger('BAC0')
logger.setLevel('INFO')
```

may or may not work

ChristianTremblay, 2018-11-27:
https://bac0.readthedocs.io/en/latest/logging.html?highlight=log
```
BAC0.log_level(level)
```

sandhiya88, 2018-11-28:
I got the level change method. But now my requirement is that I need the protocol data logs like <bacpypes.comm.PDU object at 0xb6d433cc>
pduSource = ('128.253.109.254', 47808)
pduData = x'81.04.00.37.0A.10.6D.45.BA.C0.01.28.FF.FF.00.00.B6.01.05.FD...'

as log results to analyze them further in the wireshark or any other protocol analyzer tool. Is there any way to get it done?  

between what we get in the BAC0.log is only the Info, debug or warning, error messages but not protocol level inputs. Any help on this?

sandhiya88, 2018-11-28:
> https://bac0.readthedocs.io/en/latest/logging.html?highlight=log
> 
> ```
> BAC0.log_level(level)
> ```

One quick question. If we have chnage level, where exactly in the script should I give this command. Before starting the bacnet connection ir any other rules?

ChristianTremblay, 2018-11-30:
I think it doesn't matter when... but as soon as you change the level, the log will be "updated". It really depends on what you want.

ChristianTremblay, 2018-11-30:
Regarding level protocol messages, you may have to enable logging on bacpypes itself.

closed 2020-01-29 by ChristianTremblay