ChristianTremblay/BAC0 #122: Unable to perform write operation on few properties of point objects

sandhiya88, 2018-11-28:
Unable to perform write request on point object properties of AI/AO/BI/BO in bacpypes.

I am trying the below command.
bacnet.write('2002:24 analogInput 10005 presentValue 100')

gives the error below:
raise NoResponseFromController()
NoResponseFromController

with respect to your isssue : https://github.com/JoelBender/bacpypes/issues/224, I saw few properties can be changed to commandable to write them. Could you please clarify on the steps?

I have raised the same in bacpypes lib: https://github.com/JoelBender/bacpypes/issues/232 as well.

ChristianTremblay, 2018-11-30:
Typically you will not be able to write to an analogInput. If you want to, you have to set its "out_of_service" property to true. Then you will be able to write to the present_value.
Doing this you will disconnect the variable from the hardware.

In BAC0, I call this behaviour "simulation" where I want to simulate a sensor value (like a room sensor).
Creating a device would make any write to this variable switch the "out_of_service" property for you.

ChristianTremblay, 2018-11-30:
https://github.com/JoelBender/bacpypes/issues/224 is related to objects created inside the BAC0 instance. Not remote controllers (Outside of BAC0).

sandhiya88, 2018-12-04:
Hi Christian,
As per your instructions, I tried to set the "Out_of_service" property to true and write to AnalogInput object proerties. I was able to write for the 'ReadableProperty' but 'OptionalProperty' type point proerties show error. 
I even checked on the Simulate.py file but no special preference on the property type is mentioned. But still I am not able to write. Pls help.

ChristianTremblay, 2018-12-05:
I don't get what you are trying to do. Please be more specific.

ChristianTremblay, 2019-03-10:
As per the standard, write true to « out_of_service » will make the property « presentValue » writable in some situations (like analog Inputs). 

Is this what you are trying to do ?

ChristianTremblay, 2019-08-21:
Reopen if bug still there

closed 2019-08-21 by ChristianTremblay