ChristianTremblay/BAC0 #126: Share value to N4

bbartling, 2019-01-24:
Hi Christian,

Is it possible with BAC0 to share a value to N4? (integrate a point) I am hoping to do a calculation in Python which returns an integer and then integrate this into N4, can BAC0 do this?

I know the code below doesn't work, but is it possible for `n` be integrated to N4 and trended like any ordinary BACnet controller?

```
import BAC0
import random
import time

bacnet = BAC0.lite()

for i in range (1000):
    n = random.randrange(0, 101, 5)
    print(n)
    time.sleep(1)
```

Thanks for any tips/help I am amazed at the complexity of Bacypes (and BACnet in general from a higher level than just setting up BAS) for someone such as yourself to create an entire Python package like this..

bbartling, 2019-01-24:
Whats interesting without knowing what I am doing in editing code, I can get this to work in N4 I can discover BAC0, and suck up the AI point with a present value of 99.9

```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AI
from collections import namedtuple

def _add_points(qty, device):

    # Add a points
    ais = []
    for i in range(qty):
        ais.append(create_AI(oid=i, name='ai{}'.format(i), pv=99.9))
        
    for ai in ais:
        device.this_application.add_object(ai)

#call function
device_app = BAC0.lite()
_add_points(1, device_app)
```

[image]

ChristianTremblay, 2019-01-24:
Cool heh :)

bbartling, 2019-01-24:
This will generate a random integer from 0-100 in values of 5 for the AI that discovered on N4, but would you have any ideas on how this AI0 could be continuously updated periodically?

```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AI
from collections import namedtuple

def _add_points(qty, device):

    # Add a points
    ais = []
    for i in range(qty):
        ais.append(create_AI(oid=i, name='ai{}'.format(i), pv=random.randrange(0, 101, 5)))
        
        
    for ai in ais:
        device.this_application.add_object(ai)

#call function
device_app = BAC0.lite()
_add_points(1, device_app)
```

ChristianTremblay, 2019-01-24:
You need your script to write to it

ChristianTremblay, 2019-01-24:
Depends on how you run your script.
You could add a task to your BAC0 device to update the value at a certain frequency.

you will need the make mutable thing...

```
import BAC0
from BAC0.tasks.RecurringTask import RecurringTask

bacnet = BAC0.lite()

def _make_mutable(obj, identifier='presentValue', mutable=True):
            """ 
            This function is not the way to go as it changes the class
            property...As bacpypes issue #224, it will need a lot of work
            """
            for prop in obj.properties:
                if prop.identifier == identifier:
                    prop.mutable = mutable
            return obj

bacnet.this_application.add_object(_make_mutable(create_AI(oid=0, name='MyAI', pv=Real(0))))

def changeValueOfMyAI():
     obj = bacnet.this_application.get_object_name('MyAI')
     value = obj.ReadProperty('presentValue').value
     value = value + 1 if value < 100 else 0
     obj.WriteProperty('presentValue', Real(value), priority=16, direct=True)

task = RecurringTask(changeValueOfMyAI,delay=2)
task.start()
```

Something like that seems to work

ChristianTremblay, 2019-01-24:
This will change from 0 to 100 then revert back to 0 to start over changing value each 2 seconds

bbartling, 2019-01-24:
Thanks for sending this over Christian, do you know how I can troubleshoot this traceback error pv=Real:

`Traceback (most recent call last):
  File "C:\Users\bbartling\Desktop\testBAC02.py", line 20, in <module>
    bacnet.this_application.add_object(_make_mutable(create_AI(oid=0, name='MyAI', pv=Real(0))))
NameError: name 'Real' is not defined`

ChristianTremblay, 2019-01-24:
I've updated the code..

`from bacpypes.primitivedata import Real`

was missing

bbartling, 2019-01-25:
Hi Christian,

This works on my Windows 10 laptop only... The script runs on pi and another PC I have on the LAN running Linux Mint 18.3, but they  wont come up on a BACnet discovery thru Workbench. Any ideas to troubleshoot why the BAC0 script only works on Windows laptop to come up on Workbench?

I had to also add in `from BAC0.core.devices.create_objects import create_AI`

This is the script up till now, thanks for your help & time much appreciation on my end..

```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AI
from collections import namedtuple
from BAC0.tasks.RecurringTask import RecurringTask
from bacpypes.primitivedata import Real
from BAC0.core.devices.create_objects import create_AI

bacnet = BAC0.lite()

def _make_mutable(obj, identifier='presentValue', mutable=True):
            """ 
            This function is not the way to go as it changes the class
            property...As bacpypes issue #224, it will need a lot of work
            """
            for prop in obj.properties:
                if prop.identifier == identifier:
                    prop.mutable = mutable
            return obj

bacnet.this_application.add_object(_make_mutable(create_AI(oid=0, name='MyAI', pv=Real(0))))

def changeValueOfMyAI():
     obj = bacnet.this_application.get_object_name('MyAI')
     value = obj.ReadProperty('presentValue').value
     value = value + 1 if value < 100 else 0
     obj.WriteProperty('presentValue', Real(value), priority=16, direct=True)

task = RecurringTask(changeValueOfMyAI,delay=5)
task.start()

```

JoelBender, 2019-01-26:
As a side comment, you can change the value of a property "directly", `__getattr__()` and `__setattr__()` are defined for "direct access", a little nicer than making the property mutable because that opens it up for being changed via BACnet services:
```
>>> from bacpypes.object import AnalogValueObject
>>> avo = AnalogValueObject(objectName="avo", objectIdentifier=('analogValue', 1))

>>> avo.WriteProperty('presentValue', 75.3)
# raises bacpypes.errors.ExecutionError: ('property', 'writeAccessDenied')

# direct access
>>> avo.presentValue = 75.3
>>> avo.presentValue
75.3

>>> avo.presentValue += 100
>>> avo.presentValue
175.3
```
If you change the value like this it will still respect the COV notifications.  It's also one of the main reasons I switched to lowerCamelCase property names rather then hyphenated-ones or the UPPERCASE_VERSION in the standard, the others are to make JSON serialization easier to navigate and have property names matching predicate naming conventions for RDF serialization.

bbartling, 2019-01-29:
Hello,

Any chance for a tip on making the recommendations by @JoelBender work in a script fashion :)

This code runs without an error and I can discover my BAC0 device & AV point, but I cannot get the value to change with the loop:

```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AV
from collections import namedtuple

def _add_points(qty, device):

    # Add a points
    avs = []
    for i in range(qty):
        avs.append(create_AV(oid=i, name='av{}'.format(i), pv=0))
        
    for av in avs:
        device.this_application.add_object(av)

#call function
device_app = BAC0.lite()
_add_points(1, device_app)

value = 0
#loop for a continous write 0-100
device_app.presentValue = value + 1 if value < 100 else 0
```

bbartling, 2019-01-29:
One thing I am noticing on rasp pi, this works to discover pi on N4 just using the interpreter:

```
>>> import BAC0
>>> bacnet = BAC0.lite()
```

But when I run a script .py file on pi with the same code above it doesn't work to discover pi in N4. There are no errors, it looks like it runs just fine.. Im using rasbian stretch python 3.5... Same issue with a PC I have running Linux Mint 18.3 with Anaconda 3.6, the interpreter method works to discover BAC0 in N4 but not when ran in a .py file.

On my Windows 10 laptop running IDLE python 3.6 this works just fine, with a more complicated script I can discover the BAC0 device on the JACE 8000 and bring in the mutable point that changes value... 
```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AI
from collections import namedtuple
from BAC0.tasks.RecurringTask import RecurringTask
from bacpypes.primitivedata import Real
from BAC0.core.devices.create_objects import create_AI

bacnet = BAC0.lite()

def _make_mutable(obj, identifier='presentValue', mutable=True):
            """ 
            This function is not the way to go as it changes the class
            property...As bacpypes issue #224, it will need a lot of work
            """
            for prop in obj.properties:
                if prop.identifier == identifier:
                    prop.mutable = mutable
            return obj

bacnet.this_application.add_object(_make_mutable(create_AI(oid=0, name='MyAI', pv=Real(0))))

def changeValueOfMyAI():
     obj = bacnet.this_application.get_object_name('MyAI')
     value = obj.ReadProperty('presentValue').value
     value = value + 1 if value < 100 else 0
     obj.WriteProperty('presentValue', Real(value), priority=16, direct=True)

task = RecurringTask(changeValueOfMyAI,delay=5)
task.start()
```

ChristianTremblay, 2019-01-29:
Probably because you didn’t provide a while loop to run the code until you
tell it to stop.

The code run... then exit.

Try :

Python myscript.py

You will see it run then go back to the shell. Job done ;-)

What you want is to create some kind of service that will keep the code
running.

bbartling, 2019-01-30:
Hi Christian, in Linux I have been running scripts from terminal...

On pi in the correct folder directory, it’s:

python3 runmyscript.py

bbartling, 2019-01-30:
I think I know what you mean now with the while loop. Create .py file and run it in terminal:

import BAC0

while True:
          bacnet = BAC0.lite()

bbartling, 2019-01-31:
Hi Christian,

Any chance for a tip on getting BAC0 to work on pi? I can get it to run thru the interpreter and discover the device on n4 in this fashion:

[image]

But not when running in a script form like in the screen shot below, even with a `while True`:

[image]

Any tips for what I am doing wrong would be greatly appreciated!! If the interpreter is the only method to make it work, I maybe able to just work with that if I can share a calculated value to n4... Something like what @JoelBender demonstrated on an AO

Thanks for your time, much appreciation on my end.

ChristianTremblay, 2019-02-01:
I will need time... but I’ll look at this

ChristianTremblay, 2019-02-03:
The while loop must be doing something simple... not re-running the script or you’ll end with something trying to create an infinite number of instances....

It’s probably not the best way to program a Linux service... but this can serve at a starting point proving the concept. It keeps the script alive while you can interact with another instance.

```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AI
from collections import namedtuple
from BAC0.tasks.RecurringTask import RecurringTask 
from bacpypes.primitivedata import Real
from BAC0.core.devices.create_objects import create_AI

def changeValueOfMyAI():
     obj = bacnet.this_application.get_object_name('MyAI')
     value = obj.ReadProperty('presentValue').value
     value = value + 1 if value < 100 else 0
     obj.presentValue = Real(value)

bacnet = BAC0.lite()
bacnet.this_application.add_object(_make_mutable(create_AI(oid=0, name='MyAI', pv=Real(0))))

task = RecurringTask(changeValueOfMyAI,delay=5)
task.start()

while True:
    obj = bacnet.this_application.get_object_name(‘MyAI’)
    print(‘Running and value is {}’.format(obj.presentValue)
    time.sleep(10)
```

bbartling, 2019-02-07:
Hi Christian,

This is working... AND thank you!!!!

I had to delete out the `_make_mutable()` from this line: `bacnet.this_application.add_object(_make_mutable(create_AI(oid=0, name='MyAI', pv=Real(0))))`

This is the full working script on my Windows 10 laptop that is discoverable on N4:

```
import BAC0
import random
import time
from BAC0.core.devices.create_objects import create_AI
from collections import namedtuple
from BAC0.tasks.RecurringTask import RecurringTask 
from bacpypes.primitivedata import Real
from BAC0.core.devices.create_objects import create_AI

def changeValueOfMyAI():
     obj = bacnet.this_application.get_object_name('MyAI')
     value = obj.ReadProperty('presentValue').value
     value = value + 1 if value < 100 else 0
     obj.presentValue = Real(value)

bacnet = BAC0.lite()
bacnet.this_application.add_object(create_AI(oid=0, name='MyAI', pv=Real(0)))

task = RecurringTask(changeValueOfMyAI,delay=5)
task.start()

while True:
    obj = bacnet.this_application.get_object_name('MyAI')
    print('Running and value is {}'.format(obj.presentValue))
    time.sleep(10)
```

closed 2019-02-08 by ChristianTremblay