ChristianTremblay/BAC0 #127: 0.99.944 causes point discovery to fail

akshephard, 2019-02-01:
I recently found an issue that is present in BAC0 version 0.99.944 that is not in BAC0 0.99.932
I can build points fine with version 0.99.932, but get errors with a lot of the objects. Here is some information regarding the problem. I'd be happy to help troubleshoot this issue, we have some test Bacnet devices on a non-operational network.

```
>>> import BAC0
>>> BAC0.version
'0.99.944'
>>> bacnet = BAC0.connect(ip='X.X.X.X/24')
2019-01-31 18:31:03,423 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-01-31 18:31:03,425 - INFO    | Using ip : X.X.X.X
2019-01-31 18:31:03,433 - INFO    | Starting app...
2019-01-31 18:31:03,436 - INFO    | BAC0 started
2019-01-31 18:31:03,438 - INFO    | Registered as Simple BACnet/IP App

>>> bacnet.whois()
defaultdict(<class 'int'>, {('24001:3', 24003): 1, ('24001:2', 24002): 1, ('24001:4', 73004): 1, ('X.X.X.X', 24001): 1})
>>> bacnet.devices
[('UTM', 'Badger Meter, Inc.', '24001:3', 24003), ('device24002', 'Automated Logic Corporation', '24001:2', 24002), ('TEC2645-4', 'Johnson Controls, Inc.', '24001:4', 73004), ('device24001', 'Automated Logic Corporation', 'X.X.X.X', 24001)]
>>>

>>> controller = BAC0.device('X.X.X.X', 24001, bacnet)
2019-01-31 18:31:12,313 - INFO    | Changing device state to DeviceDisconnected'>
2019-01-31 18:31:12,358 - INFO    | Changing device state to RPMDeviceConnected'>
2019-01-31 18:31:12,399 - INFO    | Device 24001:[device24001] found... building points list
2019-01-31 18:31:16,918 - WARNING | Offline: provide database name to load stored data.
WARNING:BAC0:Offline: provide database name to load stored data.
2019-01-31 18:31:16,919 - WARNING | Ex. controller.connect(db = 'backup')
WARNING:BAC0:Ex. controller.connect(db = 'backup')

>>> controller = BAC0.device('X.X.X.X', 24001, bacnet)
2019-01-31 18:31:12,313 - INFO    | Changing device state to DeviceDisconnected'>
2019-01-31 18:31:12,358 - INFO    | Changing device state to RPMDeviceConnected'>
2019-01-31 18:31:12,399 - INFO    | Device 24001:[device24001] found... building points list
2019-01-31 18:31:16,918 - WARNING | Offline: provide database name to load stored data.
WARNING:BAC0:Offline: provide database name to load stored data.
2019-01-31 18:31:16,919 - WARNING | Ex. controller.connect(db = 'backup')
WARNING:BAC0:Ex. controller.connect(db = 'backup')

After reading this [issue](https://github.com/ChristianTremblay/BAC0/issues/38)

I wrote a script which uses a try and except block to try adding each point from
the obj_list until it fails. Also I was able to do a direct read using the read
function for basically all of the points with 0.99.44 

This is one of the errors that happens while building the points:

```
Traceback (most recent call last):
  File "set_object.py", line 17, in <module>
    controller = BAC0.device('X.X.X.X', 24001, bacnet, object_list=test_obj_list, segmentation_supported=False)
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/Device.py", line 165, in __init__
    self.new_state(DeviceDisconnected)
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/Device.py", line 177, in new_state
    self._init_state()
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/Device.py", line 623, in _init_state
    self.connect()
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/Device.py", line 651, in connect
    self.new_state(RPDeviceConnected)
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/Device.py", line 177, in new_state
    self._init_state()
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/Device.py", line 402, in _init_state
    self._buildPointList()
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/Device.py", line 467, in _buildPointList
    self.custom_object_list)
  File "/usr/local/lib/python3.5/dist-packages/BAC0/core/devices/mixins/read_mixin.py", line 539, in _discoverPoints
    tl = trendLogs(point_address, self)
NameError: name 'trendLogs' is not defined

```
The first object it fails at is this:
```
obj_list[69] = ('trendLog', 7)
```

There are quite a few objects that cause issues though:
```
fail_list = [69, 78, 86, 99, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223, 224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 256, 257, 258, 259, 260, 261, 262, 263, 264, 265, 266, 267, 268, 269, 270, 271, 272, 273, 274, 275, 276, 277, 278, 279, 280, 281, 282, 283, 284, 285, 286, 287, 288, 289, 290, 291, 292, 293, 294, 295, 296, 297, 298, 299, 300, 301, 302, 303, 304, 305, 306, 307, 308, 309, 310, 311, 312, 313, 314, 315, 316, 317, 318, 319, 320, 321, 322, 323, 324, 325, 326, 327, 328, 329, 330, 331, 332, 333, 334, 335, 336, 337, 338, 339, 340, 341, 342, 343, 344, 345, 346, 347, 348, 349, 350, 351, 352, 353, 354, 355, 356, 357, 358, 359, 360, 361, 362, 363, 364, 365, 366, 367, 368, 369, 370, 371, 372, 373, 374, 375, 376, 377, 378, 379, 380, 381, 382, 383, 384, 385, 386, 387, 388, 389, 390, 391, 392, 393, 394, 395, 396, 397, 398, 399, 400, 401, 402, 403, 404, 405, 406, 407, 408, 409, 410, 411, 412, 413, 414, 415, 416, 417, 418, 419, 420, 421, 422, 423, 424, 425, 426, 427, 428, 429, 430, 431, 432, 433, 434, 435, 436, 437, 438, 439, 440, 441, 442, 443, 444, 445, 446, 447, 448, 449, 450, 451, 452, 453, 454, 455, 456, 457, 458, 459, 460, 461, 462, 463, 464, 465, 466, 467, 468, 469, 470, 471, 472, 473, 474, 475, 476, 477, 478, 479, 480, 481, 482, 483, 484, 485, 486, 487, 488, 489, 490, 491, 492, 493, 494, 495, 496, 497, 498, 499, 500, 501, 502, 503, 504, 505, 506, 507, 508, 509, 510, 511, 512, 513, 514, 515, 516, 517, 518]
```

However, when I downgraded to BAC0 version 0.99.32 I am able to build points
I recently found an issue with 0.99.944 that is not found in 0.99.932
```
>>> import BAC0
>>> BAC0.version
'0.99.932'
>>> bacnet = BAC0.connect(ip='X.X.X.X/24')
2019-01-31 19:02:24,697 - INFO    | Starting BAC0 version 0.99.932 (Lite)
2019-01-31 19:02:24,699 - INFO    | Using ip : X.X.X.X/24
2019-01-31 19:02:24,707 - INFO    | Starting app...
2019-01-31 19:02:24,710 - INFO    | BAC0 started
2019-01-31 19:02:24,712 - INFO    | Registered as Simple BACnet/IP App
>>> controller = BAC0.device('X.X.X.X', 24001, bacnet)
2019-01-31 19:02:32,317 - INFO    | Changing device state to DeviceDisconnected'>
2019-01-31 19:02:32,363 - INFO    | Changing device state to RPMDeviceConnected'>
2019-01-31 19:02:32,403 - INFO    | Device 24001:[device24001] found... building points list
2019-01-31 19:02:36,869 - INFO    | Ready!
2019-01-31 19:02:36,876 - INFO    | Polling started, values read every 10 seconds
>>> bacnet.whois()
defaultdict(<class 'int'>, {('24001:2', 24002): 1, ('24001:3', 24003): 1, ('X.X.X.X', 24001): 1, ('24001:4', 73004): 1})
>>> bacnet.devices
[('device24002', 'Automated Logic Corporation', '24001:2', 24002), ('UTM', 'Badger Meter, Inc.', '24001:3', 24003), ('device24001', 'Automated Logic Corporation', 'X.X.X.X', 24001), ('TEC2645-4', 'Johnson Controls, Inc.', '24001:4', 73004)]
>>>
```

ChristianTremblay, 2019-02-01:
Can you try with a capital T in line 539....seems like a typo.
TrendLogs

Older version (if I remember correctly for .32) didn’t support trendLogs objects so they were not added in the discovery process.

I don’t have access right now to a device with trendLogs... would help if you could do the test

akshephard, 2019-02-01:
That actually solved the problem. I read in the obj_list using:
```obj_list = bacnet.read('X.X.X.X device 24001 objectList')```
So I looped through all of the objects in the list and changed them from trendLog to Trendlog and that fixed the issue. Is this just a typo from reading in the object list?

ChristianTremblay, 2019-02-03:
Seems that the develop branch was correct...but not the master branch... TrendLog with a L is the correct class to call when creating the list. There is no tests actually in the suite to validate this portion of the code.... so it just passed silently.

closed 2019-06-20 by ChristianTremblay