ChristianTremblay/BAC0 #128: while True loop

bbartling, 2019-02-11:
Hi Christian,

I am tinkering around with a script (.py file, _not_ interpreter method >>>) in hopes can keep an eye on the hot water system temperatures with BAC0 lite (to start with)... Sort of like an additional alarm dial out feature to the BAS if I can build off of a working script.

Do I need a while True loop to keep the script alive? Like shown below...? It throws a lot of exceptions...

```
import BAC0
import time

bacnet = BAC0.lite()
hws = BAC0.device('100:99', 30099, bacnet)
print(hws.points)

while True:
    
    print(hws['PHWR-T'])
    time.sleep(60)

    if KeyboardInterrupt:
        bacnet.disconnect()
```

This script below runs but only prints hot water return temperature once... I think the service is still running in the background but how do I keep retrieving values from the hot water system temperature?

Ultimately I am hoping to use some python logic to dial out or make adjustments...

```
import BAC0
import time

bacnet = BAC0.lite()
hws = BAC0.device('100:99', 30099, bacnet)
print(hws.points)
    
print(hws['PHWR-T'])
time.sleep(60)
```

ChristianTremblay, 2019-02-21:
You could try using logging and stream to a file ?

bbartling, 2019-02-21:
Hi Christian, would you have a tip on how I can incorporate that? Thanks

ChristianTremblay, 2019-03-10:
Have a look here : 
https://docs.python.org/3/library/logging.html

And here : 
https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/utils/notes.py

bbartling, 2019-04-11:
Hi Christian, Sorry for the late response here I was busy for a while...

On your Feb 20th note, what does it mean to use logging to stream to a file? I looked at the links including the GitHub code for notes.py, but I am still a little confused on what it all does.

I copied and pasted this from the logging docs:
The basic classes defined by the module, together with their functions, are listed below.

- Loggers expose the interface that application code directly uses.
- Handlers send the log records (created by loggers) to the appropriate destination.
- Filters provide a finer grained facility for determining which log records to output.
- Formatters specify the layout of log records in the final output.

bbartling, 2019-04-11:
If I run this code:
```
import BAC0
import time

bacnet = BAC0.lite()
hws = BAC0.device('100:99', 30099, bacnet)
#print(hws.points)
    
print(hws['PHWR-T'])
time.sleep(60)
```

I will print:
```
2019-04-11 08:30:32,170 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-04-11 08:30:32,323 - INFO    | Using ip : 10.30.4.209
2019-04-11 08:30:32,339 - INFO    | Starting app...
2019-04-11 08:30:32,354 - INFO    | BAC0 started
2019-04-11 08:30:32,354 - INFO    | Registered as Simple BACnet/IP App
2019-04-11 08:30:34,359 - INFO    | Changing device state to DeviceDisconnected'>
2019-04-11 08:30:35,459 - INFO    | Changing device state to RPMDeviceConnected'>
2019-04-11 08:30:36,544 - INFO    | Device 30099:[Boiler] found... building points list
2019-04-11 08:30:38,801 - INFO    | Ready!
2019-04-11 08:30:38,833 - INFO    | Polling started, values read every 10 seconds
Boiler/PHWR-T : 133.45 degreesFahrenheit
>>> 
```

Any tips is greatly appreciated! Sorry for the low level of not understand..

bbartling, 2019-04-11:
One thing I am playing around with in IDLE not using a script (.py file) is just using the interpreter..

At the end I am running into some issues `trend = BAC0.TrendLog(1,hws)`, I don't think I know what the `1` represents in `(1,hws)`. Can you give me a tip at all? Thanks

```
>>> import BAC0
>>> import time
>>> bacnet = BAC0.lite()
2019-04-11 12:32:27,180 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-04-11 12:32:27,570 - INFO    | Using ip : 10.30.4.209
2019-04-11 12:32:27,617 - INFO    | Starting app...
2019-04-11 12:32:27,617 - INFO    | BAC0 started
2019-04-11 12:32:27,617 - INFO    | Registered as Simple BACnet/IP App
>>> hws = BAC0.device('100:99', 30099, bacnet)
2019-04-11 12:33:51,799 - INFO    | Changing device state to DeviceDisconnected'>
2019-04-11 12:33:52,455 - INFO    | Changing device state to RPMDeviceConnected'>
2019-04-11 12:33:53,877 - INFO    | Device 30099:[Boiler] found... building points list
2019-04-11 12:33:55,831 - INFO    | Ready!
2019-04-11 12:33:55,847 - INFO    | Polling started, values read every 10 seconds
>>> hws['PHWR-T'].history
2019-04-11 12:33:54.908105    134.552246
2019-04-11 12:33:57.206502    134.561066
2019-04-11 12:34:08.624656    134.569366
2019-04-11 12:34:20.094917    134.553085
2019-04-11 12:34:31.485965    134.516357
2019-04-11 12:34:42.861877    134.484360
2019-04-11 12:34:54.380536    134.438889
Name: Boiler/PHWR-T, dtype: float64
>>> trend = BAC0.TrendLog(1,hws)
```

bbartling, 2019-04-11:
Hi Christian, I am doing some more Tinkering around here... And I think I have something I can work with... (Like the issue can be closed)

```
import BAC0
import time

bacnet = BAC0.lite()
hws = BAC0.device('100:99', 30099, bacnet)

data = hws['SHWS-T'].history.round(decimals=0)
print(data.min)

while True:

    
    time.sleep(30)

    if data.min() > 140:
        print("looks good")

    else:
        print("we are going to get cold")
```

Seems to work, like its keeping the service active..! I think I can use this to dial out or even adjust a setpoint or something maybe...

[image]

closed 2019-04-12 by bbartling

ChristianTremblay, 2019-04-13:
@bbartling 
`history.min()` will give you the minimum value for all the series... not sure I would base control over this... you should probably rely on `point.history.lastValue` instead.

Regarding Trendlog
If you look here
https://bac0.readthedocs.io/en/latest/histories.html#trendlog

the "1" is the address of the TrendLog object. TendLogs are like AnalogValue and they have an address. We could do a parallel saying : 

* `AV:1`
* `TrendLog:1`