ChristianTremblay/BAC0 #129: writing binary point results in undefined enumeration

henkwitte, 2019-02-28:
Hello,

I am trying to write to a binaryValue point:

bacnet.write('192.168.25.22 binaryValue 102042 presentValue 1')

The point exists and I can write on it using e.g. Yabe, but from python I get:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/groenhol/miniconda3/lib/python3.6/site-packages/BAC0/core/io/Write.py", line 80, in write
    iocb = IOCB(self.build_wp_request(args, vendor_id=vendor_id))
  File "/home/groenhol/miniconda3/lib/python3.6/site-packages/BAC0/core/io/Write.py", line 151, in build_wp_request
    value = datatype(value)
  File "/home/groenhol/miniconda3/lib/python3.6/site-packages/bacpypes/primitivedata.py", line 1099, in __init__
    raise ValueError("undefined enumeration '%s'" % (arg,))

I tested this with python 3.6 and BAC0 0.99.994, bacpypes 0.17.5

Any suggestions?

Henk Witte

ChristianTremblay, 2019-02-28:
> File "", line 1, in

Try « active » instead of 1

henkwitte, 2019-02-28:
Perfect, "active" and "inactive" works!

I tried 0/1, false/true, on/off but not active/inactive

Did I miss something in the manual? I assume this should be documented somewhare, in YABE I just see 0 and 1 and can use that to write.

closed 2019-02-28 by henkwitte

ChristianTremblay, 2019-03-10:
@henkwitte Bacpypes is following the standard word by word and all the implementation has been made to be as close as possible to the standard. Using it, you must accept to work at a very low level of the BACnet standard.
Other softwares which are more user-oriented (instead of standard-centric) will present to the user concepts that are more « popular » or « known ». 0 and 1, true and false are much more intuitive for users... but under the hood YABE like any other will build the message as requested by the standard.

BAC0 itself, when writing to Boolean point (in the context of a BAC0 device) will accept True and False as I considered it was more intuitive.

saraembeddian, 2022-12-30:
@ChristianTremblay hi, i think the problem still persists...
[image]
This simply returns NONE.
[image]
when seeing from the above wireshark log, the presentValue is NULL.

Note: the result is same even we set the priority as '8'.

Could you help ?