ChristianTremblay/BAC0 #130: read mixin can check validity before assignment

ilayn, 2019-03-04:
First of all thank you for this nice work. It really makes life easier. I have an issue regarding the read_single behavior during controller initialization. Consider the following

```python
bacnet = BAC0.connect()
controller = BAC0.device(address='10.xx.xx.xx',
                          device_id=900,
                          network=bGrid_bacnet,
                          object_list= [('device', 900),
#                                        ('analogValue', 0),
                                        ])

```

This leads to the following stack

```bash
2019-03-04 14:19:15,314 - INFO    | Changing device state to DeviceDisconnected'>
2019-03-04 14:19:16,280 - INFO    | Changing device state to RPDeviceConnected'>
2019-03-04 14:19:17,300 - INFO    | Device 900:[GW_room] found... building points list
Traceback (most recent call last):

  File "<ipython-input-7-1f3fedd559ac>", line 7, in <module>
    ('analogValue', 4)],

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\Device.py", line 165, in __init__
    self.new_state(DeviceDisconnected)

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\Device.py", line 177, in new_state
    self._init_state()

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\Device.py", line 623, in _init_state
    self.connect()

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\Device.py", line 651, in connect
    self.new_state(RPDeviceConnected)

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\Device.py", line 177, in new_state
    self._init_state()

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\Device.py", line 402, in _init_state
    self._buildPointList()

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\Device.py", line 467, in _buildPointList
    self.custom_object_list)

  File "c:\users\evalan\appdata\local\programs\python\python37\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py", line 500, in _discoverPoints
    self.read_single('{} {} presentValue '.format(point_type, point_address))),

ValueError: could not convert string to float: 
```

Then when I modified the `_discoverPoints` it turns out that the return value of `self.read_single('{} {} presentValue '.format(point_type, point_address))),` is an empty string hence `float('')`.

Is there a given default value that you can write in instead of forcing the float value? I can send a PR if you have a default value in mind. 

(version info bacpypes 0.17.5, BAC0 0.99.944)

ChristianTremblay, 2019-03-10:
Hummm.... technically it should not give an empty string.... this is weird.
It is trying to read the presentValue of an analogValue.... which « must » be a float...

Can you give me more informations on the controller ?

ilayn, 2019-03-10:
I think this is also related to the out-of-service condition. When an endpoint object becomes out-of-service is there a special check about skipping the value?

ChristianTremblay, 2019-03-13:
[image]

ChristianTremblay, 2019-03-13:
[image]

ChristianTremblay, 2019-03-13:
It should not interfere with the reading of the present value.

ilayn, 2019-03-13:
Indeed, you are right. Thus, in our case when out-of-service is true the value is an empty string, why it is chosen that way I have no idea but when the controller is initialized it is actually looking for a valid float-convertible value and that's where it gets stuck in one of the AV endpoints. 

I guess what I am trying to understand is whether there is a way to skip the endpoint value if OoS is set. Then it won't complain about having the value not floatable. As the standard says. it can be any value anyways and it seems to me that it shouldn't be checked.

ChristianTremblay, 2019-03-13:
So it's a in-house controller ? 
Or something you bought but would have a clumsy BACnet implementation ?

I guess we could catch the exception and convert to `None`... 
Do you know the content of the string ? Is this really an empty string ? `""`

ilayn, 2019-03-24:
> So it's a in-house controller ?
> Or something you bought but would have a clumsy BACnet implementation ?

In fact it is a mixture of both. 

> Do you know the content of the string ? Is this really an empty string ? ""

It is indeed an empty string but since the message can be anything when OoS is set I guess that might bear some priority in the checks.

Romeren, 2019-09-13:
Hello Christian,
Joining in on the conversation here.
I come across this issue a lot when i integrate with a lot of devices sat up from a different vendors.
No doubt it is definitely a issue caused by the installers from these devices. For instance im sitting here with a system where a obj_cls=NumericPoint, obj_type="analog" returns a NoneType.
This causes an error right here BAC0/core/devices/mixins/read_mixin.py", line 317.

If you are interested in getting a fix. Then i have one ready that i can make a pull request for.

Btw great work so far!

ChristianTremblay, 2019-09-13:
I guess I didn’t understand the problem completely the first time. The PR make sense and I will add this resilience test if there is an empty response. I’ll probably log a warning as it is not supposed to happen as suggested. 

Thabks

closed 2021-02-13 by github-actions[bot]