ChristianTremblay/BAC0 #131: BAC0 write method

ahmedmfarahat, 2019-04-03:
Hi I want to create the write method but I can't do it.
my code is as follows:
import BAC0
myIPAddr = '192.168.1.253/47808
bacnet = BAC0.connect(ip = myIPAddr)
bacnet.write('0:698 analogOutput 1 presentValue100')
I have a bacnet gateway configured with a dimmer where I want to write to see the output on the dimmer. Please help me asap 
Thanks

ChristianTremblay, 2019-04-13:
myIPAddr should be : 
```
import BAC0
myIPAddr = '192.168.1.253/24'
bacnet = BAC0.connect(ip = myIPAddr)
bacnet.write('0:698 analogOutput 1 presentValue 100')
```

In the definition of the IP, you specify the subnet mask after the slash
You don't have to specify the port (47808) as it is the default port.

ahmedmfarahat, 2019-04-14:
So if the subnet mask is 255.255.255.0, myIPAddr will be = '192.168.1.253/255' is that right?
And for the bacnet.write method what is the 0:698? what does these stands for? I am using HDL device. Please tell me what does these numbers stands for as this is my main problem.
Note: in Analoug Output 0 i cant change the out_of_service variable to true from my BACnet simulator or my code so please advice if this will be a problem.
Thanks

ChristianTremblay, 2019-04-14:
SUBNET
=======
No it is not :)
255.255.255.0 = 24
Easiest way to find the equivalent is to look at a subnet calculator like : http://www.subnet-calculator.com/

There is a good explanation here : https://networkengineering.stackexchange.com/questions/7106/how-do-you-calculate-the-prefix-network-subnet-and-host-numbers

0:698
=====
Network number : Address of device on this network

What's weird here is that it can't be MSTP as 698 > 127 AND 698 > 255 (neither a master device or a slave device on a MSTP network)

You may have used the BACnet Device Instance number which will not work here.

If you are talking over IP, you don't need to specify the network number if you are on the same subnet.

`bacnet.write("192.168.1.100 analogOutput 1 presentValue 100")`

Analog output and out_of_service
=========================
Honestly, I don't see why in real life situation you would like to change the out_of_service property of an analog output... you should overrride the value using a priority of 8 instead. More logical way to interact with an output IMHO.
Not all simulator are made equal... and I don't use them at all. I prefer real devices.

`bacnet.write("192.168.1.100 analogOutput 1 presentValue 100 - 8")`

ahmedmfarahat, 2019-04-15:
Great it worked!! Thanks a lot Christian. 
I want to have a loop over my Bacnet device so once I make the presentValue 100 and then in the next turn is 50 and so on. Can you please advice for this

ahmedmfarahat, 2019-04-21:
@ChristianTremblay Looking forward for your prompt reply

ChristianTremblay, 2019-04-24:
Instead of using a loop you could look at this : https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/tasks/RecurringTask.py

You can define a function and pass it to a recurring task and the function will run every x seconds.

closed 2019-06-12 by ChristianTremblay