ChristianTremblay/BAC0 #132: Error when no response from controller

Kartstig, 2019-04-11:
I'm getting the following exception when trying to access BAC.lite(...).devices:
```
2019-04-11 00:24:14,501 - WARNING | APDU Abort Reject Reason : noResponse
[2019-04-11 00:24:14,501: WARNING/ForkPoolWorker-1] APDU Abort Reject Reason : noResponse
2019-04-11 00:24:14,502 - WARNING | No response from controller
[2019-04-11 00:24:14,502: WARNING/ForkPoolWorker-1] No response from controller
[2019-04-11 00:24:14,502: ERROR/ForkPoolWorker-1] discover/0
Traceback (most recent call last):
  File "/home/herman/dev/PyBacnet/beacon_web/batch/__init__.py", line 53, in discover
    for d in bactalk.find_devices(_connection)
  File "/home/herman/dev/PyBacnet/lib/bactalk.py", line 37, in find_devices
    return conn.devices
  File "/home/herman/dev/PyBacnet/venv/lib/python3.6/site-packages/BAC0/scripts/Lite.py", line 161, in devices
    '{} device {} objectName vendorName'.format(device[0], device[1]))
ValueError: not enough values to unpack (expected 2, got 1)
```

ChristianTremblay, 2019-04-13:
I will need more information because it seems you are wrapping BAC0 in something else...

Kartstig, 2019-04-14:
What information do you need?

Here's the function that is calling the BAC0 code:

```python
def find_devices(conn):
    conn.whois()
    time.sleep(WHOIS_WAIT)
    return conn.devices
```

where `conn` is the BAC0 connection instance

ChristianTremblay, 2019-04-14:
The `forkpoolworker` let me think you're trying to make something in a async way....which is probably the source of your pain.

BAC0... bacpypes... are synchronous. Every network calls are handled by 1 thread inside bacpypes. You must be careful when mixing with other threads or any async framework.

Kartstig, 2019-04-14:
Indeed. I'm using celery but I kept this to a single threaded worker because of that issue

ChristianTremblay, 2019-04-19:
Interesting. I don’t know celery. 
If your assumption is true, I would say not using`Con.devices`  This function will retrieve numerous informations about the devices which maybe you don’t need (manufacturer name, model,etc...). I’ve built it to get a sweet table in ipython when listing devices. And in your context, I suspect you really just need to get a short list of the devices. 

You could use `conn.discoveredDevices` Which hold the content of what Whois will find.

Kartstig, 2019-04-26:
I am also getting the following:
```
Traceback (most recent call last):
  File "/etc/beacon/beacon_web/batch/__init__.py", line 53, in discover
    for d in bactalk.find_devices(_connection)
  File "/etc/beacon/lib/bactalk.py", line 44, in find_devices
    return conn.devices
  File "/etc/beacon/venv/lib/python3.6/site-packages/BAC0/scripts/Lite.py", line      161, in devices
    '{} device {} objectName vendorName'.format(device[0], device[1]))
  File "/etc/beacon/venv/lib/python3.6/site-packages/BAC0/core/io/Read.py", line      290, in readMultiple
    reason = find_reason(apdu)
  File "/etc/beacon/venv/lib/python3.6/site-packages/BAC0/core/io/Read.py", line      555, in find_reason
    if apdu.pduType == RejectPDU.pduType:
AttributeError: 'KeyError' object has no attribute 'pduType'
```

If I use `discoveredDevices` I end up only getting something like the following:
```python
dict_items([(('192.168.1.10', 1000), 1)])
```

I think this will work fine for what I'm doing, but in case I wanted to get more information (name / manufacturer), how would I go about safely checking or loading?

I'm wondering if the `devices` list should be protected against exceptions such as KeyError like in the example above

ChristianTremblay, 2019-04-26:
What is this controller ?
I didn't know an apdu could be ripped off its type...
I'll check with bacpypes

ChristianTremblay, 2019-04-26:
Can you have a look to develop branch.... I've modified the code so you won't have the keyerror

Kartstig, 2019-04-30:
Sorry for the delayed response. I've been super busy and away from this code. I should be able to test this out later in the week

Kartstig, 2019-05-23:
@ChristianTremblay I was able to use the following version: `BAC0==19.4.30.2` and this issue is resolved

Thank you for your support!

closed 2019-05-23 by Kartstig