ChristianTremblay/BAC0 #133: AttributeError: 'FlaskServer' object has no attribute '_log'

bbartling, 2019-04-11:
I am tinkering around with trying to get the Full version of BAC0 going... SO I did a `pip install` of bokeh, flask-bootstrap, & flask...

I hope I didn't screw anything up but now I have an issue when I try to run BAC0 thru Python interpreter:

```
Python 3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:54:40) [MSC v.1900 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> import BAC0
>>> bacnet = BAC0.connect()
2019-04-11 14:35:06,425 - INFO    | Starting BAC0 version 0.99.944 (Complete)
2019-04-11 14:35:06,628 - INFO    | Using ip : 10.30.4.209
2019-04-11 14:35:06,644 - INFO    | Starting app...
2019-04-11 14:35:06,659 - INFO    | BAC0 started
2019-04-11 14:35:06,659 - INFO    | Registered as Simple BACnet/IP App
2019-04-11 14:35:08,674 - INFO    | Server started : http://10.30.4.209:8111
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\web\FlaskServer.py", line 148, in task
    self.startServer()
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\web\FlaskServer.py", line 48, in startServer
    self.flask_app.run(port=self.port, host='0.0.0.0')
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\flask\app.py", line 938, in run
    cli.show_server_banner(self.env, self.debug, self.name, False)
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\flask\cli.py", line 629, in show_server_banner
    click.echo(message)
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\click\utils.py", line 218, in echo
    file = _default_text_stdout()
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\click\_compat.py", line 675, in func
    rv = wrapper_func()
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\click\_compat.py", line 436, in get_text_stdout
    rv = _get_windows_console_stream(sys.stdout, encoding, errors)
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\click\_winconsole.py", line 295, in _get_windows_console_stream
    func = _stream_factories.get(f.fileno())
io.UnsupportedOperation: fileno

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\web\FlaskServer.py", line 41, in run
    self.process()
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\web\FlaskServer.py", line 45, in process
    self.task()
  File "C:\Users\bbartling\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\web\FlaskServer.py", line 150, in task
    self._log.warning('Flask server already running', err)
AttributeError: 'FlaskServer' object has no attribute '_log'
>>> 
```

ChristianTremblay, 2019-04-13:
It should not create any trouble... 
Are you able to run a simple Flask example script ? I would start by testing if Flask works alone.

bbartling, 2019-04-25:
I am having some issues running a flask server, I need to look into stackoverflow to see what the deal is. Even with a new computer installing Python 3.6 with all the packages with PIP I have the same issue.

bbartling, 2019-04-25:
Hi Christian,

I got BAC0 complete version to work (as with Flask on a simple hello world script..) when running the .py file from the command prompt on my Windows 10 laptop... Initially, I was trying to do this all thru the Python interpreter `>>>` where I ran into the issue trying to run complete version of BAC0... Out of curiosity do you use IPython for BAC0?

```
import BAC0
import time

bacnet = BAC0.connect()

while True:
    print("howdy")
    time.sleep(10)
```

[image]

Does one need a `while True` loop to keep the script alive? It dies if I dont have this... The embedded web page looks pretty cool Im looking forward to trying this out in the field...

Do you ever utilize some sort of a hot spot cell service device to check on controllers remotely with BAC0? IE, JACE isn't installed yet and it would be nice to keep an eye on something if I had a remote computer running BAC0 and some sort of a cell service/hot spot where I could remote into the BAC0 PC.

Any tips on how you use BAC0 would be greatly appreciated, especially remote field devices!!! Also no more JCI Fx for me its all Schneider electric now

Thanks,
Ben

ChristianTremblay, 2019-04-26:
Hi Ben, it will be cool, you will be able to test BAC0 on devices I don’t have access to and find new bugs/features. 

I use ipython a lot. And Jupiter Notebook. 

Typically, my use case is short term scripts. So I didn’t elaborate much on the long running script thing. Best would be to make some researches and make tests. I know some modules are made to build services from python scripts that can be started and stop in Windows services or Linux. That would be cool. 

Until now, I don’t rely on BAC0 for remote jobs. I do use raspberry pi with OpenVPN for temporary remote field access during construction and this RPi can be used with BAC0 if needed. My typical use case is startup/configuration phase. The Jace is always ready when data gathering is important. 

One example configuration use is k factor calculation for jobs where balancing tools cannot be used. I reversed engineered the computation and find the k factor myself using BAC0.

Another use case would be define a device, make a bunch of tests and as everything is trended automatically I know exactly what happened and this will ease troubleshooting. 

Those use cases are on site. The other use case is in the office before we go on-site. I define feedback functions that simulate the environment of the sequence. Start the fan, fan status go on. Start heating, discharge air will go up. It is very easy to come out with a simple test design that will be repeatable. This was the main goal of BAC0 when I designed it. Allowing me to test the sequences before being on-site, on real équipements.

bbartling, 2019-04-26:
Hi Christian, whenever you have a chance to respond could you tell me more about how you use BAC0 in the office to test HVAC sequences? Do you have a function script in Python that can simulate like an AHU or heating system?

Im trying to envision how this would work... Would you have a test bench in your office with a live PCG or PCA running a program and then use BAC0 to put an input "out of service" with a calculated value from a Python script? And then see how the program responds??? 

Do you have any github HVAC function scripts? That is hard to find to simulations of HVAC in a Python tool unless its an engineering tool for like building load simulations where it is overly complicated... Thank you!

ChristianTremblay, 2019-04-26:
Yes I work with live controllers I download them, then I start my script. So instead of manually doing "test" on points, everything is automatic.... almost everything.

You can have a look to this project : 
https://github.com/ChristianTremblay/ddcsequences

it is at a very early stage and work with
https://github.com/ChristianTremblay/ddcmath

closed 2019-05-16 by bbartling