ChristianTremblay/BAC0 #136: Unicast communication across subnets

dirkdevriendt, 2019-05-03:
First off, thanks for developing and maintaining such an impressive library! The attention to making it friction free and approachable is very much appreciated :-)
I understand my request is a probably a tad unorthodox, but bear with me if you will :-)

The absolute minimum to read a value (without doing identification/discovery etc) is something along the lines of:

```python
source_ip = "192.168.1.2/24"
target_ip = "10.10.10.11"
bacnet = BAC0.lite(source_ip)
r = bacnet.read("{} analogOutput 1 presentValue".format(target_ip))
```

As you see I'm trying to communicate with a device on a different subnet. In theory the read command triggers (or could trigger) a unicast request, so if the device in question replies with a unicast communication as well, that could work over standard IP routing mechanisms (i.e. without BBMD).

Is this the case or are there for instance other broadcast/multicast (whois/iam) requests being sent in the background that would prevent this from working?
Or do (some/most/all) BACnet devices require an initial handshake before they start responding to these read requests?

Safrone, 2019-05-03:
I would suggest testing but direct reading to a given address is possible since it should reply back to the requester barring some firewall settings perhaps.

Device discovery will not work since that relies on broadcasts over the subnet so `whois` is out. 

In my case, I was able to read from a bacnet device that was on a public IP (security concerns aside for now)

dirkdevriendt, 2019-05-03:
Thanks for those insights, I did test it and it didn't work, but I assume that can be because the device is replying with multicast or broadcast packets (it's in a restricted environment, so I can't monitor the remote network). The device in question is a JCI NAE controller.

I wanted to verify my general understanding of the library and double check if I was maybe missing some hidden broadcasts required by `BAC0` to be able to do its magic.

dirkdevriendt, 2019-05-03:
... and if there is currently a need for broadcast messages for some reason, I'd be interested to discuss the possibility of (and if it is possible, contribute to) a lightweight, unicast-only solution that would work across subnets.

ChristianTremblay, 2019-05-04:
@Safrone is correct and pointing directly to an IP should be sufficient to communicate with a device. It is in fact one of the biggest security issue with BACnet... It is super easy to connect to any public device not behind a firewall or a VPN.

Broadcast are really limited to whois and iam... or some timesync action (not yet supported by BAC0) and they should not be an issue most of the time. At least if you know where to point your requests.

BAC0 also support connecting to a BBMD as a foreign device. See https://bac0.readthedocs.io/en/latest/connect.html?highlight=bbmd#use-bac0-on-a-different-subnect-foreign-device
It will cover a lot of situations. If using BAC0 as a client, it is better than declaring BAC0 as a BBMD for your subnet... There are some parameters to respect when declaring BBMD (only one per subnet for example) which would create potential bug if done bad...

In your case, I suspect some kind of firewall blocking port UDP 47808 from crossing subnets.

Let me know if this doesn't answer well to your question or if you need more details.

ChristianTremblay, 2019-05-04:
Just saw that I will have to correct documentation : 
To do so, use the syntax:
```
my_ip = '10.8.0.2/24'
bbmdIP = '192.168.1.2:47808'
bbmdTTL = 900
bacnet = BAC0.connect(ip='xxx.xxx.xxx.xxx/mask', bbdmAddress=bbmdIP, bbmdTTL=bbmdTTL)```

dirkdevriendt, 2019-05-05:
This is indeed how I imagined it should work, I did some further tests and I can only conclude it must be either at the network level or maybe the device restricting itself to responding locally or by broadcast only.
So I'll close this issue and report back if it turns out there was something to learn from it after all :-)
Thanks again!

closed 2019-05-05 by dirkdevriendt

Safrone, 2019-05-06:
btw @dirkdevriendt I found wireshark/tshark extremely useful when debugging some bacnet issues early on. They have nice bacnet filters if you can sniff on the devices network.

In my case I found the responses were being directed towards another IP, probably the gateway or something like that.