ChristianTremblay/BAC0 #137: Unable to run BAC0

bbartling, 2019-05-16:
Hi Christian,

I am finally getting around to playing with these Schnieder boards, but would you know why I am getting this socket error?

```
Python 3.6.8 (tags/v3.6.8:3c6b436a57, Dec 24 2018, 00:16:47) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> import BAC0
>>> bacnet = BAC0.connect()
2019-05-16 11:26:10,512 - INFO    | Starting BAC0 version 0.99.944 (Complete)
2019-05-16 11:26:10,745 - INFO    | Using ip : 192.168.111.75
2019-05-16 11:26:10,774 - ERROR   | an error has occurred: [WinError 10013] An attempt was made to access a socket in a way forbidden by its access permissions
ERROR:BAC0:an error has occurred: [WinError 10013] An attempt was made to access a socket in a way forbidden by its access permissions
Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    bacnet = BAC0.connect()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\scripts\Complete.py", line 151, in __init__
    Lite.__init__(self, ip=ip, mask=mask, port=port, bbmdAddress=bbmdAddress, bbmdTTL=bbmdTTL)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\scripts\Lite.py", line 96, in __init__
    self.whois_answer = self.update_whois()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\scripts\Lite.py", line 100, in update_whois
    return (self.whois(), str(datetime.now()))
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\functions\WhoisIAm.py", line 57, in whois
    'BACnet stack not running - use startApp()')
BAC0.core.io.IOExceptions.ApplicationNotStarted: BACnet stack not running - use startApp()
>>> 
```

ChristianTremblay, 2019-05-17:
You sure no other software uses port 47808 ?

bbartling, 2019-05-21:
Hi Christian,

I think on this issue I somehow had two different BAC0 windows open...

Anyway I am diving into this Schneider code (looks pretty nasty). Its in Visio and what is lame you cant see live values unless special 'tags' are defined. PCT was pretty nice :^D

[image]

So in BAC0, I can this to work in full version to connect to the controller. But it appears these boards dont accept segmentation so BAC0 try's to read one point at a time, and there appears to a million of points/variables. The script below will connect but i never seen it finish reading everything..

```
import BAC0
import time

bacnet = BAC0.connect()

controller = BAC0.device('192.168.111.235', 101,
                         bacnet,
                         segmentation_supported=False)

while True:
    print(controller.points)
    time.sleep(10)
```

So I think I have to define some points so only a few are read... Is it possible to read the inputs circled red and then write some values to the OAT sensor to see what the program does? Its in the office on the test bench to simulate the sequences...

Im trying to get the OSA sensor defined, how do I go about that?

```
import BAC0
import time

bacnet = BAC0.connect()

# Define OAT
my_obj_list = [('file', 1),
             ('analogValue', AV300)]

controller = BAC0.device('192.168.111.235', 101,
                         bacnet, object_list = my_obj_list,
                         segmentation_supported=False)

while True:

    OAT = controller['AV300']
    
    value = OAT + 1 if value < 100 else 0
    time.sleep(10)

    controller['AV300'] = value
    

```

If I run that Ill get an error:

```
2019-05-21 14:44:32,771 - INFO    | Starting BAC0 version 0.99.944 (Complete)
2019-05-21 14:44:32,860 - INFO    | Using ip : 192.168.111.75
2019-05-21 14:44:32,862 - INFO    | Starting app...
2019-05-21 14:44:32,863 - INFO    | BAC0 started
2019-05-21 14:44:32,864 - INFO    | Registered as Simple BACnet/IP App
2019-05-21 14:44:34,870 - INFO    | Server started : http://192.168.111.75:8111
Traceback (most recent call last):
  File "tryit.py", line 8, in <module>
    ('analogValue', AV300)]
 * Serving Flask app "BAC0.web.FlaskServer" (lazy loading)
NameError: name 'AV300' is not defined
 * Environment: production
```

Thanks-
Ben

ChristianTremblay, 2019-05-21:
Wow. Visio programming. Did that in 2003 with Staefa controllers.

On your object list, don’t use AV300... use (analogValue,300)

From there it should work.

And don’t forget to use OAT.lastValue where possible to « not read every
time ». Especially when segmentation is not supported.

bbartling, 2019-05-22:
Christian,

In bacnet do I need to distinguish UIs from AIs in my BAC0 script?

I think this Schneider board has all UIs and a few BIs. Like UI 1 thru 4,
then BI 5 & 6.

Also in BAC0 do address for hardware points start at 0 and go to 4, for UI
1-5?

Thanks

bbartling, 2019-05-22:
Hi Christian,

Disregard that question above about UIs... I discovered this controller on a JACE and took some screenshots.. Im still having some issues tho, I get an error that AV300 doesnt exist but it exists on the JACE discovery. Any ideas to ry?

JACE screenshot:
[image]

```
C:\Users\benb\Desktop\BAC>py tryIt.py
2019-05-22 10:06:48,180 - INFO    | Starting BAC0 version 0.99.944 (Complete)
2019-05-22 10:06:48,196 - INFO    | Using ip : 192.168.111.75
2019-05-22 10:06:48,196 - INFO    | Starting app...
2019-05-22 10:06:48,196 - INFO    | BAC0 started
2019-05-22 10:06:48,196 - INFO    | Registered as Simple BACnet/IP App
2019-05-22 10:06:50,212 - INFO    | Server started : http://192.168.111.75:8111
2019-05-22 10:06:50,212 - INFO    | Changing device state to DeviceDisconnected'>
 * Serving Flask app "BAC0.web.FlaskServer" (lazy loading)
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-22 10:06:50,566 - INFO    | Changing device state to RPDeviceConnected'>
2019-05-22 10:06:50,682 - INFO    | Device 101:[HP_LOOP] found... building points list
2019-05-22 10:06:50,807 - WARNING | Unknown property 192.168.111.235 analogValue 300 description
WARNING:BAC0:Unknown property 192.168.111.235 analogValue 300 description
2019-05-22 10:06:50,932 - INFO    | Ready!
2019-05-22 10:06:50,932 - INFO    | Polling started, values read every 10 seconds
2019-05-22 10:06:50,932 - ERROR   | AV300 doesn't exist in controller
ERROR:BAC0:AV300 doesn't exist in controller
Traceback (most recent call last):
  File "tryIt.py", line 26, in <module>
    OAT = controller['AV300'].lastValue
AttributeError: 'NoneType' object has no attribute 'lastValue'
```

bbartling, 2019-05-22:
```
import BAC0
import time

# counting function for OAT
def count(OAT):
    OAT = OAT + 1
    if OAT < 100:
        return 0
    else:
        return OAT

# create BACnet IP device on local PC
bacnet = BAC0.connect()

# Define OAT
my_obj_list = [('file', 1),('analogValue',300)]

# Define MNB1000 board
controller = BAC0.device('192.168.111.235', 101,
                         bacnet, object_list = my_obj_list,
                         segmentation_supported=False)

while True:
    #define OAT
    OAT = controller['AV300'].lastValue    
    newOAT = count(OAT)
        
    time.sleep(10)

    #Write a value to OAT
    controller['AV300'] = newOAT
    print(newOAT)
    

```

bbartling, 2019-05-22:
Sorry to keep bombarding you, but I have just attempted to run this code below without defining points.

```
import BAC0
import time

# create BACnet IP device on local PC
bacnet = BAC0.connect()

# Define MNB1000 board
controller = BAC0.device('192.168.111.235', 101, bacnet)

#print points
print(controller.points)  
```

I can see that BAC0 recognizes segmentation isn't supported and it does read each point which takes a few minutes for it to do that. But something happens on the printing the `controller.points`

Any tips greatly appreciated as well as your time :) Heres the last bits of the trace back:

```
WARNING:BAC0:Unknown property 192.168.111.235 binaryValue 12501 description
2019-05-22 10:23:23,694 - WARNING | Unknown property 192.168.111.235 binaryValue 722 description
WARNING:BAC0:Unknown property 192.168.111.235 binaryValue 722 description
2019-05-22 10:23:24,005 - WARNING | Unknown property 192.168.111.235 binaryValue 720 description
WARNING:BAC0:Unknown property 192.168.111.235 binaryValue 720 description
2019-05-22 10:23:24,314 - WARNING | Unknown property 192.168.111.235 binaryValue 13353 description
WARNING:BAC0:Unknown property 192.168.111.235 binaryValue 13353 description
2019-05-22 10:23:24,495 - INFO    | Ready!
2019-05-22 10:23:24,497 - INFO    | Polling started, values read every 10 seconds
Exception in thread rpm_poll:
Traceback (most recent call last):
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 45, in run
    self.process()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 52, in process
    self.task()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\tasks\Poll.py", line 77, in task
    self.device.read_multiple(list(self.device.points_name), points_per_request=25)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py", line 452, in read_multiple
    self.read_single(each,points_per_request=1, discover_request=discover_request)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py", line 459, in read_single
    return self.properties.network.read(request)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\io\Read.py", line 87, in read
    args_split, arr_index=arr_index, vendor_id=vendor_id, bacoid=bacoid))
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\io\Read.py", line 310, in build_rp_request
    addr, obj_type, obj_inst, prop_id = args[:4]
ValueError: not enough values to unpack (expected 4, got 2)

Traceback (most recent call last):
  File "tryIt2.py", line 13, in <module>
    print(controller.points)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\Points.py", line 666, in __repr__
    return '{}/{} : {}'.format(self.properties.device.properties.name, self.properties.name, self.enumValue)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\Points.py", line 639, in enumValue
    return self.properties.units_state[int(self.lastValue) - 1]
TypeError: int() argument must be a string, a bytes-like object or a number, not 'tuple'
```

ChristianTremblay, 2019-05-23:
First I would try : 
```
controller['OaT'].lastValue
```

On the description issue... I'll have to look at it.... not sure you are at latest version anyway. Please upgrade and send me more details.

bbartling, 2019-05-23:
Just thru some experimentation on a different job, with an MSTP controller I can get this code to work where it prints the `chiller.points` one time. But I get an error if I try to use `chiller['OaT'].lastValue`

```
import BAC0
import time

bacnet = BAC0.lite()

# Define OAT
my_obj_list = [('file', 1),('analogValue',300),
               ('analogInput', 3),
               ('analogInput', 4),
               ('analogInput', 5),
               ('analogOutput', 1),
               ('analogOutput', 2),
               ('analogValue', 710)]

chiller = BAC0.device('10:43', 1043, bacnet,
                      object_list = my_obj_list,
                      segmentation_supported=False)

while True:
    
    #print(chiller['OaT'].lastValue)
    print(chiller.points)
    time.sleep(60)

    if KeyboardInterrupt:
        bacnet.disconnect()
```

This will print `chiller.points` just one time but then errors out I think when the script goes thru the `time.sleep` for another `print(chiller.points`). I also did a `pip --upgrade` too where BAC0 updated

```
Python 3.6.8 (tags/v3.6.8:3c6b436a57, Dec 24 2018, 00:16:47) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
================ RESTART: C:\Users\benb\Desktop\BAC\tryIt3.py ================
2019-05-23 10:31:21,929 - INFO    | Starting BAC0 version 19.4.30.2 (Lite)
2019-05-23 10:31:22,093 - INFO    | Using ip : 10.10.100.4
2019-05-23 10:31:22,109 - INFO    | Starting app...
2019-05-23 10:31:22,115 - INFO    | BAC0 started
2019-05-23 10:31:22,121 - INFO    | Registered as Simple BACnet/IP App
2019-05-23 10:31:24,126 - INFO    | Changing device state to DeviceDisconnected'>
2019-05-23 10:31:25,069 - INFO    | Changing device state to RPDeviceConnected'>
2019-05-23 10:31:26,363 - INFO    | Device 1043:[CHILLER2] found... building points list
2019-05-23 10:31:30,939 - INFO    | Ready!
2019-05-23 10:31:30,942 - INFO    | Polling started, values read every 10 seconds
Exception in thread rpm_poll:
Traceback (most recent call last):
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 44, in run
    self.process()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 50, in process
    self.task()
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\tasks\Poll.py", line 79, in task
    list(self.device.points_name), points_per_request=25
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py", line 550, in read_multiple
    each, points_per_request=1, discover_request=discover_request
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py", line 560, in read_single
    return self.properties.network.read(request)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\io\Read.py", line 110, in read
    args_split, arr_index=arr_index, vendor_id=vendor_id, bacoid=bacoid
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\io\Read.py", line 335, in build_rp_request
    addr, obj_type, obj_inst, prop_id = args[:4]
ValueError: not enough values to unpack (expected 4, got 2)

[CHILLER2/OaT : 50.15 degreesFahrenheit, CHILLER2/CwDP : 5.38 poundsForcePerSquareInch, CHILLER2/CwsT : 71.14 degreesFahrenheit, CHILLER2/CwrT : 71.26 degreesFahrenheit, CHILLER2/CwpAVfd : 0.00 percent, CHILLER2/CwpBVfd : 0.00 percent, CHILLER2/CwstStpt : 50.57 degreesFahrenheit]
2019-05-23 10:32:32,575 - INFO    | BACnet stopped
Traceback (most recent call last):
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\Points.py", line 125, in value
    str(self.properties.address),
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\io\Read.py", line 97, in read
    raise ApplicationNotStarted("BACnet stack not running - use startApp()")
BAC0.core.io.IOExceptions.ApplicationNotStarted: BACnet stack not running - use startApp()

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\benb\Desktop\BAC\tryIt3.py", line 23, in <module>
    print(chiller.points)
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\Points.py", line 563, in __repr__
    self.value,
  File "C:\Users\benb\AppData\Local\Programs\Python\Python36\lib\site-packages\BAC0\core\devices\Points.py", line 130, in value
    raise Exception("Problem reading : {}".format(self.properties.name))
Exception: Problem reading : OaT
>>> 
```

In the traceback if you see right above:
`2019-05-23 10:32:32,575 - INFO    | BACnet stopped`
I did retrieve all of the point values once, but it doesnt appear to not let me print a value by point name.

Just FYI :)

bbartling, 2019-05-23:
I have been running the script a few times with uncommenting the `chiller['OaT'].lastValue` and it does appear to be giving me a like a saved value from memory or something...

ChristianTremblay, 2019-05-23:
```
while True:
    
    #print(chiller['OaT'].lastValue)
    print(chiller.points)
    time.sleep(60)

    if KeyboardInterrupt:
        bacnet.disconnect()
```

This part... I'm not sure... you are not quiting the loop.... so it'll try to read the point again. And in fact, it creates errors at the end.

Each time a point is read, BAC0 fills a list for all points.
Look here : https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/Points.py#L68

`_history` property contains two list... timestamp and value...

When you call `.history` BAC0 will turn those two into a pandas serie (pandas series are IMO the best tool to deal with time series...)

`.lastValue` is just a shortcut to the last entry of `.history`

So the basic idea is to let BAC0 device do the polling... no need to read points directly. Polling will use read property multiple which is optimized vs one message per read. Accessing the last read point should be sufficient in most applications.

If you need the "value" part of a point :  `chiller['OaT'].value` it will return just the value... instead of the "point object" containing everything (like properties, etc...)

bbartling, 2019-05-29:
Do you use IPython alot with BAC0?

ChristianTremblay, 2019-05-29:
Mostly. And Jupiter Notebooks

ChristianTremblay, 2019-06-20:
@bbartling 
The error you reported I saw that with device that do not support segmentation and the develop branch contains a fix for that. Please try and let me know if it works for you.

bbartling, 2019-06-20:
Hi Christian, can you give me a tip for installing delevop branch? I can't remember do I git clone the repository and then run it somehow in develop mode?

 I recently just downloaded anaconda 3.7 so I have Ipython now too. Would I need to pip install BAC0 as well?

ChristianTremblay, 2019-06-20:
You 
```
git clone https://github.com/ChristianTremblay/BAC0
cd BAC0
git checkout develop
python setup.py install
```

ChristianTremblay, 2019-07-03:
Still need this @bbartling ?

bbartling, 2019-07-03:
Yes sorry for the delay Ill be able to test later on this week

bbartling, 2019-07-08:
It seems like its working to me. Is it possible to tell if I am running in full or lite version?
[image]

bbartling, 2019-07-08:
On these Schneider electric boards (I dont have this learned into a JACE or anything) in the Visio program you need 'write' to the point like an OSA sensor to test sequences. For example to test the mechanical sequences that change with outside air temperature in Visio I have to double click on the block, type in a value:
[image]

And then right click 'write' for the controller to accept the value:
[image]

Is it possible to do that in BAC0? It appeared to have accepted the new value of 50 but its almost like I couldnt do the 'write' for the Schneider board to accept the value of 50.

[image]

Anyways thanks for your help... If its possible to do some office/test bench of sequences like this it would be pretty cool to change OSA values and watch the Bokeh outputs of graphics etc... :)

bbartling, 2019-07-10:
I think I got this to work Christian on a pump differential pressure point and flow meter.  On the test bench in the office, I can write to the controller with BAC0 on an AI (dp) & AV (flow) with a sign wave function and on a different computer running the Visio Schneider software we can verify the mechanical sequences are correct by watching the values as if it was a live HVAC system. 

:^D Pretty cool, working really good in Jupyter lab note books

```
import numpy as np
import time

def sinWave(rng,stp):
    x = np.arange(0,4*np.pi,.1)   # start,stop,step
    return np.sin(x) *rng + stp

dp = sinWave(2,20)
flow = sinWave(30,180)

while True:
    for x in range(len(flow)):
        time.sleep(.5)
        controller['DiffPr'] = dp[x]
        controller['Flw'] = flow[x]
```

bbartling, 2019-07-10:
The only thing I can tell is if I am running full or Lite version of BAC0. It must be lite version as I dont think I installed the packages yet

closed 2019-07-10 by bbartling

bbartling, 2019-07-10:
https://gist.github.com/bbartling/b6637cf4ada139332ec16378a5c078e6

ChristianTremblay, 2019-07-11:
Excellent ! I'll try to document specific sequence functions one day...
It allow an easy way to make inputs "match the value" of an output... for example : 
```
device['SF-S'].match(device['SF-C'])
```
you can even define function and make the variable match the result : 
```
def function():
    deltaT_of_SCR = 20
    gain = device['RH-O'].lastValue * deltaT_of_SCR / 100
    return device['MA-T'].lastValue + gain

device['DA-T'].match_value(function)
```

I've used threads for that so you will be able to continue working on the Notebook and make other changes... because actually your while loop is blocking everything. Match are called every 10(or was it 5?) seconds... so it creates some kind of "real life effect".

PS : I may have some typos here... going from memory but I'm sure you see the idea behind.

ChristianTremblay, 2019-07-11:
To run complete mode... you will need flask, flak_bootstrap, bokeh....

I still need to work a lot on the logger function and I should find a way to tell the user what is missing from their install to run the complete mode.