ChristianTremblay/BAC0 #138: Is there a way to print out an object's priorityArray?

mosterme1003, 2019-05-17:
Hello everyone!
I try to read out the priorityArray of an analogValue because I wanna know whether the presentValue is overwritten on priority 8 or not. 

`bacnet = BAC0.connect(ip=ipLocal)`
`array = bacnet.read('%s analogValue 1 priorityArray' %(ipDevice))`

When I try to read the propertyArray and print it I receive: 
> <bacpypes.basetypes.PriorityArray object at 0x00000...>

When I try to iterate the array and print each priorityValue I receive:
> <bacpypes.basetypes.PriorityValue object at 0x00000...>

None of these objects seem to have a __str__ method. Is there a way to get the value of each priority element? I would appreciate your help. Thank you!

ChristianTremblay, 2019-05-17:
If you define a device, each point can be verified using is_overridden property. 
https://bac0.readthedocs.io/en/latest/_modules/BAC0/core/devices/Points.html?highlight=Is_overridden

mosterme1003, 2019-05-17:
Perfect. Thank you!

closed 2019-05-17 by mosterme1003