ChristianTremblay/BAC0 #140: Maximum Recursion Depth Exceeded on BAC0.connect() and sometimes on other commands

LincolnL, 2019-05-23:
Executing

`import BAC0`
`bacnet=BAC0.connect("192.168.123.201/24")`

results in recursive launching of flask servers and crash of python.

`2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
2019-05-23 11:23:09,700 - INFO    |  * Running on http://0.0.0.0:8111/ (Press CTRL+C to quit)
`
...ad nauseum until crash

`BAC0.version`
`'19.4.30.2'`
Python 3.7.3 on Windows 10

ChristianTremblay, 2019-05-23:
Can you check if you are able to test a simple Flask example... just to be sure it's not a Flask issue on your installation ?

Does BAC0.lite() works ? (no web server)

LincolnL, 2019-05-24:
BAC0.lite() works, but I also ended getting with a RecursionDepthExceeded error when I did the same script with 
`bacnet=BAC0.lite()
bacnet.whois()
lgr=bacnet.devices[0]`
The recursion error had something to do with ReadPropertyMultiple, but I didn't think to capture error when it happened, and I haven't been able to reproduce the error today.  It seems relevant though (maybe).

ChristianTremblay, 2019-05-28:
If you ever reproduce, please send me more details... I never saw this but there are so many weird devices out there :)

LincolnL, 2019-05-28:
This was an LGR25 from Automated Logic that I was connecting to, but I
think it was a problem with my environment, because I don't have the
problem on my other machines.  You can probably close this issue.

fgervais, 2019-06-17:
I think I get the same thing here. It looks like as soon as something gets logged, it gets into a recursive loop.

```
python --version
Python 3.7.3

pipenv graph
BAC0==0.99.944
  - bacpypes [required: Any, installed: 0.17.6]
```

I can see I'm running the same python version as @LincolnL when here originally reported the issue so there might be something with that version.

I can reproduce with this code:

```
import BAC0

import logging
logging.getLogger().setLevel(logging.DEBUG)    

bacnet = BAC0.lite()
dut = BAC0.device("10.59.50.43", "403", bacnet)
bacnet.disconnect()
exit()
```

```
python3 log_test_with_cronjob.py 
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)

[truncated]

2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.7/logging/__init__.py", line 1034, in emit
    msg = self.format(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 880, in format
    return fmt.format(record)
  File "/usr/lib/python3.7/logging/__init__.py", line 620, in format
    if self.usesTime():
  File "/usr/lib/python3.7/logging/__init__.py", line 588, in usesTime
    return self._style.usesTime()
  File "/usr/lib/python3.7/logging/__init__.py", line 430, in usesTime
    return self._fmt.find(self.asctime_search) >= 0
RecursionError: maximum recursion depth exceeded while calling a Python object
Call stack:
  File "log_test_with_cronjob.py", line 23, in <module>
    bacnet = BAC0.lite()
  File "/home/fgervais/4tb/ai/BAC0/BAC0/scripts/Lite.py", line 69, in __init__
    version, self.__module__.split('.')[-1]))
Message: 'Starting BAC0 version 0.99.944 (Lite)'
Arguments: ()
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
2019-06-17 18:40:18,751 - INFO    | Starting BAC0 version 0.99.944 (Lite)
[... 25 lines cut ...]
  File "/usr/lib/python3.7/logging/__init__.py", line 880 in format
  File "/usr/lib/python3.7/logging/__init__.py", line 1034 in emit
  File "/usr/lib/python3.7/logging/__init__.py", line 905 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle

[truncated]

  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  File "/usr/lib/python3.7/logging/__init__.py", line 1591 in callHandlers
  File "/usr/lib/python3.7/logging/__init__.py", line 1529 in handle
  ...
Aborted (core dumped)
```

ChristianTremblay, 2019-06-18:
Reopening.... I myself saw this behaviour on some devices. Can't yet tell why...

ChristianTremblay, 2019-06-18:
Can anyone validate it happens when segmentation errors occur ?

fgervais, 2019-06-18:
I'm not sure how I can confirm that.

Is this something I would see with wireshark?

ChristianTremblay, 2019-06-18:
In wireshark, I see the last request with a "abort" response. Recursion seems to occur when BAC0 tries to reconnect to the device without using segmentation. 
My implementation is not so good for those case. 
For example, I have this trouble with TEC3000 from JCI when reading a multistate value with 255 different stateText. The answer bust the segmentatioin limit... for this only few points... it would mean I would turn off segmentation entirely for the device... which will slow things down a lot... with no real reason.

I must find a way to recover from that and use a simple read when necessary.

For now, I think the crash comes from logging being unable to handle the recursion... but if no logging was crashing, I would probably end in an infinite loop of reconnection failing.

Not cool.

fgervais, 2019-06-18:
I think is fails before doing any transfers. See it fails even with this patch:

```
diff --git a/BAC0/scripts/Lite.py b/BAC0/scripts/Lite.py
index c713be7..9633e25 100755
--- a/BAC0/scripts/Lite.py
+++ b/BAC0/scripts/Lite.py
@@ -67,6 +67,7 @@ class Lite(Base, WhoisIAm, ReadProperty, WriteProperty, Simulation):
     def __init__(self, ip=None, port=None, mask = None, bbmdAddress=None, bbmdTTL=0):
         self._log.info("Starting BAC0 version {} ({})".format(
             version, self.__module__.split('.')[-1]))
+        exit()
         self._log.debug("Configurating app")
         self._registered_devices = weakref.WeakValueDictionary()
         if ip is None:
```

If I put the `exit()` just 1 line before, before the log call, it does not fail.

fgervais, 2019-06-18:
Yes it seems. 

If I apply the [proposed fix](https://github.com/python/cpython/commit/6a7a9f1d83cef628d2bacd71ee568b93f53fd6b4) manually it does not crash python. I still get the recursion though.

I think that would mean that there is a recursive logging somewhere.

Maybe the way logging is configured?

ChristianTremblay, 2019-06-18:
Probably I did something wrong somewhere... you can disable it by tweaking the note.py file. I guess removing all logs will work :-)

fgervais, 2019-06-18:
Ok I'll see if I can find something

fgervais, 2019-06-18:
One thing that fixes it on my end is this:

```
diff --git a/BAC0/core/utils/notes.py b/BAC0/core/utils/notes.py
index c6fff8a..a129672 100644
--- a/BAC0/core/utils/notes.py
+++ b/BAC0/core/utils/notes.py
@@ -100,8 +100,8 @@ def note_and_log(cls):
     cls.logname = '{} | {}'.format(cls.__module__, cls.__name__)
     root_logger = logging.getLogger()
     cls._log = logging.getLogger('BAC0')
-    if not len(root_logger.handlers):
-        root_logger.addHandler(cls._log)
+    #if not len(root_logger.handlers):
+    #    root_logger.addHandler(cls._log)
 
     # Console Handler
     ch = logging.StreamHandler()
```

Maybe it breaks something else I'm not sure. Nothing obvious on my end at least.

LincolnL, 2019-06-19:
I'm not very experienced with the Logging library, and only recreationally use Python, but shouldn't `root_logger = "BAC0"`, and the cls._log be `logging.getLogger(cls.logname)`, and then only attach handlers to the root_logger, and let propagation handle the message passing?

ChristianTremblay, 2019-06-19:
Thanks @LincolnL, I'll have a look. For now, I removed the addHandler thing as Francois found. It's temporary, goal being to be able to have something working.

For now, @fgervais and @LincolnL I would be very grateful if you could try the develop branch. I tried to stabilize the thing and I hope it'll work on your side.

ChristianTremblay, 2019-06-20:
Starting to think that playing with root logger is probably the source of the recursion trouble... and it is probably not needed for a library to interfere with it... so I'll just won't touch it and remove the part where I add handler to root logger.

closed 2019-06-20 by ChristianTremblay