ChristianTremblay/BAC0 #142: Cannot discover points for controller

Kartstig, 2019-06-05:
Context:
- Device: Carrier
- Lots of points on the device (1033)

I am able to find the device via `bacnet.devices`:
```python
>>> import BAC0
>>> bacnet = BAC0.lite(maxAPDULengthAccepted="1024000", maxSegmentsAccepted="1024000")
2019-06-05 13:14:31,574 - INFO    | Starting BAC0 version 19.5.22dev (Lite)
2019-06-05 13:14:36,634 - INFO    | Using ip : 10.58.32.201
2019-06-05 13:14:36,635 - INFO    | <class 'bacpypes.pdu.Address'>
2019-06-05 13:14:36,641 - INFO    | Starting app...
2019-06-05 13:14:36,643 - INFO    | BAC0 started
2019-06-05 13:14:36,644 - INFO    | Registered as Simple BACnet/IP App
>>> bacnet.devices
2019-06-05 13:14:45,504 - WARNING | APDU Abort Reject Reason : unrecognizedService
WARNING:BAC0:APDU Abort Reject Reason : unrecognizedService
10.58.32.5
249999
[('i-Vu Plus 249999', 'Carrier Corporation', '10.58.32.5', 249999), ('PS3037-E  P371905157', 'DENT Instruments, Inc.', '10.58.32.204', 215896), ('device249912', 'Carrier Corporation', '10.58.32.8', 249912), ('device249913', 'Carrier Corporation', '10.58.32.7', 249913), ('device249911', 'Carrier Corporation', '10.58.32.6', 249911)]

```

However, when I create a controller, I can't find the points on it:

```
>>> controller = BAC0.device('10.58.32.6', 249911, bacnet)
2019-06-05 13:15:36,867 - INFO    | Changing device state to DeviceDisconnected'>
2019-06-05 13:15:36,902 - INFO    | Changing device state to RPMDeviceConnected'>
2019-06-05 13:15:36,936 - INFO    | Device 249911:[device249911] found... building points list
2019-06-05 13:15:37,610 - WARNING | Offline: provide database name to load stored data.
WARNING:BAC0:Offline: provide database name to load stored data.
2019-06-05 13:15:37,611 - WARNING | Ex. controller.connect(db = 'backup')
WARNING:BAC0:Ex. controller.connect(db = 'backup')
>>> controller.points
[]
```

I looked at https://github.com/ChristianTremblay/BAC0/issues/38 and decided to try and do `bacnet.read` and I was successfully able to find the points:
```python
>>> points = bacnet.read('10.58.32.6 device 249911 objectList')
>>> len(points)
1033
>>> points[0]
(900, 1)
>>> points[-1]
('binaryValue', 684)
```

I also tried to create the BAC0.device with `segmentation_supported=False` but that didn't work either

My usage of BAC0 has been primarily for the convenience of connecting and discovering points, so if we can figure out these

ChristianTremblay, 2019-06-12:
Having 1033 points to read regularly on the device is not a good idea. I think you should create a custom_object list with the points that you need and use this custom object list when defining the device.
I'm almost certain that a lot of those 1033 points are useless.

https://bac0.readthedocs.io/en/latest/controller.html?highlight=custom%20object%20list#object-list

Kartstig, 2019-06-12:
> Having 1033 points to read regularly on the device is not a good idea

👍 💯 
Agreed! All the points will not be read regularly, but only at an initial setup mode where we need to choose a point by name and/or type. 

I haven't found a way around `bacnet.read('10.58.32.6 device 249911 objectList')` in my situation. I'm unable to discover any points using the traditional method. For some reason, `controller.points` is empty.

```python
controller.points
[]
```

Is there another way to force this to be repopulated?

Kartstig, 2019-06-12:
After reading manually, it is working. However, I'm encountering another issue:
```
BAC0.core.io.IOExceptions.UnknownPropertyError: Unknown property 192.168.10.180 analogValue 56 profileName
[2019-06-12 09:19:17,585: ERROR/ForkPoolWorker-1] Unknown property 192.168.10.180 analogValue 57 profileName
Traceback (most recent call last):
  File "/home/herman/dev/PyBacnet/lib/bactalk.py", line 75, in read_point_key
    return conn.read(f'{ip_address} {point_type} {name} {key}')
  File "/home/herman/dev/PyBacnet/venv/lib/python3.6/site-packages/BAC0/core/io/Read.py", line 176, in read
    raise UnknownPropertyError("Unknown property {}".format(args))
BAC0.core.io.IOExceptions.UnknownPropertyError: Unknown property 192.168.10.180 analogValue 57 profileName
```

Per bacnet standard, it seems like `profileName` is a required field on `AnalogValue` :thinking: 

[image]

Shouldn't this property always return? (This may not be a BAC0 issue, but a device / point not abiding by the standard)

ChristianTremblay, 2019-06-14:
Those are for Analog Value Functional blocks which are different than Analog Values. From what I understand, those points are BACnet representation of non-BACnet point (like KNX). You are not looking at the right section.

Kartstig, 2019-06-14:
Ah okay. Thank you. I will remove that property from the read command.

One question no have is whether or not you want to raise an exception when a property is not available. Is this a Bacpypes behavior or BAC0?

If we can safe read, then it may be more performant. At the moment I'm wrapping everything in a try / except to avoid halting a scan. 

If there's anything to do to get `device.points` to work, let me know, otherwise I'll close this issue.

I appreciate your help! I'm a Bacnet noob

ChristianTremblay, 2019-07-03:
I know it's been a while you had this bug, but did you figured out how to make it work ?

Kartstig, 2019-07-03:
I'm using `Read.py` and the following:
```python
def read_point_key(conn, ip_address, point_tuple, key):
    point_type, name = point_tuple
    try:
        return conn.read(f'{ip_address} {point_type} {name} {key}')
    except Exception as e:
        logger.exception(e)
        sentry_sdk.capture_exception(e)
        return None
```

I think this is fine for my workaround

Kartstig, 2019-07-03:
If there's a process for figuring out why some of these points don't get placed into the `device.points` method, let me know. I'm happy to submit a PR to help improve the discovery of devices on BAC0. Otherwise I'm closing this, as I've found a workaround for my situation

closed 2019-07-03 by Kartstig

ChristianTremblay, 2019-07-03:
Many points don't appear in `device.points` because I implemented (for now) analogInput, analogOutput, analogValue, binaryInput, binaryOutput, binaryValue, multiStateInput, multiStateOutput, multiStateValue and trendLogs. Everything else is not yet implemented in device.... mostly because I don't use anything else 99% of the time.

mohamednayif, 2022-08-26:
I tried to connect to a bacnet device which succeeded, but didn't shows any point. 

`bacnet = BAC0.lite(maxAPDULengthAccepted="1024000", maxSegmentsAccepted="1024000")`

2022-08-26 10:43:06,949 - INFO    | Starting BAC0 version 21.12.03 (Lite)
2022-08-26 10:43:06,950 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2022-08-26 10:43:06,950 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2022-08-26 10:43:06,950 - INFO    | Starting TaskManager
2022-08-26 10:43:06,957 - WARNING | Netifaces not installed on your system. BAC0 can't detect the subnet.
Please provide subnet for now, we'll consider 255.255.255.0 (/24).
You can install netifaces using 'pip install netifaces'.
2022-08-26 10:43:06,958 - INFO    | Using ip : 10.0.1.216
2022-08-26 10:43:07,027 - INFO    | Starting app...
2022-08-26 10:43:07,027 - INFO    | BAC0 started
2022-08-26 10:43:07,028 - INFO    | Registered as Simple BACnet/IP App
2022-08-26 10:43:07,063 - INFO    | Update Local COV Task started

`mycontroller = BAC0.device('10.0.1.40', 79, bacnet)`

2022-08-26 10:43:21,266 - INFO    | Changing device state to DeviceDisconnected'>
2022-08-26 10:43:21,494 - INFO    | Changing device state to RPMDeviceConnected'>
2022-08-26 10:43:21,682 - INFO    | Device 79:[NETIX_DEMO_POINTS_79] found... building points list
2022-08-26 10:43:21,746 - INFO    | Ready!
2022-08-26 10:43:21,747 - INFO    | Device defined for normal polling with a delay of 10sec
2022-08-26 10:43:21,747 - INFO    | Polling started, values read every 10 seconds

`bacnet.devices`

[('NETIX_DEMO_POINTS_79', 'Tridium', '10.0.1.40', 79), ('CAR_PARK_01_15', 'Tridium', '10.0.1.127', 15)]

But no points are showing.

`mycontroller.points`

[]
[image]
[image]

ChristianTremblay, 2022-08-26:
For a Jace to expose bacnet points you need to configure a bacnet export table. Refer to Niagara doc.

bbartling, 2022-08-26:
BACnet export tables are really easy to setup on the JACE....from what I can remember its like a point discovery process but on the JACE itself, then the points magically show up as discoverable BACnet points underneath the JACE. this PDF looks a little dated but hopefully its gives you some tips..
[Bacnet Export Table (2).pdf](https://github.com/ChristianTremblay/BAC0/files/9433116/Bacnet.Export.Table.2.pdf)
.

bbartling, 2022-08-29:
Make sure when you pull or map in the points of the BACnet device to the JACE on the initial BACnet discovery process before making the export table that the points are mapped in as "writeable" and "not read only" which I think is default. This may help else Im sure there is a million YouTubes on Workbench discovery process for BACnet.

https://hvac-talk.com/vbb/attachment.php?attachmentid=798687&d=1520972559