ChristianTremblay/BAC0 #144: Cannot stablish a connection between a device simulator on PC and a Raspberry

HAMH96, 2019-06-14:
Hello, I am tryring to make a BACnet server on a Raspberry and python code, so I made use that code lines to start the communication with a simulated device:

import BAC0
bacnet=BAC0.connect(ip='192.168.1.5/24')
bacnet.devices 

I connect the Raspberry with a PC via ethernet LAN cable, the PC has a BACnet device simulator made by CBMS Studio, in which I added a BACnet device and an object, I do this because I have not access to a real BACnet device yet, the simulator says that the network adapter has the same IP address that I wrote in the codeline above, but when I run the code using Python 3.4.2 shell in the Raspberry, it gives me the following error:

Traceback (most recent call last):
  File "/home/pi/bacnet.py", line 2, in <module>
    bacnet=BAC0.connect(ip='192.168.1.5/24')
  File "/usr/local/lib/python3.4/dist-packages/BAC0/scripts/Lite.py", line 96, in __init__
    self, localIPAddr=ip_addr, bbmdAddress=bbmdAddress, bbmdTTL=bbmdTTL
  File "/usr/local/lib/python3.4/dist-packages/BAC0/scripts/Base.py", line 91, in __init__
    "IP Address provided ({}) invalid.".format(localIPAddr)
BAC0.core.io.IOExceptions.InitializationError: IP Address provided (192.168.1.5) invalid.

can you help me please?

ChristianTremblay, 2019-06-14:
The IP must the IP of the Rapsberry. The local IP of the device running BAC0. 

Also, have a look to the test folder. There are example running two different instances of BAC0 talking to each other. Could be another way to make some tests.

HAMH96, 2019-06-14:
I had a suspicion about that, and also tried with the raspberry IP address and not only there was no error but also there was no return of the devices list, like it was frozen, I had to press ctrl+c to interrupt it

ChristianTremblay, 2019-06-14:
start with `bacnet.whois()` instead of `bacnet.devices`. It is supposed to output a quick list of devices. Also, be sure that the RPi is in the same subnet than the PC.

ChristianTremblay, 2019-06-14:
You can send this whois a couple time and see things appearing

HAMH96, 2019-06-17:
Well, I tried using bacnet.whois() instead, but there is no output of the devices only I see this in the shell
[image]
If I type bacnet.whois() or bacnet.devices, I got the answer:
[image]
but this is not what I need, because I need this to appear automatically when I run the code.
So, keeping doing things I added to my code the following line:

mycontroller = BAC0.device('192.168.1.4',1001,bacnet)

and when I run the code appears a very very long error message
like this:
[image]

[image]

[image]

Kartstig, 2019-06-18:
According to your results from `bacnet.devices` you are only able to find one controller:
[image]

Then you attempt to create an instance of a controller for `192.168.1.4`. There is no device 1001 @ 192.168.1.4 so BAC0 will assume it needs to work in offline mode. Try using the proper IP and see if that fixes your issue

HAMH96, 2019-06-18:
Oh, thank you so much Karstig, finally it is working!! and also I want to add that if I want to see the response of bacnet.whois() or bacnet.devices, I have to put them inside a print()

ChristianTremblay, 2019-06-19:
The time where other support others... pretty neat !

ChristianTremblay, 2019-06-19:
@Kartstig @HAMH96 
I'm working on the recursion error when connecting to devices actuall on the develop branch. If you can test it on your side, it would be great.

HAMH96, 2019-06-21:
Thanks to all for the help, I have another question, how can I get the Object Type of each Object of all devices connected?

HAMH96, 2019-06-26:
Hello, could you help me please answering my last question? also I want to know how to know how many objects and which objects a device have, because I have a simulator and I added an AccessCredential and an AccessDoor objects, but they are not listed by ".points()" and if I try to run for example: control1['AccesDoor1'] in order to get the state of the door, an error appears saying: "AccessDoor1" doesn't exist in the controller, please help!

Kartstig, 2019-06-27:
@HAMH96 

```python
conn.read(f'{ip_address} device {device_id} objectList')
```

where `conn` is your bac0 connection instance

HAMH96, 2019-06-27:
[image]

I Run it and it returns Syntax Error 

the 0xe... address are beacuse the simulator, that are the address that let me work with it

Safrone, 2019-06-27:
watch out for the last line, you have the bracket outside of the `'`

HAMH96, 2019-06-27:
> watch out for the last line, you have the bracket outside of the `'`

You are right, but I put the first bracket inside and also keep getting syntax error

Kartstig, 2019-06-27:
Are you running Python3? A non-f-string solution will be more compatible:
```python
bacnet.read('{ip_address} device {device_id} objectList'.format(ip_address=ip_address, device_id=device_id)
```
FYI: https://www.python.org/dev/peps/pep-0498/

HAMH96, 2019-06-27:
> Are you running Python3? A non-f-string solution will be more compatible:
> 
> ```python
> cmd = '{ip_address} device {device_id} objectList'.format(ip_address=ip_address, device_id=device_id)
> bacnet.read(cmd)
> ```
> 
> FYI: https://www.python.org/dev/peps/pep-0498/

[image]

Yeah! It worked well, I got this, could you please tell me how to access to an specific object? and what (55,1) (57,1) (58,1) (59,1) means at the end?

Kartstig, 2019-06-27:
You are given a list of tuples of `(pointType, pointAddress)`

I'm not entirely sure of the ones on the end. In my experience, those may be proprietary points.

HAMH96, 2019-06-27:
> You are given a list of tuples of `(pointType, pointAddress)`
> 
> I'm not entirely sure of the ones on the end. In my experience, those may be proprietary points.

Well for an index "2" that is AnalogValue as you can see in the image above, can I only get the name and the instance address? What if I want to know the present value of that object? If I have a MultistateInput, how I can read the value of each state?

ChristianTremblay, 2019-06-27:
Number points are proprietary objects. If you know what they are, you will be able to create them but some work is needed. See here for an example : https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/proprietary_objects/jci.py

MultistateInput should be added to the point list and if you grab the properties of the point, you'll see the `units_state` information that contains all the states of the point.

Let say the name of the point is `mypoint`
```
device['mypoint'].properties
```

By default, `device['mypoint']` will give you the state (ex. 'cooling'...and not the integer value... if you need the value.... use `device['mypoint'].enumValue`

Keep in mind that some object are not implemented in BAC0... accumulator, PID, accessZone, etc... They will not be discovered and you will need to access them with `bacnet.read`

HAMH96, 2019-06-27:
> Number points are proprietary objects. If you know what they are, you will be able to create them but some work is needed. See here for an example : [...]

Thanks a lot for the aclaration, but what if I need to access to a property of an object not implemented in BAC0, for example how to obtain the UserName of an AccessUser object?

ChristianTremblay, 2019-06-28:
I would look into bacpypes object to see what are the property names that you need to access. 
https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py#L836

Then you can play with bacnet.read(‘20:000adress AccessUser(?) UserName’)

You’ll have to work a little ;-)

HAMH96, 2019-06-28:
> I would look into bacpypes object to see what are the property names that you need to access.
> https://github.com/JoelBender/bacpypes/blob/master/py34/bacpypes/object.py#L836
> 
> Then you can play with bacnet.read(‘20:000adress AccessUser(?) UserName’)
> 
> You’ll have to work a little ;-)

I got it, also I looked into Read.py in BAC0 to understand well, finally I could got the UserName, and I think that I will be able to obtain all the properties that I will need, thank you so much!

ChristianTremblay, 2019-07-03:
If you ever get to something, please share your code, it could be included in the library and make those points discoverable.

closed 2019-07-03 by ChristianTremblay