ChristianTremblay/BAC0 #145: Multiple errors and warnings with bacnet.devices

mosterme1003, 2019-07-18:
Hello everyone! I try to browse the network to get all devices that are online. This is the simple code:

```
def getDevices(ipLocal):
      bacnet = BAC0.connect(ip=ipLocal)
      devs = bacnet.devices
      return devs
```

Well in my laboratory network there's no problem. I'll get a response of all devices that are online. At work I was testing this in a much bigger BACnet network with like 100 controllers. Now I receive following errors and warnings.

```
...

2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
--- Logging error ---
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 992, in emit
    msg = self.format(record)
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 838, in format
    return fmt.format(record)
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 576, in format
    if self.usesTime():
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 544, in usesTime
    return self._style.usesTime()
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 388, in usesTime
    return self._fmt.find(self.asctime_search) >= 0
RecursionError: maximum recursion depth exceeded in comparison
Call stack:
  File "C:/Users/Both/sciebo/Labor/Python/opine/testScripts/bacnet_test.py", line 198, in <module>
    Application.getDevices(ipLocal)
  File "C:/Users/Both/sciebo/Labor/Python/opine/testScripts/bacnet_test.py", line 19, in getDevices
    devs = bacnet.devices
  File "C:\Program Files\Python36\lib\site-packages\BAC0\scripts\Lite.py", line 171, in devices
    "{} device {} objectName vendorName".format(device[0], device[1])
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\io\Read.py", line 315, in readMultiple
    self._log.warning("APDU Abort Reject Reason : {}".format(reason))
Message: 'APDU Abort Reject Reason : noResponse'
Arguments: ()
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse

...

2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
2019-07-18 14:23:43,076 - WARNING | APDU Abort Reject Reason : noResponse
Fatal Python error: Cannot recover from stack overflow.

Thread 0x0000079c (most recent call first):
  File "C:\Program Files\Python36\lib\asyncore.py", line 144 in poll
  File "C:\Program Files\Python36\lib\asyncore.py", line 207 in loop
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\core.py", line 155 in run
  File "C:\Program Files\Python36\lib\threading.py", line 864 in run
  File "C:\Program Files\Python36\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Program Files\Python36\lib\threading.py", line 884 in _bootstrap

Current thread 0x000007c0 (most recent call first):
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 388 in usesTime
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 544 in usesTime
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 576 in format
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 838 in format
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 992 in emit
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1070 in emit
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 863 in handle
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1514 in callHandlers
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1452 in handle
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1514 in callHandlers

...

```
Now do you get something out of this? I just want to collect the instances and ip addresses of all devices available. I don't know how to catch the errors when there's a problem with the controllers which cause these. I would appreciate some help. Thank you very much!

ChristianTremblay, 2019-07-21:
Try using bacnet.whois() instead of devices. Devices will try to get
information on device like the vendor and name which you don’t need
necessarily.

mosterme1003, 2019-07-23:
bacnet.whois() works perfectly fine in my case. Thank you!

closed 2019-07-23 by mosterme1003