ChristianTremblay/BAC0 #147: Install error using Python 3.7

wayne1203, 2019-07-29:
Hi, 

I tried to run setup.py or using pip install BAC0 in python 3.7 but will got same error "Systax error"
It will be work when I using python 3.6

ChristianTremblay, 2019-07-29:
Any details on the error ? I'm running 3.7 here not problem.

wayne1203, 2019-07-29:
Error message like #45 
But I'm using Ubuntu 18.04 LTS
I tried use develop brunch to run install but same error as well

ChristianTremblay, 2019-07-29:
Absolutely sure it's not Python 2 that get triggered ?

wayne1203, 2019-07-29:
Pretty sure I'm running python 3.7.x
I checked python version (-V) before I run the setup.py

ChristianTremblay, 2019-07-29:
Can you please provide screen shots ?

wayne1203, 2019-07-30:
Sure
[image]

ChristianTremblay, 2019-07-30:
Haaaa ! sudo !
If you “sudo Python -v” you will probably see Python 2 ;-)

Try installing without sudo. In fact, you should never sudo pip. It won’t point to your user install folder but will target the system install.

wayne1203, 2019-07-30:
Oh... Yes, it's show 2.7.15+
I got Permission denied if I not use sudo

[image]

ChristianTremblay, 2019-07-30:
Then you get some choices... you could force python3 with sudo....`sudo python3 setup.py install`
Or grant yourself permission...
Or I think there is a —user flag possible (?) never used it
Or use Virtual Environments. The latter being probably the best solution.

See : https://docs.python-guide.org/dev/virtualenvs/

ChristianTremblay, 2019-07-30:
On Ubuntu I have always used Anaconda... was easier as they provide everything I needed.
(www.anaconda.org)

wayne1203, 2019-07-30:
Thank you for your support.

closed 2019-07-30 by wayne1203