ChristianTremblay/BAC0 #148: Error while reading presentValues

mosterme1003, 2019-07-29:
Hello everyone!
I want to read the present value from some objects for some minutes with a short intervall. For that I use bacnet.read(). I tested this in a loop and every now and then I get following error:

> Ouput:
> ...
>Try 80: 29.19°C
>Try 81: 29.19°C
>Try 82: 29.19°C
>Try 83: 29.2°C

>Traceback (most recent call last):
>  File "C:/Users/.../testScripts/bacnet_test.py", line 28, in getValue
       print('Try %s: %s°C' %(k, round(bacnet.read('10.64.134.162 analogInput 6008 presentValue'), 2)))
>  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\io\Read.py", line 155, in read
       reason = find_reason(apdu)
>  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\io\Read.py", line 578, in 
find_reason
       if apdu.pduType == RejectPDU.pduType:
>AttributeError: 'KeyError' object has no attribute 'pduType'

Could you help me with this issue? Thank you!

ChristianTremblay, 2019-07-29:
Probably a communication error and the packet received cannot be decoed correcltly... so we don't get the reject reason. How many points do you read ? All in one controller or on multiple devices ?

I suspect you are polling a lot, using non-optimal read calls

mosterme1003, 2019-07-29:
At 'worst' I poll the values of two points from the same controller every second. Last time I defined the device, set the history size and got the data from the point's history but there I got the same and some other errors from time to time which is why I switched back to the bacnet.read() way.. I think said errors all had to do with some packet loss or response failures. Sometimes I get these errors after a minute and sometimes there are no issues for hours.

What would you think is the optimal way to poll the values? Could you give me an example?

Can you tell me how to recover from such an error so that I kinda 'reconnect' and get back on track with polling? I would really appreciate a solution! Thank you!

ChristianTremblay, 2019-07-29:
For 2 points... my comment makes no sense.
bacnet.read use single reqd request instead of readMultiple requests used by the device when defined which are more efficient and create less network traffic, leading to less errors. But as I said... for 2 points, this is not that.

But... you are probably pushing the limit with a every second read. What is the thing you need to read every second ? Usually we are working with temperature reading or commands that will have inertia... so there is not often a reason to read as often.

But you may be needing to detect changes to a value. This would be a good reason to read it often... but this would not be the better way. In that case, a CoV registration would be the way to go. Sadly, BAC0 do not implement COV yet... But there are a few examples on bacpypes for that.

IMO, to optimize reading to some points over a long period of time I would : 

- Define the custom point list for the only points I need
- Use this object list to create a `BAC0.device`
- Choose a decent polling delay (depending on the use case)
- Use dev['point'].lastValue to interact with the actual value

I do not use BAC0 myself over very long period of time, so I have not met all "field" network errors...
To protect the app against data loss, I would try saving the device (`device.save()`)and maybe reset the script once in a while using `bacnet.disconnect()`.

Device save needs some love in a way but I've used that in the past. It stores everything locally in a sqlite db and it is then possible to come back later and browse the history (when offline if I'm not getting it wrong... been a while sorry)

Also, you error message looks like it is not at the latest version. The find `reason` function have been reworked A keyError should not raise an exception but just return a string.

mosterme1003, 2019-07-29:
I need to record some step responses to determine the plant's characteristics and I need the data to be detailed. That's why I want to read the values every second. 

COV would probably not reduce the read data because the temperatures change quite fast after the steps so the record interval would still be small.

I will definetly try again to create the bac0.device with the point list and will hopefully recover from reading issues. Also I will update to the latest version.

Thank you for your help so far. I will get back with an update soon.

ChristianTremblay, 2019-07-30:
Humm.... I see... an interesting use case. You will have better results with a device + custom object list I'm sure.

I'm about to release a new version with better support for proprietary objects... I'll wait for your feedback.

Your characterization ? Is the goal PID optimization ?

ChristianTremblay, 2019-07-30:
... and you may want to have a look to the new function `bacnet.time_sync()` I added in the new develop version. Thanks to your uestion on bacpypes :)

mosterme1003, 2019-07-30:
Yes, my goal is to automatically evaluate the controller performance by analysing a setpoint step response and in the next step I wanna optimise the PID controller after analysing the plant's characteristics. Today I'll try it with the custom object list. 
That's great with the time_sync()! Thank you for your support :)

mosterme1003, 2019-07-30:
I'm trying to do it the way you suggested like in this test script but even if I change the polling delay I mainly get values with a 5 sec delay like you can see in the output. Am I doing something wrong?How do I get the value every second?

```
import BAC0
import time

def polling():

    bacnet = BAC0.connect(ip='192.168.0.2')
    object_list= [('analogInput', 5005), ('analogValue', 201)]

    # I tried both ways of changing the poll delay
    device = BAC0.device('192.168.0.3', 99, bacnet, object_list=object_list, poll=1)
    device['flowTemperature'].poll(delay=1)

    device.update_history_size(100)
    device.clear_histories()

    for i in range(30):
        time.sleep(1)

    print(device['flowTemperature'].history)

if __name__ == '__main__':
    polling()
```

First I changed the poll delay when creating the device and got following output for a 30 seconds for-loop:

`device = BAC0.device('192.168.0.3', 99, bacnet, object_list=object_list, poll=1)`
>2019-07-30 11:21:45.738494    0.0
>2019-07-30 11:21:50.850685    0.0
>2019-07-30 11:21:56.896068    0.0
>2019-07-30 11:22:02.934397    0.0
>2019-07-30 11:22:09.012302    0.0
>2019-07-30 11:22:14.037146    0.0
>Name: Device/flowTemperature, dtype: float64

After that I tried to change the delay of the wanted point I got following history output which still has no 1 second interval:

`device['flowTemperature'].poll(delay=1)`
>2019-07-30 11:26:36.178225    0.0
>2019-07-30 11:26:36.240722    0.0
>2019-07-30 11:26:41.407621    0.0
>2019-07-30 11:26:46.540393    0.0
>2019-07-30 11:26:46.602890    0.0
>2019-07-30 11:26:51.760876    0.0
>2019-07-30 11:26:56.854847    0.0
>2019-07-30 11:26:56.867837    0.0
>2019-07-30 11:27:02.912656    0.0
>Name: Device/flowTemperature, dtype: float64

mosterme1003, 2019-07-30:
Now for the last hour I tried to poll one value with the shown method and I received following errors within the loop. Can you get something out of it?

> 2019-07-30 12:02:47,186 - ERROR   | exception: KeyError(None,)
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\app.py", line 372, in indication
    helperFn(apdu)
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\app\ScriptApplication.py", line 78, in do_WhoIsRequest
    BIPSimpleApplication.do_WhoIsRequest(self, apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\service\device.py", line 102, in do_WhoIsRequest
    self.i_am(address=apdu.pduSource)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\service\device.py", line 127, in i_am
    self.request(iAm)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\app.py", line 460, in request
    super(ApplicationIOController, self).request(apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\app.py", line 354, in request
    super(Application, self).request(apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\appservice.py", line 1505, in sap_indication
    self.request(xpdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\appservice.py", line 1326, in sap_indication
    self.request(apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\comm.py", line 282, in request
    self.clientPeer.indication(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\netservice.py", line 292, in indication
    adapter = self.local_adapter or self.adapters[None]
KeyError: None
Fatal Python error: Cannot recover from stack overflow.

>Thread 0x00002f5c (most recent call first):
  File "C:\Program Files\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 58 in process
  File "C:\Program Files\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 44 in run
  File "C:\Program Files\Python36\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Program Files\Python36\lib\threading.py", line 884 in _bootstrap

>Current thread 0x00000b2c (most recent call first):
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 338 in getMessage
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 575 in format
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 838 in format
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 992 in emit
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1070 in emit
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 863 in handle
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1514 in callHandlers
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1452 in handle
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1514 in callHandlers
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1452 in handle
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1514 in callHandlers
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1452 in handle
  File "C:\Program Files\Python36\lib\logging\__init__.py", line 1514 in callHandlers
  ...

>Thread 0x00000810 (most recent call first):
  File "C:/Users/.../testScripts/testPolling.py", line 20 in polling
  File "C:/Users/.../testScripts/testPolling.py", line 26 in <module>

>Process finished with exit code -1073740791 (0xC0000409)

ChristianTremblay, 2019-07-30:
I may have limited polling to 5 seconds...let me check

ChristianTremblay, 2019-07-30:
Yep..here : https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/tasks/Poll.py
You can change that on your side...

I’m struggling with the logger issue making stack overflow... can you please test using the develop branch ? This way you will see if the logging feature keeps bugging.... and you will be able to modify the poll.py file, changing 5 by 1 (if delay < 1, delay =1)

mosterme1003, 2019-07-30:
While I tested the polling using the develop branch I received this exceptions from time to time. Sometimes I can poll values for an hour without issues and the next time these errors occur after a minute of polling and I won't get any new values until I 'reconnect' and start over.

> exception: KeyError(None,)
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\app.py", line 372, in indication
    helperFn(apdu)
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\app\ScriptApplication.py", line 78, in do_WhoIsRequest
    BIPSimpleApplication.do_WhoIsRequest(self, apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\service\device.py", line 102, in do_WhoIsRequest
    self.i_am(address=apdu.pduSource)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\service\device.py", line 127, in i_am
    self.request(iAm)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\app.py", line 460, in request
    super(ApplicationIOController, self).request(apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\app.py", line 354, in request
    super(Application, self).request(apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\appservice.py", line 1505, in sap_indication
    self.request(xpdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\comm.py", line 574, in request
    self.elementService.sap_indication(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\appservice.py", line 1326, in sap_indication
    self.request(apdu)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\comm.py", line 282, in request
    self.clientPeer.indication(*args, **kwargs)
  File "C:\Program Files\Python36\lib\site-packages\bacpypes\netservice.py", line 292, in indication
    adapter = self.local_adapter or self.adapters[None]
KeyError: None
Exception in thread rp_poll:
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\devices\Points.py", line 125, in value
    str(self.properties.address),
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\io\Read.py", line 162, in read
    reason = find_reason(apdu)
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\io\Read.py", line 585, in find_reason
    if apdu.pduType == RejectPDU.pduType:
AttributeError: 'KeyError' object has no attribute 'pduType'

>During handling of the above exception, another exception occurred:

>Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\threading.py", line 916, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 44, in run
    self.process()
  File "C:\Program Files\Python36\lib\site-packages\BAC0\tasks\TaskManager.py", line 50, in process
    self.task()
  File "C:\Program Files\Python36\lib\site-packages\BAC0\tasks\Poll.py", line 48, in task
    self._point.value
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\devices\Points.py", line 130, in value
    raise Exception("Problem reading : {}".format(self.properties.name))
Exception: Problem reading : flowTemperature

mosterme1003, 2019-08-01:
Further questions regarding the recovery when having this issue: Do I simply reconnect by calling `bacnet.disconnect()` and then just settting up `bacnet = BAC0.connect()` again? Do I have to define the device again? And how would I load the data that has been saved by `device.save()`? I hope that you can help me with this problem because it's essential for recording step responses. I really appreciate your help!

ChristianTremblay, 2019-08-02:
Once a `device.save` is done, the data can be retrieved if you disconnect the device.
`device.disconnect()`
Once disconnected, it should connect to the database and you will be able to interact with the points and look for their history.

Accuracy : I never thought about it. I start a polling task which is a thread using a delay... should be ±1 sec of error.... I guess... depending if some network latency occurs... honestly, I don't know.

Regarding error handling, it depends on the level of error.. use auto-save feature and yes, you could disconnect and reconnect after...

ChristianTremblay, 2019-08-02:
@mosterme1003 
I've made improvements to this workflow in the develop branch this morning. 
bacnet.disconnect and device.disconnect should work better and allow you to get what you want.

Please let me know if it is working for you

mosterme1003, 2019-08-03:
It seems that your improvements did it. Thank you very much for your time and work!
I polled with 1 sec delay for the last 8 hours and none of these errors occured again. Could you explain briefly what you improved? 

Another thing: As I understand your `bacnet.time_sync()` method synchronises the device's localDate and localTime with the current system time.. Is it possible for you to include the option of synchronising with a manually selected time and date that I pass to this method?

ChristianTremblay, 2019-08-05:
What I suspect is that I found some network requests that were bypassing the main request threads... it was possible for a request to be passed before its turn arrive... I think it may be that.
I also disabled global broadcast for whois, that can limit network flooding by iam responses...
And a few other bug fixes.

I'll check for the function, should not be hard to allow that.

ChristianTremblay, 2019-08-06:
Check the develop branch. You can pass a bacpypes.basetypes.DateTime object to the time sync function.

ChristianTremblay, 2019-08-06:
I just want to be sure. To poll 1 sec you modified the Poll.py file or you set it differently (or bypassed the init limit...)

Just to be sure because it is not necessarily a good thing to put a really small number in there... in 99% of the cases...

mosterme1003, 2019-08-06:
Thank you! Yes I modified the Poll.py to poll 1 sec. We will see if it is making any trouble. But at least I need a 2 sec delay for my analyses.

ChristianTremblay, 2019-08-06:
In that case, please tell me if this work. I think I will create a fast
polling function. For those special cases.

ChristianTremblay, 2019-08-06:
When you have time, check out the develop branch. I've changed how I deal with polling and if the delay is under 10sec, a new class will be triggered (DeviceFastPoll) ... nothing really different but it could be useful to have those separated from normal polling threads.

Be aware that the delay will not be guaranteed. Every requests are sent to one processor so there is a bottleneck that will slow down the requests.
Without filtering the object list, I'm able to poll between 1-2 seconds a complete controller.

If the controller do not support segmentation, I suggest using polling on 1 point
`dev['point'].poll(delay=1)`

I've changed that to allow delay under 5 seconds.

mosterme1003, 2019-08-20:
Sorry for not responding.. FYI since your last changes I never encountered a logging error again and the polling 1 sec works fine in my case, although the interval will not be exact like you mentioned but it's not a big deal for sure. Thank you for your time. I really appreciate it!

ChristianTremblay, 2019-08-21:
This is great news.
This means that the next version published will be stronger.
Thanks for testing

closed 2019-08-21 by ChristianTremblay