ChristianTremblay/BAC0 #149: Complete mode for custom API?

wayne1203, 2019-08-06:
Hi, I'm using the develop version of BAC0

The Web Service is start normally but will end when task finish, how can I keep the web server running?
Also, can I add route for the Flash Server which is provided in BAC0? In other word, can I develop some APIs using the Flask Server provided in BAC0?

ChristianTremblay, 2019-08-07:
Good question... never thought about that.

I understand that you are working with a script file that get executed and when it's finished, everything close. Right ?
When I'm using the Web Server, I either work on ipython or a Jupyter Notebook and I keep 
```
bacnet.connect()
```
running for all the time I need it.

If you want the servers to keep working, you will need to keep this part On.

wayne1203, 2019-08-08:
Yes, that's my situation.
So using Jupyter is the only way to hold the process, right?

ChristianTremblay, 2019-08-09:
Not necessarily. You can work in the shell or create a script that would run as a service...

But I must say that I’ve created the web interface as a feature that was alive while I was using the script... for the life of the bacnet instance if I may say so. For example, in real time I can add trends to the live trend feature from the shell, or the Notebook. 

One problem, I think, that would make it hard to separate the web site from the bacnet instance is that there is not really something between the web site and the bacnet side. Goal was to be able to play with the devices and have something that would react. Usually, I think people use some kind of database or an API. 

I would probably need some kind of API in the instance so the Web site could connect to the API, instead of having the bacnet instance talking directly to the web site. 

Maybe you have ideas ? I’m open !

wayne1203, 2019-08-09:
I'll try to run the script as a service later because I don't know how to run it as a a service...
Currently, I work around using database to make an API.
I save devices and point data into MySQL and read it via php. So that the API is written in php

I have no idea how to create .route in BAC0 Flask

ChristianTremblay, 2019-08-12:
Interesting. What is the goal of that ? If I may ?

wayne1203, 2019-08-12:
Sure. I would like to make a custom UI for reading those bacnet devices information.
As all I know can do that is create an API call and use ajax to get the data from server.
So that I can use bootstrap or other library to build a dashboard

mbio16, 2019-08-12:
I want to make almost the same thing. My idea is to make scripts one for web app (Flask) and the second with BAC0. When user define action on the web app script will write to file or database the request for the action. The second script (BAC0) will have infinity loop which look into file or db and when there is new request it will do the action in BacNet. Let me know if you have any other idea to solve this problem.

ChristianTremblay, 2019-08-13:
You probably saw that actually, I made the web server in a way that it has access to the bacnet network... 

I also tried to build some helpers that are creating the network stats, etc...

Main goal was to have a simplement way to have access to Live Bokeh plots... the rest was only “playing around” with Flask. 

That said, I’m open to any suggestion that would keep the live trending feature easy... and provide a intermediate layer to a web site. 

An API would probably be something nice... but it is easy to create something bad :-)

I also know that there was an effort to create some kind of BACnet Web service... not sure I want to go down this path...  

An API that would replicate the REPL and allow to interact with the script would be cool...

Ideas ?

closed 2021-02-13 by github-actions[bot]