ChristianTremblay/BAC0 #150: Run full version on a pi?

bbartling, 2019-08-14:
Hi Christian 

Can the full version run on a pi?

For a job we have to make an energy dash board to display building power and fuel use. 

I was hoping to do most calcs for energy on the jace and then use an export table for bac0. 

Then hopefully some basic math in python to compile units of power into energy. IE, kW to kWh..

Could this data then be displayed thru bac0 bokeh server on a pi in a control panel?

Thanks, 
Ben

ChristianTremblay, 2019-08-14:
Yes. Use berryconda to install Pandas and Bokeh... or something like that (miniconda). Those are already compiled for the RPi in those packages.

bbartling, 2019-08-14:
Cool. Is it also possible for BAC0 to log calculated values in Python?

For example if I get electrical kW by polling the JACE export table, and then in Pandas resample previous days kW data to calculate kWh, would I be able to use BAC0 bokeh server to display that data thru a web browser connection to the pi? Or is it only data that is being polled thru BACnet..?

bbartling, 2019-08-14:
Hopefully that makes sense!

ChristianTremblay, 2019-08-15:
Hummm..... i never thought of that but I think it should be possible. Trends are sent to your BACnet instance. You must give a pandas série. Let me check the code.

ChristianTremblay, 2019-08-15:
So... BAC0 wants the format of the thing sent to be a point. With name,units,properties and history. Pretty sure you could define a point (kind of virtual point) that would point to your result with the history. We should make some tests. When we find a good way, let’s build something convenient.

ChristianTremblay, 2019-08-21:
I’ll close this one and open a new one with the trend request

closed 2019-08-21 by ChristianTremblay

bbartling, 2019-09-18:
Hi Christian, which new issue with the trend request was opened after this one was closed?

ChristianTremblay, 2019-09-20:
https://github.com/ChristianTremblay/BAC0/issues/154 ?