ChristianTremblay/BAC0 #151: BAC0 segmentation

bbartling, 2019-08-15:
Hi Christian,

A few months back you made some changes where I could connect to these larger Schneider MNB1000 boards which works great... Looks like this:

[image]

But when I attempt to connect to a Schneider VAV box controller thru an MSTP trunk there is appears to be a looping disconnect error which does the same thing if `, segmentation_supported=False` or not...

[image]

[image]

Just FYI... Any tips help! Thank you!!

bbartling, 2019-08-15:
[image]

ChristianTremblay, 2019-08-15:
If you pull the latest version ? (Develop branch)

bbartling, 2019-08-16:
Hi Christian,

How do I know if I am using develop branch? I have a new laptop in the field today with a new version of Python installed along with BAC0 & dependencies...

Running the code on this PC:

```
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
2019-08-16 10:43:01,616 - ERROR   | Cannot retrieve object list, disconnecting...
Traceback (most recent call last):
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 517, in _buildPointList
    self.custom_object_list
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\mixins\read_mixin.py", line 574, in _discoverPoints
    self.properties.address, self.properties.device_id
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\io\Read.py", line 184, in read
    "APDU Abort Reason : {}".format(reason)
BAC0.core.io.IOExceptions.NoResponseFromController: APDU Abort Reason : bufferOverflow

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/coryo/Documents/VAV/vavTest.py", line 5, in <module>
    controller = BAC0.device('9501:1', 950101, bacnet)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 186, in __init__
    self.new_state(DeviceDisconnected)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 202, in new_state
    self._init_state()
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 675, in _init_state
    self.connect()
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 711, in connect
    self.new_state(RPDeviceConnected)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 202, in new_state
    self._init_state()
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 444, in _init_state
    self._buildPointList()
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\BAC0\core\devices\Device.py", line 522, in _buildPointList
    self._log.error("Cannot retrieve object list, disconnecting...")
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1407, in error
    self._log(ERROR, msg, args, **kwargs)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1514, in _log
    self.handle(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1524, in handle
    self.callHandlers(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1586, in callHandlers
    hdlr.handle(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1524, in handle
    self.callHandlers(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1586, in callHandlers
    hdlr.handle(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1524, in handle
    self.callHandlers(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1586, in callHandlers
    hdlr.handle(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1524, in handle
    self.callHandlers(record)
  File "C:\Users\coryo\AppData\Local\Programs\Python\Python37-32\lib\logging\__init__.py", line 1586, in callHandlers
```

bbartling, 2019-08-16:
```
>>> import BAC0
>>> print(BAC0.version)
19.4.30.2
>>> 
```

For develop branch do I need to git clone the repository and then use Git to check out develop?? Thanks

ChristianTremblay, 2019-08-21:
Yes... FYI I’ll plan updating Pypi very soon

bbartling, 2019-08-21:
Cool, drop me a line when that happens :^)

ChristianTremblay, 2019-09-09:
@bbartling It's there ! Version 19.9.8

bbartling, 2019-09-09:
Sounds good, can the new version be pip installed or should I git clone and
check out develop?

ChristianTremblay, 2019-09-09:
It’s on Pypi. Just pip install (you may have to pip uninstall first if you
played with too many versions...)

bbartling, 2019-09-10:
Hi Christian,

I have the updated version of BAC0 :)

Would you have any ideas for reading points from these Schneider VAV controllers?

In this Gist, I can connect to the MNB-1000 controllers (`test_bench` controller)  just fine but still issues with the VAVs (`test_vav`) on an MSTP trunk...

```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
C:\ProgramData\Anaconda3\lib\site-packages\IPython\core\formatters.py in __call__(self, obj)
    700                 type_pprinters=self.type_printers,
    701                 deferred_pprinters=self.deferred_printers)
--> 702             printer.pretty(obj)
    703             printer.flush()
    704             return stream.getvalue()

C:\ProgramData\Anaconda3\lib\site-packages\IPython\lib\pretty.py in pretty(self, obj)
    383                 if cls in self.type_pprinters:
    384                     # printer registered in self.type_pprinters
--> 385                     return self.type_pprinters[cls](obj, self, cycle)
    386                 else:
    387                     # deferred printer

C:\ProgramData\Anaconda3\lib\site-packages\IPython\lib\pretty.py in inner(obj, p, cycle)
    561                 p.text(',')
    562                 p.breakable()
--> 563             p.pretty(x)
    564         if len(obj) == 1 and type(obj) is tuple:
    565             # Special case for 1-item tuples.

C:\ProgramData\Anaconda3\lib\site-packages\IPython\lib\pretty.py in pretty(self, obj)
    400                         if cls is not object \
    401                                 and callable(cls.__dict__.get('__repr__')):
--> 402                             return _repr_pprint(obj, self, cycle)
    403 
    404             return _default_pprint(obj, self, cycle)

C:\ProgramData\Anaconda3\lib\site-packages\IPython\lib\pretty.py in _repr_pprint(obj, p, cycle)
    695     """A pprint that just redirects to the normal repr function."""
    696     # Find newlines and replace them with p.break_()
--> 697     output = repr(obj)
    698     for idx,output_line in enumerate(output.splitlines()):
    699         if idx:

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Points.py in __repr__(self)
    615         polling = self.properties.device.properties.pollDelay
    616         if polling < 60 and polling > 0:
--> 617             val = float(self.lastValue)
    618         else:
    619             val = float(self.value)

C:\ProgramData\Anaconda3\lib\site-packages\BAC0\core\devices\Points.py in lastValue(self)
    266         """
    267         if _PANDAS:
--> 268             return self.history.dropna().iloc[-1]
    269         else:
    270             return self._history.value[-1]

C:\ProgramData\Anaconda3\lib\site-packages\pandas\core\indexing.py in __getitem__(self, key)
   1498 
   1499             maybe_callable = com.apply_if_callable(key, self.obj)
-> 1500             return self._getitem_axis(maybe_callable, axis=axis)
   1501 
   1502     def _is_scalar_access(self, key):

C:\ProgramData\Anaconda3\lib\site-packages\pandas\core\indexing.py in _getitem_axis(self, key, axis)
   2228 
   2229             # validate the location
-> 2230             self._validate_integer(key, axis)
   2231 
   2232             return self._get_loc(key, axis=axis)

C:\ProgramData\Anaconda3\lib\site-packages\pandas\core\indexing.py in _validate_integer(self, key, axis)
   2137         len_axis = len(self.obj._get_axis(axis))
   2138         if key >= len_axis or key < -len_axis:
-> 2139             raise IndexError("single positional indexer is out-of-bounds")
   2140 
   2141     def _getitem_tuple(self, tup):

IndexError: single positional indexer is out-of-bounds
```

ChristianTremblay, 2019-09-11:
Let's open a new issue...

closed 2019-09-11 by ChristianTremblay