ChristianTremblay/BAC0 #152: Error when passing a specific dateTime to TimeSync function

mosterme1003, 2019-08-20:
Hey! I just tested the TimeSync function from the develop branch and encountered a TypeError when passing a specific dateTime to the TimeSync function. The synchronisation process works perfectly fine.

```
2019-08-20 16:17:26,953 - INFO    | Starting BAC0 version 19.8.06.dev3 (Lite)
2019-08-20 16:17:26,953 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2019-08-20 16:17:26,953 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2019-08-20 16:17:26,953 - INFO    | Using ip : 192.168.0.2
2019-08-20 16:17:26,954 - INFO    | Starting app...
2019-08-20 16:17:26,954 - INFO    | BAC0 started
2019-08-20 16:17:26,954 - INFO    | Registered as Simple BACnet/IP App
Traceback (most recent call last):
  File "D:/Sciebo/Daten/Labor/Python/timesync_bac0/timeSync.py", line 38, in <module>
    timeSynchronisation(ipLocal, ipDevice, idDevice, localTime, localDate)
  File "D:/Sciebo/Daten/Labor/Python/timesync_bac0/timeSync.py", line 19, in timeSynchronisation
    bacnet.time_sync(ipDevice, datetime=_datetime)
  File "C:\Program Files\Python365\lib\site-packages\BAC0\core\functions\TimeSync.py", line 99, in time_sync
    year, month, day, dow = _datetime.date
TypeError: 'Date' object is not iterable
```

Is it right that when I don't pass an ip to TimeSync the time will be synchronised on all devices in the network and when I pass an ip the time will be only synchronised on this specific device like in the following examples?

`bacnet.time_sync(datetime=_datetime)`
`bacnet.time_sync('192.168.1.3', datetime=_datetime)`

ChristianTremblay, 2019-08-21:
Ho ! I see.
Please try using 
```
_datetime = DateTime(date=_date.value, time=_time.value)
```

I’ll update documentation

mosterme1003, 2019-08-21:
Excellent, thank you!

closed 2019-08-21 by mosterme1003