ChristianTremblay/BAC0 #154: Use any Pandas series in Bokeh Live Trend

ChristianTremblay, 2019-08-21:
Cool. Is it also possible for BAC0 to log calculated values in Python?

For example if I get electrical kW by polling the JACE export table, and then in Pandas resample previous days kW data to calculate kWh, would I be able to use BAC0 bokeh server to display that data thru a web browser connection to the pi? Or is it only data that is being polled thru BACnet..?

@bbartling

bbartling, 2019-09-20:
This is another thing I have been day dreaming about... I am setting up a net zero building (VRF Mitsubishi ground loop type system) that requires alot of power monitoring via modbus to a JACE 8000.

If I could create an export table on the JACE for some kW BACnet points, could BAC0 ever poll these values? And it would be cool if I could use the live full version of BAC0 as an energy dashboard to display some line charts for total_load, PV_load, HVAC_load, misc_loads

And then super cool if these kW values in Python could be re-sampled for 15 minute averages to convert to kWh, and then totaled where it could maybe be displayed in a line chart?

Could bokeh do line charts to display building power in kW, and a bar chart to display previous days energy consumption in kWh from resampling kW data?

Else any other ideas for an energy dashboard :) I have played around with creating Dash dashboard from CSV data and drop down menus, etc... Can bokeh do anything like that??

ChristianTremblay, 2019-09-20:
https://panel.pyviz.org

bbartling, 2019-09-20:
Thanks is that something that would work with getting data from BACO?

ChristianTremblay, 2019-09-20:
Keep in mind that all histories in BAC0 are plain Pandas series and Dataframe... Nothing more. Once you have the data... you can do whatever you want with it.

closed 2021-02-13 by github-actions[bot]