ChristianTremblay/BAC0 #155: controller['SaStPr'].history.describe()

bbartling, 2019-08-30:
Hi Christian,

Is it possible to retrieve summary statistics from a pandas series point?

Ive had a connection to an AHU controller for about an hour, but a `controller['SaStPr'].history` only gives me 4 values.

And the `controller['SaStPr'].history.describe()` gives me the summary statistic for the 4 values but its been running for a while on BAC0 default parameters.
```
2019-08-30 14:43:59.167388    0.51
2019-08-30 14:44:47.357993    0.49
2019-08-30 14:47:36.292022    0.51
2019-08-30 15:55:38.994624    0.99
Name: AHU1/SaStPr, dtype: float64
```

It does seem tho that can print new values of the duct pressure sensor in IPython with a `controller['SaStPr']`...

I was curious to see if I could periodically print the summary statistic of a pandas series when monitoring in the field.

Thanks!

ChristianTremblay, 2019-08-31:
Looks like OS going to sleep for an hour...
Or the network being shut as power efficiency setup.

bbartling, 2019-09-03:
Hi Christian,

On a different job I used the same code except on a discharge sensor for an AHU. On the last cell in Jupyter lab, I thru in a:

```
import time

while True:

    print(controller['DaT'].history.describe())
    time.sleep(120)
```
This prints:
```
count     3.000000
mean     62.526663
std       0.061102
min      62.459995
25%      62.499996
50%      62.539997
75%      62.559998
max      62.579998
Name: AHU1/DaT, dtype: float64
count     3.000000
mean     62.526663
std       0.061102
min      62.459995
25%      62.499996
50%      62.539997
75%      62.559998
max      62.579998
Name: AHU1/DaT, dtype: float64
count     3.000000
mean     62.526663
std       0.061102
min      62.459995
25%      62.499996
50%      62.539997
75%      62.559998
max      62.579998
Name: AHU1/DaT, dtype: float64
```

But it still just prints the last logged `controller['DaT'].history`. Behind the scenes does BAC0 keep a pandas series running where the points are logged at a default 10 second intervals? 

Would there be a way to keep a script running and print the summary statistics of a point being logged at the default 10 second parameters? [Heres a link to the gist](https://gist.github.com/bbartling/5a4cb0149eba69250d5becf37fa3c4bf) if it helps. Thanks!

bbartling, 2019-09-03:
Whats interesting is if I use this loop it seems like the values are updated better.

```
for i in range(20):
​
    print(controller['DaT'])
    time.sleep(120)
```
Outputs:
```

AHU1/DaT : 62.18 degreesFahrenheit
AHU1/DaT : 62.13 degreesFahrenheit
AHU1/DaT : 62.09 degreesFahrenheit
AHU1/DaT : 62.06 degreesFahrenheit
AHU1/DaT : 62.03 degreesFahrenheit
AHU1/DaT : 62.03 degreesFahrenheit
AHU1/DaT : 62.01 degreesFahrenheit
AHU1/DaT : 61.97 degreesFahrenheit
AHU1/DaT : 61.93 degreesFahrenheit
AHU1/DaT : 61.87 degreesFahrenheit
AHU1/DaT : 61.85 degreesFahrenheit
AHU1/DaT : 61.75 degreesFahrenheit
AHU1/DaT : 61.71 degreesFahrenheit
AHU1/DaT : 61.67 degreesFahrenheit
AHU1/DaT : 61.58 degreesFahrenheit
AHU1/DaT : 61.54 degreesFahrenheit
AHU1/DaT : 61.50 degreesFahrenheit
AHU1/DaT : 61.40 degreesFahrenheit
AHU1/DaT : 61.37 degreesFahrenheit
AHU1/DaT : 61.34 degreesFahrenheit
```
`controller['DaT'].history.describe()`

```
count    25.000000
mean     61.898797
std       0.314396
min      61.339996
25%      61.669998
50%      61.969997
75%      62.129997
max      62.349998
Name: AHU1/DaT, dtype: float64
```
Printing standard deviation:
`controller['DaT'].history.std()`
`0.31439560778886844`

If the point is not polled, is this what you are referring too as network optimization where the point maybe not updated? Thanks.. Any tips for best practices??

ChristianTremblay, 2019-09-03:
Thanks for the gist... this is clearer.
When defined, this controller fails to support polling. Too much data and it probably don't support segmentation.
First question : 
Do you need all the variables ?

If not, I strongly suggest you define a custom object list when you define the controller. This way you will focus on what's needed.

If you need everything but you need the value to be updated... as polling is not activated on your controller, you can start polling for a specific point.

`controller['DA-T'].poll(60)`

for example. This will assure that the value will be read once every 60 seconds.

To "optimize", when you access the point value, I suggest you use 
`controller['DA-T'].lastValue`

If not, when you issue a read, you will emit a read request on the network (read property) and this may not be optimal.

## Note : 
In the next version, if a controller is using polling and polling is frequent, BAC0 will use the lastValue by default...  

Now that the polling work, you 
`controller['DA-T'].history`

will be up to date, so the `describe()` function.

closed 2019-09-04 by bbartling