ChristianTremblay/BAC0 #156: Install BAC0

clavay, 2019-09-06:
Hello,
I have an issue installing BAC0.
I'm on debian 9.
I'm trying to install BAC0 but I don't use conda, only pip.
I installed bacpypes, flask, flask-bootstrap, bokeh, pandas.
Do I need to use conda to resolve this ?
Here is the error : 

```
sudo pip install BAC0
Collecting BAC0
  Using cached https://files.pythonhosted.org/packages/df/a8/b1af02733b570c6a39f7c08c1f375891581882d03488640845e826aa9b73/BAC0-19.4.30.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-LBxNuK/BAC0/setup.py", line 5, in <module>
        from BAC0 import infos
      File "BAC0/__init__.py", line 16, in <module>
        from . import core
      File "BAC0/core/__init__.py", line 3, in <module>
        from . import app
      File "BAC0/core/app/__init__.py", line 3, in <module>
        from . import ScriptApplication
      File "BAC0/core/app/ScriptApplication.py", line 43
        def __init__(self, *args, bbmdAddress=None, bbmdTTL=0):
                                            ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-LBxNuK/BAC0/

```
Thanks in advance
clavay

clavay, 2019-09-06:
the install works with pip3 but non pip.

ChristianTremblay, 2019-09-06:
I don’t support Python 2 and your system Python is probably 2.7. 

This is why you need to specify pip3. 

When using it, be sure to start python3 ... or you’ll get into troubles with 2.7.

closed 2019-09-09 by ChristianTremblay