ChristianTremblay/BAC0 #160: Index error

ChristianTremblay, 2019-09-11:
https://gist.github.com/bbartling/7866f2aa673c52975d7949c0cc36ca64

ChristianTremblay, 2019-09-11:
@bbartling

ChristianTremblay, 2019-09-11:
Probably related to the fact that when disconnecting bacnet, all device are getting saved... then still active but linked to the database (which have been saved).
You may face some kind of conflict.

Instead of redefining the device, you could try reconnecting it.

```
# Given a device which type is DeviceFromDB
controller.connect(network=bacnet)
```

bbartling, 2019-09-11:
Hi Christian,

I have a feeling that first Gist that you linked I forgot to update BAC0 prior to the test...

After updating BAC0 properly..

More testing on a jobsite, and it seems to work fine with a Schneider VAV controller when I define the points to poll, IE:
```
vav_obj = [('file', 1),
('analogValue', 309)] #DmprPos
```
(seems to work OK in this Gist below)
[https://gist.github.com/bbartling/3647d0579bfcbebbd566e82932883410](https://gist.github.com/bbartling/3647d0579bfcbebbd566e82932883410)

This Gist below without the `vav_obj` defined there is an odd recurring loop/error...
[https://gist.github.com/bbartling/937da8dcdae3ab3f7ab6e086c53fcf5b](https://gist.github.com/bbartling/937da8dcdae3ab3f7ab6e086c53fcf5b)

```
2019-09-11 11:32:37,184 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:32:38,087 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:32:39,648 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:32:40,787 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:32:40,790 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:32:42,107 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:32:43,427 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:32:44,506 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:32:44,509 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:32:45,827 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:32:47,325 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:32:47,506 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:32:47,508 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:32:48,827 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:32:50,386 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:32:50,627 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:32:50,629 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:32:51,885 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:32:53,387 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:32:54,526 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:32:54,529 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:32:55,788 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:32:57,526 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:32:58,667 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:32:58,670 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:32:59,927 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:33:01,309 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:33:02,386 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:33:02,389 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:33:03,825 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:33:04,547 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:33:05,687 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:33:05,690 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:33:06,947 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:33:08,387 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:33:09,526 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:33:09,529 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:33:10,789 - INFO    | Changing device state to RPDeviceConnected'>
2019-09-11 11:33:12,289 - INFO    | Device 950105:[EVAV2W] found... building points list
2019-09-11 11:33:13,366 - ERROR   | Cannot retrieve object list, disconnecting...
2019-09-11 11:33:13,369 - INFO    | Changing device state to DeviceDisconnected'>
2019-09-11 11:33:14,688 - INFO    | Changing device state to RPDeviceConnected'>
```

ChristianTremblay, 2019-09-11:
You need to check if the object can give you the objectList

Try

```
bacnet.read(‘9501:5 device 950105 objectList’)
```

This should give you something. If not, there is a problem with the VAV. May not be configured !?

ChristianTremblay, 2019-09-13:
I could not tell... I saw only bacnet points not their code ;-)

closed 2020-01-29 by ChristianTremblay