ChristianTremblay/BAC0 #164: AttributeError: 'NoneType' object has no attribute 'read'

bbartling, 2019-09-16:
Hi Christian,

I am playing around most likely attempting BAC0 for a purpose not originally created for...

Basically its a program that will attempt to connect to each VAV box in an AHU system and append an empty list with a VAV box damper position and then reset the AHU duct static pressure setpoint to save fan energy. (ASHRAE trim respond sequence for VAV)

The code is sort 'jenky' and this is bit of an experiment that doesn't include an infinite loop to continuously run thru the `i in range()` but it works. Ive seen the code cycle thru the complete `for i in range()` (all VAVs and then the AHU) but every once and while it will throw an exception like this and halt:

```
Exception in thread rp_poll:
Traceback (most recent call last):
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\core\devices\Points.py", line 124, in value
    res = self.properties.device.properties.network.read(
AttributeError: 'NoneType' object has no attribute 'read'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\TaskManager.py", line 59, in run
    self.process()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\TaskManager.py", line 65, in process
    self.task()
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\tasks\Poll.py", line 53, in task
    self._point.value
  File "C:\Users\benb\AppData\Local\Continuum\anaconda3\lib\site-packages\BAC0\core\devices\Points.py", line 134, in value
    raise Exception("Problem reading : {} | {}".format(self.properties.name, e))
Exception: Problem reading : DprPosOut | 'NoneType' object has no attribute 'read'
```

I also notice for each VAV box that it creates like a database file.. a `.hex` & `.bin` in the file directory... is this anything BAC0 could be developed for in the future?? This is definetely a coding experiment for myself as I know this stuff could just be setup on the JACE but Python is more fun...

[Here is a gist for the code I am using](https://gist.github.com/bbartling/aa36b4a3b151c132e1ab83eb6f0d5a02)... :)

Safrone, 2019-09-16:
I have an application where I query specific known points at specific intervals and found that the bacnet network interface works just fine without the overhead of the Device objects.

You're constantly creating and then disconnecting the device objects which causes a whole bunch of extra queries on the network. 

generally you can simplify your tooling with something like this:

```
bacnet = BAC0.connect()

value = bacnet.read("%s analogValue 814 presentValue" % (address))
```

you can then loop that value with sleeps to get more precise timings and avoid some of the added complexity of the device polling

bbartling, 2019-09-17:
👍 thanks Christian. Still learning how the bacnet stack works :^)

closed 2019-09-17 by bbartling

ChristianTremblay, 2019-09-17:
All credits to @safrone for the answer :-)
I would just add : for scripts like this, 

- use `bacnet = BAC0.lite()`
- Never put the definition of the network inside your loop. No gain. Keep the bacnet network active. 
- If working with devices this means you want to work with polling. So in cases where logic would need to be done on points, use a custom object list and use an explicit polling of more than 60sec.
- define your network and devices once at the beginning
- stop/save/disconnect once everything is done
- use `point.lastValue` Anytime possible. This way you don’t generate read property on the network and you rely on the read property multiple from device polling
- if you don’t work with devices and make your own requests, try to use `bacnet.readMultiple` if possible. 
- use `point.write(value,priority=x)` syntax in case where it must. I mean, you are overriding the sequence but do you really want to override the output ? Maybe writing at priority 9 would make sense ?
- try working with tasks from BAC0. It will launch thread so you notebook will still be usable while the function runs. 
- provide subnet mask if you provide the ip for a network
- 5 sec polling is very fast