ChristianTremblay/BAC0 #165: Time synchronisation doesn't work for all devices

mosterme1003, 2019-09-23:
Hey! I've tested the time sync from your develop branch again but now with some devices from other vendors. In the device object the time synchronisation is listed as a supported service but the synchronisation simply doesn't work. It only works for a few devices. Most of them from the same vendor. I'm not receiving any error and I don't know why there is a difference between the devices. Do you have any idea? Do the device have to fulfill any requirements to make this synchronisation service work except that it should be listed as a supported service?
Thank you!

ChristianTremblay, 2019-09-23:
Do you have some wireshark trace that would tell us the response (if there is any) ?

Also, I am aware of some device that don’t like unicast requests... other that don’t like global broadcast...

It can be anything... it could also be some internal config that would turn off network sync. 

Hard to tell.

mosterme1003, 2019-09-25:
I tested the BAC0 timeSync and the timeSync via enteliWeb and got some wireshark trace. The only difference that I can see is that enteliweb sends the utcTimeSynchronisation request in addition to the timeSynchronisation. Do you get something out of it?

[image]
[image]

ChristianTremblay, 2019-09-25:
Interesting. It means that some controllers require a UTC sync which are time zone aware. 

Actually BAC0 do not create a UTC request but it is not super hard to implement I guess. 

If you feel for it you can try to build the request. Using bacpypes and BAC0 (look into the timesync function of BAC0).

ChristianTremblay, 2019-10-07:
@mosterme1003 
You can try the develop branch... there is a new feature : 

`bacnet.time_sync(UTC=True)`

mosterme1003, 2019-10-07:
Sorry for responding so late. I tried it on my own and just rewrote your timeSync function with importing the UTCTimeSynchronisation service from bacpypes and also just tried your UTC function. The wireshark trace showed me that it sent a request just like enteliweb but the time on the device couldn't be changed.. I also tried to sync the time via a bacnet4J java application and it worked without the UTCTimeSynchronisation.

Summary: 

- EnteliWeb sends timeSync and UTCtimeSync requests - time synced

- BAC0 sends the same requests as enteliWeb (even tried your UTC function) - time not synced (only synced time for the devices from one vendor)

- Bacnet4J java application just sends timeSync request like the other methods - time synced

It seems strange to me that all methods leave the same wireshark trace apparently but the BAC0 timeSync just works for one vendor while the others work for other devices too.. Do you have any other idea? Thanks for your help.

ChristianTremblay, 2019-10-07:
Probably the device needs the time sync be sent "direclty" to it.
I've commited a new develop version.
Please try : 
`bacnet.time_sync(UTC=True, destination='10.64.139.227')`

destination can be any address (`3:4` or ip, or you can use `local`or `global`)

mosterme1003, 2019-10-09:
Hey I tried to send it directly like you said but it still doesn't seem to work. I recorded the wireshark trace for the request on a Delta Controls device and a SysCom device. SysCom always seem to work but no other vendors. The traces look exactly the same to me.. Tell me if you need any more information.

Btw I still receive this error when using your time_sync method:
```
Traceback (most recent call last):
  File "C:/.../Python/timesync_bac0/timeSync_bac0.py", line 81, in <module>
    timeSynchronisation(ipLocal, idDevice, localTime, localDate)
  File "C:/.../Python/timesync_bac0/timeSync_bac0.py", line 33, in timeSynchronisation
    bacnet.time_sync(datetime=_datetime, UTC=True, destination=ipDevice)
  File "C:\Program Files\Python36\lib\site-packages\BAC0\core\functions\TimeSync.py", line 128, in time_sync
    year, month, day, dow = _datetime.date
TypeError: 'Date' object is not iterable
```

[image]

ChristianTremblay, 2019-10-09:
Regarding the error, can you show me the line of code you used ? 

Do you have the trace for a working time sync for .124 ?

mosterme1003, 2019-10-10:
Well I thought the .124 was syncable but I couldn't set the datetime with other applications either.. It just says "write access denied". Bad example.. But I already had uploaded wireshark trace further up where it worked with enteliWeb and didn't work with your method. The only difference there was the UTCTimeSync request which apparently wasn't the issue since I don't get the time synced with your UTCTimeSync..

Regarding the error, it is from your time_sync method: (line 3 here)
```
# Unconfirmed request...so wait until complete
iocb.wait()  # Wait for BACnet response
year, month, day, dow = _datetime.date
year = year + 1900
hour, minutes, sec, msec = _datetime.time
d = dt.datetime(year, month, day, hour, minutes, sec, msec)
self._log.info("Time Sync Request sent to network : {}".format(d.isoformat()))
```

ChristianTremblay, 2019-10-19:
Can you give me the code call you made to get this error ?

ChristianTremblay, 2020-06-16:
If still needed, we'll reopen

closed 2020-06-16 by ChristianTremblay