ChristianTremblay/BAC0 #166: BACnet issue

NagarjunaReddy1242, 2019-09-23:
Hi sir,
I was installed BAC0 on my ubuntu. and i was connected bacnet device  as a TCP connection.
i tried 
like below program
import BAC0
bacnet=BAC0.connect(ip='169.254.24.7/24')
 iam getting error
BAC0.core.io.IOExceptions.InitializationError: IP Address provided (169.254.24.7) invalid.

how to solve this error.

ChristianTremblay, 2019-09-24:
By providing a valid IP. Your interface seems to be disconnected.

NagarjunaReddy1242, 2019-09-24:
Sir if i provide a valid IP what can i get the details given by
bacnet.whois() and can you please give the small program to read of our
bacnet register mapping. Please find below data mapping.

1,AV-11, 4264_1_1, Output Apparent Power Rating (System
Information->Ratings)

2,AV-12, 4103_1_1, System Input Nominal Frequency (System
Information->Ratings)

3,AV-13, 4102_1_1, System Input Nominal Voltage (System
Information->Ratings)

4,AV-14, 4261_1_1, System Output Nominal Frequency (System
Information->Ratings)

5,AV-15, 4260_1_1, System Output Nominal Voltage (System
Information->Ratings)

Please sir i need your help in this regard.

ChristianTremblay, 2019-09-29:
Not sure I understand this part in your description  "4260_1_1"
Is your device IP or MSTP
What is the network number ?
What is the device ID of the device ?

ChristianTremblay, 2020-01-29:
If still needed, please reopen

closed 2020-01-29 by ChristianTremblay