ChristianTremblay/BAC0 #169: Problem with segmentation

HAMH96, 2019-10-17:
Hi Christian, I am trying to connect a Fire Alarm Control Panel from Simplex, exactly 4007ES, via BACnet IP, to a raspberry pi model 3B which has the following python 3 code(using NINJA IDE):

import BAC0
bacnet = BAC0.connect(ip='192.168.0.101/24')
controlBacnet = BAC0.device('192.168.0.101/24',32403,bacnet)

When I run that code, there appear a lot of WARNING alerts that say: "Segmentation not supported...will read properties one by one..." Finally the execution crashes

So I try to modify the "connect" line with:
controlBacnet = BAC0.device('192.168.0.101/24', 32403, bacnet, segmentation_supported=False)
and also 
controlBacnet = BAC0.device('192.168.0.101/24', 32403, bacnet, segmentation_supported=True)
in both cases the WARNING  alerts still appearing and the execution crashes again

I was looking for another similar issues like this, so I found you suggested someone to use a custom object list, so I tried it too, but I didn't understand well how it works, I used:
my_obj_list =[(digitalInput,1),(digitalInput,2),(digitalInput,3)]
I suppose that the tuple only need contain the ObjectType and the instance of each object. then there is neither error nor warnings, but I don't know how to access to the value of that specific objects, for example one of the objects it's called "Low Battery", but if I run the line:
print(controlBacnet["Low Battery"]),  an error appears and says that Low Battery doesn't exist in controller. 
I will be grateful by your help with this, thanks for your time.

ChristianTremblay, 2019-10-19:
First thing I see, when you connect BAC0, you are creating a BACnet device so you need to provide the IP address of the computer on which you work...in that case, the Raspberry Pi.

When you define the device, you provide the IP address (or BACnet Address) of the Fire Panel you want to talk to.

Try that and give me feedback on your results, we'll start from there.

HAMH96, 2019-10-21:
Excuse me, there was an error from me, here is the actual code:

import BAC0
bacnet = BAC0.connect(ip='192.168.0.101/24')
controlBacnet = BAC0.device('192.168.0.240/24',32403,bacnet)

ChristianTremblay, 2019-12-16:
I see this is probably related to your other question, hope you'll forgive the time I took to answer.

ChristianTremblay, 2020-01-29:
https://github.com/ChristianTremblay/BAC0/issues/170

closed 2020-01-29 by ChristianTremblay