ChristianTremblay/BAC0 #174: IP address provided is invalid

leo-nerd, 2019-12-11:
Hello. I have a BACnet router at IP address 192.168.92.68/24, and defined my computer's (Windows) **IP address =192.168.92.69** on **subnet mask = 255.255.255.0** with **gateway = 192.168.92.1**. 
I get the 'IP address provided invalid' with these settings when running `BAC0.connect(ip='192.168.92.69/24')` (or `BAC0.connect(ip='192.168.92.68/24')` but from what I've read on the conversations here it's the IP address of the computer sending the BACnet commands that we're entering at this point, not the router).

Maybe worth mentioning:
- My setup: I'm using a Wattstopper LMBC300 to bridge the router to the LMRC-102 controller which is ultimately the piece of equipment I'm interested in controlling
- With the aforementioned IP settings I am able to control the LMRC-102 without any issues from the Wattstopper desktop software.
- I do not get this error when I use `BAC0.connect` with the wifi adapter IP address (which is connected to Internet), or when I do `BAC0.connect()` it goes by default to this address as explained in the BAC0 documentation, but of course this isn't the network I want to be using.
- I do not seem to get the error when I use an automatically-assigned IP using DHCP (169.x.y.z) but in this case I cannot communicate with my devices anymore (I tried changing their IP addresses to be on the same subnet but no luck)

ChristianTremblay, 2019-12-11:
Would it be possible that the IP 192.168.92.69 is already used by the Wattstopper software ? If so, you cannot use the same IP as the socket is already taken.
You can define another IP though : 
[image]

Then you will be able to run 2 BACnet software at the same time, using the same network interface.

leo-nerd, 2019-12-11:
YES! That did it. Thank you! And awesome job on the thorough documentation.

closed 2019-12-11 by leo-nerd