ChristianTremblay/BAC0 #175: How to avoid the "Read property" printout?

leo-nerd, 2019-12-13:
Hello again. 
When calling bacnet.device() (bacnet being created using BAC0.connect()) I get about 3 minutes of console printout of what seems to be all the properties on the controller. Here's an excerpt:
```
...
12:59:59: 
12:59:59: ###################################
12:59:59: # Read property
12:59:59: ###################################
12:59:59: ['2001:87', 'analogValue', '901', 'objectName']
12:59:59: ###################################
12:59:59: REQUEST              <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x277385032b0>
12:59:59: iocb                 <bacpypes.iocb.IOCB(287) instance at 0x277389afe10>
13:00:00: value                datatype            
13:00:00: 'MS/TP MAC'          <class 'bacpypes.primitivedata.CharacterString'>
13:00:00: 
13:00:00: ###################################
13:00:00: # Read property
13:00:00: ###################################
13:00:00: ['2001:87', 'analogValue', '901', 'description']
13:00:00: ###################################
13:00:00: REQUEST              <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x277388d4358>
13:00:00: iocb                 <bacpypes.iocb.IOCB(288) instance at 0x277388c9ef0>
13:00:02: value                datatype            
13:00:02: 'MS/TP MAC Address'  <class 'bacpypes.primitivedata.CharacterString'>

...etc etc
```
This also happens throughout the execution of the script (not only in the initialization), when I write values to the controller. See below the console output right after setting a controller value: 
```
13:03:37: Current DIMMER-2 setting: 0
13:03:37: Setting DIMMER-2 to 100
13:03:37: 
13:03:37: ###################################
13:03:37: # Read property
13:03:37: ###################################
13:03:37: ['2001:87', 'analogOutput', '2', 'presentValue']
13:03:37: ###################################
13:03:37: REQUEST              <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x27738503be0>
13:03:37: iocb                 <bacpypes.iocb.IOCB(466) instance at 0x277386c2d68>
13:03:37: value                datatype            
13:03:37: 0.0                  <class 'bacpypes.primitivedata.Real'>
13:03:37: 
13:03:37: ###################################
13:03:37: # Write property
13:03:37: ###################################
13:03:37: ['2001:87', 'analogOutput', '2', 'presentValue', '100', '-', '8']
13:03:37: ###################################
13:03:37: 
13:03:37: ===================================
13:03:37: Creating Request
13:03:37: ===================================
13:03:37: indx                 priority             datatype             value               
13:03:37: None                 8                    <class 'bacpypes.primitivedata.Real'> '100'               
13:03:37: Encodeable value     <bacpypes.primitivedata.Real object at 0x0000027738A79F60> <class 'bacpypes.primitivedata.Real'>
13:03:37: REQUEST              <bacpypes.apdu.WritePropertyRequest(15) instance at 0x27738a79128>
13:03:37: iocb                 <bacpypes.iocb.IOCB(467) instance at 0x27738a79f60>
13:03:38: 
13:03:38: ###################################
13:03:38: # Read property
13:03:38: ###################################
13:03:38: ['2001:87', 'analogOutput', '2', 'presentValue']
13:03:38: ###################################
13:03:38: REQUEST              <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x277386c2b38>
13:03:38: iocb                 <bacpypes.iocb.IOCB(468) instance at 0x2773889c358>
13:03:39: value                datatype            
13:03:39: 100.0                <class 'bacpypes.primitivedata.Real'>
13:03:39: beginning shutdown
```
This is delaying the script and hinders its critical functions. Is there any way to turn that off? This happens even when using `BAC0.log_level('silence')` or `BAC0.log_level('error')`

I'm attaching here the script as well as the entire console output: 
[bactools.zip](https://github.com/ChristianTremblay/BAC0/files/3962181/bactools.zip)

Thanks again for your help
Léo.

ChristianTremblay, 2019-12-16:
You should not see that except if something activated "debug" logging level for the console AND BAC0 logger level is also set as debug. 

I don't have time to dig down this rabbit hole, but I would suggest something if it's acceptable for you : 

```
# -*- coding: utf-8 -*-
"""
Created on Thu Dec 12 15:09:52 2019

@author: Adam.Jaffe
"""

import BAC0 as bac
import concurrent.futures
import threading
import logging
import time as t

def open_bac_cxn(config):
    ''' Open connection to BAC device '''
    
    ip = config['bac_ip']
    
    bacnet = bac.lite(ip=ip) # You don't need all the extra stuff of connect
    # bacnet.whois() # this is not required
    controller = bac.device('2001:87', 946399, bacnet, poll=0,)# segmentation_support=False)
    
    config['bacnet'], config['controller'] = bacnet, controller
    
    return config

def test_bac_cxn(config, events):
    ''' Test connection to BAC device '''
    
    controller = config['controller']
    log = config['bacnet']._log
    exit_event = events['exit']
    
    log.info('CURRENT DIMMER STATUS')
    log.info('Dimmer-1: %s', controller['DIMMER-1'])  # red light
    log.info('Dimmer-2: %s', controller['DIMMER-2'])  # green light
    
    d1, d2 = 0, 0
    
    while not exit_event.is_set():
        dim = int(input('Select dimmer to edit (1 or 2): '))
        val = int(input('Set dimmer value (0 to 100): '))
        
        if dim == 1:
            log.info('Current DIMMER-1 setting: %s', d1)
            d1 = val
            log.info('Setting DIMMER-1 to %s', val)
            controller['DIMMER-1'] = val            
        elif dim == 2:
            log.info('Current DIMMER-2 setting: %s', d2)
            d2 = val
            log.info('Setting DIMMER-2 to %s', val)
            controller['DIMMER-2'] = val
        
    log.info('test_bac_cxn received EXIT event. Exiting.')
    close_bac_cxn(config)
    
    return

def close_bac_cxn(config):
    
    bacnet, controller, log = config['bacnet'], config['controller'],config['bacnet']._log
    
    # Set both to zero
    log.info('Setting both dimmers to off.')
    controller['DIMMER-1'], controller['DIMMER-2'] = 0, 0
    log.info('Both dimmers off.')
    
    bacnet.disconnect()
    
    return
    

if __name__ == "__main__":
    
#    
    config = dict()
    config['bac_ip'] = str(input('Enter IP address for BAC device: '))
    log = config['bacnet']._log
    config = open_bac_cxn(config)
    
    events = dict()
    events['exit'] = threading.Event()
    
    with concurrent.futures.ThreadPoolExecutor(max_workers=1) as executor:
        executor.submit(test_bac_cxn, config, events)

        t.sleep(20)
        
        log.info('beginning shutdown')
        events['exit'].set()
```

Just using what is already there
[image]

adammjaffe, 2019-12-17:
Hi, thanks. Your suggestions works on the standalone snippet but not in the full code set.

In the full code set, we begin the main function with a formatting statement setting logging level to INFO:

`fmt = "%(asctime)s: %(message)s"
logging.basicConfig(format=fmt, level=logging.INFO, datefmt="%H:%M:%S")`

We then open a series of threads to interface with different pieces of hardware, including one thread for the bacnet device. All logging done in the various threads is done through logging.info.

`with concurrent.futures.ThreadPoolExecutor(max_workers=6) as executor:
        executor.submit(check_timer, config, events)  # timer to check accels 
        executor.submit(light_timer, config, events)  # timer to turn off light after cooldown
        if config['device'] != 'none':
            executor.submit(control_light, config, events)
        executor.submit(watch_stream, config, events)
        executor.submit(to_aws, config, events)
        executor.submit(call_cxcapture, config, events)`

It's unclear at what point the logging level gets set to DEBUG. Could this happen automatically somehow because of the threaded structure? I'm doubtful, but still unable to resolve...

ChristianTremblay, 2019-12-18:
I'm pretty sure the basicConfig plays with a global root logger that can affect all loggers underneath.

Also, in the doc, care should be taken when using this function with threads (which you are clearly doing)...
I would create a new logger, a format as you want it and add the handlers myself...

ref : 
https://docs.python.org/3/library/logging.html#logging.basicConfig
https://docs.python.org/3/library/logging.html#logging.log

closed 2020-01-29 by ChristianTremblay