ChristianTremblay/BAC0 #176: BAC0 fails using DNS names

palisadoes, 2019-12-17:
If you use a resolvable DNS name versus an IP address you get this message.

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/BAC0/core/io/Read.py", line 118, in read
    args_split, arr_index=arr_index, vendor_id=vendor_id, bacoid=bacoid
  File "/usr/local/lib/python3.6/dist-packages/BAC0/core/io/Read.py", line 399, in build_rp_request
    request.pduDestination = Address(addr)
  File "/usr/local/lib/python3.6/dist-packages/bacpypes/pdu.py", line 52, in __init__
    self.decode_address(args[0])
  File "/usr/local/lib/python3.6/dist-packages/bacpypes/pdu.py", line 278, in decode_address
    raise ValueError("unrecognized format")
```

palisadoes, 2019-12-17:
Adding code like this should help fix it.

```
import ipaddress
import socket

def get_ip_address(device):
    """Get IP address for a device.

    Args:
        device: Device for which address is required

    Returns:
        result: IP address

    """
    # Initialize key variables
    result = None

    # Check if IP address
    try:
        _ = ipaddress.ip_address(device)
        result = device
    except:
        # Not and IP address
        try:
            result = socket.gethostbyname(device)
        except:
            result = None

    # Return
    return result
```

ChristianTremblay, 2019-12-18:
Do you often use DNS locally to identify IP devices on your network ?
Or are you accessing remote devices through port 47808 over the web ?

I'm not against including support for dns_names in BAC0... but also want to promote that over the web can lead to significant security risks as everything is open.

palisadoes, 2019-12-18:
For use internally within the confines of our company.

ChristianTremblay, 2019-12-19:
Excellent answer :)
Do you want to make a PR ?

closed 2021-02-13 by github-actions[bot]