ChristianTremblay/BAC0 #177: A strange problem when adding a device

sedlacekdavid, 2019-12-21:
Hi Guys, mainly - thanks for the great software!

My issue - I have a bacnet device on a IP/UDP network and I can successfully discover it like:

`>>> bacnet.discover()
2019-12-21 22:02:31,115 - INFO    | 192.168.100.2 network number is 0
2019-12-21 22:02:33,052 - INFO    | 192.168.100.2 router to [0]
2019-12-21 22:02:35,044 - INFO    | Addr : 192.168.100.2
2019-12-21 22:02:37,045 - INFO    | Found those networks : {0}
2019-12-21 22:02:37,045 - INFO    | Discovering network 0
2019-12-21 22:02:37,047 - DEBUG   | do_whois ['0:*', '0', '4194303']
2019-12-21 22:02:37,050 - DEBUG   |   - request: <bacpypes.apdu.WhoIsRequest(8) instance at 0x880b6f0>
2019-12-21 22:02:38,328 - DEBUG   | do_IAmRequest <bacpypes.apdu.IAmRequest(0) instance at 0x1405410>
[('192.168.100.2', 123)]`

The trouble is that when I add it, it is responding to BAC0's ReadProperty Device Name, but from some reason BAC0 just can't see that response. Using Wireshark I can see that there is a response to the request, in the very same way as for .discover() function.

`>>> fx = BAC0.device('0:2', 123, bacnet)
2019-12-21 22:06:59,581 - DEBUG   | BAC0.core.devices.Device | Device | BAC0.core.devices.Device | Device | Controller initialized
2019-12-21 22:06:59,582 - INFO    | Changing device state to DeviceDisconnected'>
2019-12-21 22:06:59,583 - DEBUG   |
2019-12-21 22:06:59,586 - DEBUG   | ###################################
2019-12-21 22:06:59,586 - DEBUG   | # Read property
2019-12-21 22:06:59,587 - DEBUG   | ###################################
2019-12-21 22:06:59,588 - DEBUG   | ['0:2', 'device', '123', 'objectName']
2019-12-21 22:06:59,589 - DEBUG   | ###################################
2019-12-21 22:06:59,593 - DEBUG   | REQUEST              <bacpypes.apdu.ReadPropertyRequest(12) instance at 0x1405370>
2019-12-21 22:06:59,596 - DEBUG   | iocb                 <bacpypes.iocb.IOCB(82) instance at 0x01405510>
2019-12-21 22:07:09,601 - WARNING | Offline: provide database name to load stored data.
2019-12-21 22:07:09,602 - WARNING | Ex. controller.connect(db = 'backup')`

`No.	Time	Source	Destination	Protocol	Length	Info
30277	1332.199620	192.168.100.1	192.168.100.2	BACnet-APDU	54	Confirmed-REQ   readProperty[ 26] device,123 object-name 
30292	1332.407882	192.168.100.2	192.168.100.1	BACnet-APDU	59	Complex-ACK     readProperty[ 26] device,123 object-name 
`

Is there anything I'm missing or I should check? Otherwise I don't have any issues communicating with the device. It seems like BAC0 doesn't see data coming back from the device, although I can see there is a correct response.

Thanks & regards, David.

ChristianTremblay, 2019-12-21:
Hi David,

Your device is IP... so provide the IP for the address. And as both BAC0 and the device are on the same BACnet network, no need to specify the network number (0). 

`fx = BAC0.device(192.168.100.2, 123, bacnet)`

The syntax `0:2` Serves for MSTP network (network number : MAC address of the device on the MSTP Bus)

sedlacekdavid, 2019-12-21:
Hi Christian, thank you for the very quick response. You are right, it's worked! Wonderful.

Just because of my curiosity, you mentioned MSTP support. Recently, I needed to work with a MS/TP bus on a PC. I had to create my own BACNET/IP/UDP to MS/TP RS485-based network convertor using an Arduino & an appropriate Python script, because I found almost impossible to directly communicate on a MS/TP bus with a PC. it's basically an MS/TP to UDP/IP convertor or a gateway if you want. I made it because it's very difficult to keep MS/TP timings etc on a pc.  So now I can use your BAC0 and it obviously works 100 %, which is excellent.

But back to your MSTP support, I saw it in your documentation, but didn't found any explanation?

BTW keep up the good work : )

closed 2019-12-22 by sedlacekdavid