ChristianTremblay/BAC0 #178: Simulate Multiple BACNET devices on the same PC

jayantpancholi, 2019-12-27:
Hi Christian,

I am currently working on Bacnet simulator using YABE and your code.
my requirement is to connect multiple devices instead of only single device under same network.

this is my BACpypes.ini file :-

[BACpypes]
objectName: Betelgeuse
address: 10.16.33.135
objectIdentifier: 123
maxApduLengthAccepted: 1024
segmentationSupported: segmentedBoth
vendorIdentifier: 15
foreignBBMD: 255.255.0.0
foreignTTL: 100
I am able to connect to this device using YABE, I want to create one more device ...how Can I do that.

Also, this one I am connecting using my Wifi IP and not the ethernet one. I want to connect using Ethernet one so that I can create multiple ethernet IPs and all can avt as a individual Bacnet devices.

Please let me know how can i solve this problem.

Looking forward to hearing from you.

ChristianTremblay, 2019-12-30:
To declare multiple instances, you will need multiple IP Addresses. See : https://github.com/ChristianTremblay/BAC0/issues/174

I showed how to do that in Windows.

ChristianTremblay, 2020-01-15:
Did you make it work ?

closed 2020-06-16 by ChristianTremblay