ChristianTremblay/BAC0 #185: read_mixin binary object with present activeText property returns wrong units_state

imrePurret, 2020-01-28:
In read_mixin.py file _process_nex_objects method includes small bug where active state is valued with property inactiveText instead of property activeText. Occurs only in case of binary object with present propertys inactiveText and activeText.

[Link to specific line of code](https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/core/devices/mixins/read_mixin.py#L332)

Current code:
```
_inactive = point_infos[_find_propid_index("inactiveText")]
_active = point_infos[_find_propid_index("inactiveText")]
point_units_state = (_inactive, _active)
```

Should be replaced with:
```
_inactive = point_infos[_find_propid_index("inactiveText")]
_active = point_infos[_find_propid_index("activeText")]
point_units_state = (_inactive, _active)
```

ChristianTremblay, 2020-01-29:
Good catch ! I reviewed the PR
https://github.com/ChristianTremblay/BAC0/pull/186

ChristianTremblay, 2020-01-29:
Merged

closed 2020-01-29 by ChristianTremblay