ChristianTremblay/BAC0 #187: IP Address provided (192.168.3.12) invalid.

hvacking, 2020-02-10:
hi , it is excited to find this bacnet tool. I installed a simulator . But when i run the codes you published. 
I meet an IP issue. 

Could you tell me how to solve this ?The followings are error information. Thanks . 

import BAC0
import pandas as pd
%matplotlib inline
# Define network and connect to it
bacnet = BAC0.connect()

2020-02-10 21:14:17,851 - INFO    | Starting BAC0 version 19.9.9 (Lite)
2020-02-10 21:14:17,855 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2020-02-10 21:14:17,856 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2020-02-10 21:14:17,936 - INFO    | Using ip : 192.168.3.12
---------------------------------------------------------------------------
InitializationError                       Traceback (most recent call last)
<ipython-input-2-5648ee4ad7c9> in <module>
      3 get_ipython().run_line_magic('matplotlib', 'inline')
      4 # Define network and connect to it
----> 5 bacnet = BAC0.connect()

~\Anaconda3\lib\site-packages\BAC0\scripts\Lite.py in __init__(self, ip, port, mask, bbmdAddress, bbmdTTL, **params)
    106             bbmdAddress=bbmdAddress,
    107             bbmdTTL=bbmdTTL,
--> 108             **params
    109         )
    110 

~\Anaconda3\lib\site-packages\BAC0\scripts\Base.py in __init__(self, localIPAddr, localObjName, deviceId, firmwareRevision, maxAPDULengthAccepted, maxSegmentsAccepted, segmentationSupported, bbmdAddress, bbmdTTL, modelName, vendorId, vendorName)
    112         else:
    113             raise InitializationError(
--> 114                 "IP Address provided ({}) invalid.".format(localIPAddr)
    115             )
    116 

InitializationError: IP Address provided (192.168.3.12) invalid.

ChristianTremblay, 2020-02-10:
Are you running another BACnet software on the same network interface ?
If so, the socket is already taken.

hvacking, 2020-02-11:
Hi,Christian 

Thanks for your reply. I restarted computer, and this issue is solved.
I opened a bacnet simulator ,but why the FX is null ?

thanks

LL

hvacking, 2020-02-11:
i put simulator at another computer in LAN,  i use bacnet scan ,and can find this simulator. But when i use codes ,it has issue:

192.168.43.16/47808
import BAC0
import pandas as pd
%matplotlib inline
# Define network and connect to it
bacnet = BAC0.connect(ip="192.168.43.16/47808")
2020-02-11 20:27:15,920 - INFO    | Starting BAC0 version 19.9.9 (Lite)
2020-02-11 20:27:15,924 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2020-02-11 20:27:15,926 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-1-f827ca21c16c> in <module>
      3 get_ipython().run_line_magic('matplotlib', 'inline')
      4 # Define network and connect to it
----> 5 bacnet = BAC0.connect(ip="192.168.43.16/47808")

~\Anaconda3\lib\site-packages\BAC0\scripts\Lite.py in __init__(self, ip, port, mask, bbmdAddress, bbmdTTL, **params)
     99             if not port:
    100                 port = 47808
--> 101             ip_addr = Address("{}/{}:{}".format(ip, mask, port))
    102         self._log.info("Using ip : {ip_addr}".format(ip_addr=ip_addr))
    103         Base.__init__(

~\Anaconda3\lib\site-packages\bacpypes\pdu.py in __init__(self, *args)
     69 
     70         if len(args) == 1:
---> 71             self.decode_address(args[0])
     72         elif len(args) == 2:
     73             self.decode_address(args[1])

~\Anaconda3\lib\site-packages\bacpypes\pdu.py in decode_address(self, addr)
    190                     addrstr = socket.inet_aton(local_ip_addr)
    191                     self.addrIP = struct.unpack('!L', addrstr)[0]
--> 192                     self.addrMask = (_long_mask << (32 - int(local_ip_net))) & _long_mask
    193                     self.addrHost = (self.addrIP & ~self.addrMask)
    194                     self.addrSubnet = (self.addrIP & self.addrMask)

ChristianTremblay, 2020-02-11:
Problem is here : 
```
bacnet = BAC0.connect(ip="192.168.43.16/47808")
```
The part after the '/' must be the network mask. Not the port. So typically something like : 
```
bacnet = BAC0.connect(ip="192.168.43.16/24")
```

hvacking, 2020-02-11:
Hi，Christian 

Thanks for your patience. when i connect the simulator device. System show the error of wrong unknow 
object.
But when i use scanner , the device id is 1, why show the wrong object ? 

Thanks

LL

ChristianTremblay, 2020-02-11:
You will need to provide more detail. I don't know what you are talking about...

hvacking, 2020-02-12:
Hi,Christian 

Let me explain. I run the codes :
*************
import BAC0
import pandas as pd
%matplotlib inline
# Define network and connect to it
bacnet = BAC0.connect(ip="192.168.43.47/24")

**************
it works well, but when i run the next codes, it occurs some issues,system show "UnknownObjectError: Unknown object 192.168.43.16/24 device 1 objectName" error. But  i use bacnet scanner , scanner show that the devcie NO is "1" . So i do not know which cause this error.
*******************
fx = BAC0.device('192.168.43.16/24',1,bacnet)
*******************

Thanks 

LL

ChristianTremblay, 2020-02-13:
That tells us that the device doesn't provide an objectName.... which is mandatory for a BACnet device. 

You could try : 
```
bacnet.read('192.168.43.16 device 1 objectName')
```
and you will probably get the same error.

```
bacnet.read('192.168.43.16 device 1 objectList')
```
will give you the list of objects but you are probably using some "not well implemented/incomplete" device.

hvacking, 2020-02-18:
Hi，Christian 

I  have not fixed this issue yet, i can not get the objects yet. :(
I tried to set some details in bacnet device simulator ,but still show unknow object name.

Regards

LL

Closed #187.

ChristianTremblay, 2020-02-18:
Let's open another issue then.
I will need info about this "simulator"...  because it seems at the root of the problem

hvacking, 2020-02-18:
Thanks, i used the ‘Bacnet Device Simulator“ ， i downloaded from website
https://www.rocketdownload.com/program/bacnet-device-simulator-424960.html

You can download it , i tried to send you the simulator ,butyour mail replies that you mail rejects the attachment for its size.
The size of this simulator is about 6 M.

Regards

LL

ChristianTremblay, 2020-02-18:
Issshhh. No thanks.
First you should not trust web sites like that.
Second, http://www.scadaengine.com/software6.html seems more leggit but it's not a free tool.

If you really want to simulate bacnet points, at no cost, please have a look here : 
https://github.com/JoelBender/bacpypes/blob/master/samples/OpenWeatherServer.py
or
https://github.com/JoelBender/bacpypes/blob/master/samples/IP2VLANRouter.py

hvacking, 2020-04-19:
Dear Christian&nbsp;

BAC0 is an easy-used tool, i desire to use it in my project. But i met a problem in finding the controllers. Could you help to see where i made mistake ?
&gt;&gt;&gt; import pandas as pd
&gt;&gt;&gt; import BAC0
&gt;&gt;&gt; bacnet= BAC0.connect()
2020-04-19 21:16:03,288 - INFO&nbsp; &nbsp; | Starting BAC0 version 20.02.20 (Lite)
2020-04-19 21:16:03,304 - INFO&nbsp; &nbsp; | Use BAC0.log_level to adjust verbosity of the app.
2020-04-19 21:16:03,320 - INFO&nbsp; &nbsp; | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2020-04-19 21:16:05,597 - INFO&nbsp; &nbsp; | Using ip : 192.168.8.100
2020-04-19 21:16:05,613 - INFO&nbsp; &nbsp; | Starting app...
2020-04-19 21:16:05,628 - INFO&nbsp; &nbsp; | BAC0 started
2020-04-19 21:16:05,628 - INFO&nbsp; &nbsp; | Registered as Simple BACnet/IP App
&gt;&gt;&gt; bacnet.devices
Traceback (most recent call last):
&nbsp; File "<pyshell#3&gt;", line 1, in <module&gt;
&nbsp; &nbsp; bacnet.devices
&nbsp; File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36-32\lib\site-packages\BAC0\scripts\Lite.py", line 345, in devices
&nbsp; &nbsp; for device in list(self.discoveredDevices):
TypeError: 'NoneType' object is not iterable

We installed BAC0 in the same netwwork with Johnson control bacnet BMS , i can see the controller with bacnet scan tool, but why i got 'TypeError'&nbsp;
when use bacnet.devices?

Best Regards

Liu Lan

ChristianTremblay, 2020-04-22:
You need to first call `bacnet.discover()`

hvacking, 2020-04-23:
yes, I tried this command, but got [ ].
I also tried use bacnet scan tool, it can find  devices. So i did not why BACO can not find devices.

LL

ChristianTremblay, 2020-04-23:
What if you 
```
bacnet.discover(global_broadcast=True)
```

ChristianTremblay, 2020-06-16:
Recently, develop branch have been modified to better handle simple networks. Can you try and tell if it improves your case ?

ChristianTremblay, 2020-08-05:
Reopen if needed.

closed 2020-08-05 by ChristianTremblay

dheerajdevalekar, 2022-11-16:
Hello,
I used YABE(Yet Another Bacnet Explorer) as a BacNet Simulator. I used `devices = bacnet.discover(global_broadcast=True)` this command to find a connected Devices. I found the Bacnet Devices also. But I need to connect one of them, so how can I do?
Also I need to fetch the present objects in that device...
Can you help in this???