ChristianTremblay/BAC0 #188: log level doesn't seem to be respected in Lite

gsainsbury86, 2020-02-20:
I am using BAC0 within a Flask application. I need to read outputs from bacnet and access them via http.

I have the line ` BAC0.log_level(stdout='critical', stderr='error') ` in my application but when I start the flask app, I see the following:

```
2020-02-20 13:47:21,730 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,730 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,731 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,732 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,733 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,733 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,733 - WARNING | Changed log level of console stderr to ERROR
2020-02-20 13:47:21,733 - WARNING | Changed log level of console stdout to CRITICAL
2020-02-20 13:47:21,733 - WARNING | Changed log level of console stderr to ERROR
```

I'm not sure why it prints 20 times, and then when I actually use bacnet.read commands, they all print full debug to stderr.

ChristianTremblay, 2020-02-20:
Yeah .... not super proud of that part :-/

Do you need the logs in your app ? 
If not (if you are logging from somewhere else) you can turn it off using `BAC0.log_level('silent')`
I know some users had troubles because they were using basic config from logging and this was reconfiguring everything... not sure I remember all the details.

I think remember having some weird results with BAC0's flask implementation.
https://github.com/ChristianTremblay/BAC0/blob/master/BAC0/web/FlaskServer.py#L50

Don't hesitate to make a PR if you have suggestions to improve BAC0's logging.

gsainsbury86, 2020-02-24:
> Do you need the logs in your app ?

Not really, no.  I've tried `BAC0.log_level('silent')` but get the following error: 

```
    "Wrong log level use one of the following : {}".format(_valid_levels)
ValueError: Wrong log level use one of the following : [10, 20, 30, 40, 50]
```

> I think remember having some weird results with BAC0's flask implementation.

Yes, I wondered if Flask and BAC0's logger implementations were stepping on each other's toes.

ChristianTremblay, 2020-02-24:
BAC0.log_level(‘silence’)

gsainsbury86, 2020-02-24:
Thanks for your help. Still no luck unfortunately.

Here's a slightly simplified version of what I'm trying to do

```
@app.route('/tick/bms_alerts')
def get_bms_alert_tick():

    my_ip = config.get('General', 'LOCALHOST_IP')

    bbmdIP = config.get('General', 'BBMD_IP')
    bbmdTTL = 900

    bacnet = BAC0.lite(ip=my_ip, bbmdAddress=bbmdIP, bbmdTTL=bbmdTTL)
    BAC0.log_level('silence')

    try:

        bacnet.discover(networks=['50'])

        data_flat = {}

        for device in bacnet.devices:
            object_list = bacnet.read("{} device {} objectList".format(device[2], device[3]))
            for obj in [o for o in object_list if o[0] == 'binaryValue']:
                    name = bacnet.read("{devID} {dt} {i} objectName".format(devID=device[2], dt=obj[0], i=obj[1]))
                    value = bacnet.read("{devID} {dt} {i} presentValue".format(devID=device[2], dt=obj[0], i=obj[1]))
                    data_flat[name] = value
    except:
        pass
    finally:
        bacnet.disconnect()
    return jsonify(data_flat)
```

And here's a sample from my httpd log. It looks like the log levels are changed to CRITICAL and then immediately I am receiving INFO and DEBUG messages.

```
[Mon Feb 24 13:13:38.414362 2020] [core:notice] [pid 27787] AH00094: Command line: '/usr/sbin/httpd -D FOREGROUND'
[Mon Feb 24 13:13:43.778909 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,778 - WARNING | Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.778959 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.778999 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,778 - WARNING | Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779009 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779040 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779049 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779075 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779083 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779110 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779119 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779144 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779153 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779177 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779186 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779211 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779219 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779245 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779254 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779285 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779294 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779319 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779327 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779350 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779358 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779383 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779392 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stderr to CRITICAL
[Mon Feb 24 13:13:43.779417 2020] [wsgi:error] [pid 27791] 2020-02-24 13:13:43,779 - WARNING | Changed log level of console stdout to CRITICAL
[Mon Feb 24 13:13:43.779425 2020] [wsgi:error] [pid 27791] WARNING:BAC0_Root.BAC0.scripts.Base.Base:Changed log level of console stdout to CRITICAL
[... cut ...]
[Mon Feb 24 13:13:43.785382 2020] [wsgi:error] [pid 27791] [client x.x.x.x:p] DEBUG:BAC0_Root.BAC0.scripts.Lite.Lite:Running
[Mon Feb 24 13:13:43.785402 2020] [wsgi:error] [pid 27791] [client x.x.x.x:p] DEBUG:BAC0_Root.BAC0.scripts.Lite.Lite:finally
[Mon Feb 24 13:13:43.785586 2020] [wsgi:error] [pid 27791] [client x.x.x.x:p] DEBUG:BAC0_Root.BAC0.scripts.Lite.Lite:do_iam
[Mon Feb 24 13:13:43.785669 2020] [wsgi:error] [pid 27791] [client x.x.x.x:p] DEBUG:BAC0_Root.BAC0.scripts.Lite.Lite:  - request: <bacpypes.apdu.IAmRequest(0) instance at 0x7f1a2ac45d30>
```

gsainsbury86, 2020-02-24:
I updated BAC0 (from 19.9.9 to 20.02.20) and also realised I had `logging.basicConfig(stream=sys.stderr)` in my WSGI file.

The problem seems to be resolve now. Thanks again.

closed 2020-02-24 by gsainsbury86

ChristianTremblay, 2020-02-24:
Thanks for letting me know ! It is not the first time I hear someone fighting with `logging.basicConfig` when something else already tried to configure logging.