ChristianTremblay/BAC0 #190: problem reading device propoerties

cbonbled, 2020-03-13:
Hello Christian,
I am novice wit BACnet and wanted to make a quick test. I set one PC with address 192.168.1.41 and started a BACnet set a BACnet simulator to simulate devices. with BACeye and   Yabe I checked the server is correctly running.  My problem is reading the properties of the simulated units. 
I want to read the present value with syntax like bacnet.read('20:0 analogInput 0 presentValue'). But the object is not recognized:
This is the feedback I am getting:
020-03-13 15:56:22,528 - INFO    | Starting BAC0 version 20.02.20 (Lite)
2020-03-13 15:56:22,528 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2020-03-13 15:56:22,528 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2020-03-13 15:56:22,555 - INFO    | Using ip : 192.168.1.58
2020-03-13 15:56:22,559 - INFO    | Starting app...
2020-03-13 15:56:22,560 - INFO    | BAC0 started
2020-03-13 15:56:22,560 - INFO    | Registered as Simple BACnet/IP App
2020-03-13 15:56:26,567 - INFO    | Found those networks : set()
2020-03-13 15:56:26,567 - INFO    | Discovering network 20
2020-03-13 15:56:26,821 - INFO    | 192.168.1.41 router to [20]
discover [('20:0x010000000000', 1), ('20:0x020000000000', 2), ('192.168.1.41', 4194302)]
whois1:  [('20:0x010000000000', 1), ('20:0x020000000000', 2), ('192.168.1.41', 4194302)]
Devices:  [('DEV 1', 'SCADA Engine', '20:0x010000000000', 1), ('New Virtual Device', 'SCADA Engine', '20:0x020000000000', 2), ('CBMS - BACnet Device Simulator', 'SCADA Engine', '192.168.1.41', 4194302)]
Read object1: SCADA Engine
Read object2: ('device', 4194302)
Read result0 CBMS - BACnet Device Simulator
2020-03-13 15:56:33,324 - WARNING | Unknown object 20:0 analogInput 0 presentValue
Read result1 failed
2020-03-13 15:56:33,369 - WARNING | Unknown object 20:2 analogInput 0 presentValue
Read result2 failed
What I am doing wrong?

cbonbled, 2020-03-14:
I finally found. The correct syntax to read the value is 
bacnet.read('20:0x010000000000 analogInput 0 presentValue' )

ChristianTremblay, 2020-03-16:
Glad you found !

closed 2020-03-16 by ChristianTremblay