ChristianTremblay/BAC0 #191: 'dict' object has no attribute 'dtypes'

pzg250, 2020-03-18:
Hi ,
When I was trying to disconnect the device, I met an error. Anyone can help ? Thanks.

```
>>> ddc = BAC0.device("172.17.0.15", 4000, bacnet, poll=5)

>>> ddc.disconnect()
2020-03-18 15:06:57,928 - INFO    | Wait while stopping polling
2020-03-18 15:06:58,134 - INFO    | Polling stopped
Traceback (most recent call last):
  File "/home/major/envs/bacnet/lib/python3.6/site-packages/BAC0/sql/sql.py", line 82, in backup_histories_df
    if point.history.dtypes == object:
AttributeError: 'dict' object has no attribute 'dtypes'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/major/envs/bacnet/lib/python3.6/site-packages/BAC0/core/devices/Device.py", line 456, in disconnect
    self.save()
  File "/home/major/envs/bacnet/lib/python3.6/site-packages/BAC0/sql/sql.py", line 130, in save
    df_to_backup = self.backup_histories_df()
  File "/home/major/envs/bacnet/lib/python3.6/site-packages/BAC0/sql/sql.py", line 90, in backup_histories_df
    if point.history.dtypes == object:
AttributeError: 'dict' object has no attribute 'dtypes'
```

ChristianTremblay, 2020-03-30:
Hi, sorry for the delay, things are quite weird these days with COVID-19...
Is pandas installed ? If not, I may have forgot to handle the case where pandas is not in the system.

bernhard-steve, 2020-04-01:
Also seeing this error.  No pandas (running in Lite).  Workaround is to manually unregister devices via `bacnet.unregister_device(device)`

pzg250, 2020-04-02:
@ChristianTremblay thanks for your reply. Yes, I haven't install pandas. After install it, the error disappeared. Thanks

pzg250, 2020-04-02:
> Also seeing this error. No pandas (running in Lite). Workaround is to manually unregister devices via `bacnet.unregister_device(device)`

Thank you very much! Got it.

closed 2020-04-02 by pzg250