ChristianTremblay/BAC0 #192: Connecting to Niagara controller and writing/hosting tags

dclark3774, 2020-04-02:
Hello,

I am running:
BAC0 version 20.2.20
bacpypes version 0.18.0
On an ARMv7.

I am new to BACnet, so I apologize if there is a simple fix to the issues I am having. 
I am trying to connect to a Niagara controller through BACnet to read and write variables. I also want to host variables on my device that can be read/written.

I am able to connect to the Niagara controller using the code below:
```
import BAC0
bacnet = BAC0.connect('10.10.10.10',port=55,mask=24)
niagara = BAC0.device('10.10.10.100', 25, bacnet)
```
Once I am connected I can read the variables using:
`niagara.points`
which returns
"[C2C_25/points.realVar : 66.60 noUnits, C2C_25/points.boolVar : True, C2C_25/points.stringVar : test string]"

But if I try to discover the devices on my BACnet network:
`bacnet.discover(global_broadcast=True)`
It returns nothing

At the same time I can see that I am connected to the Niagara using the command:
`bacnet.registered_devices` and it returns `[C2C_25 / Connected]`

Is this the correct behavior or is this an issue?

As for writing to points, I am using the below command:
`niagara['points.boolVar'] = True` 
It returns the abort reason of writeAccessDenied. Even though in Niagara we have the BACnet writable selected for all tags. Is there something else I am missing?
I have also tried this:
niagara['points.realVar'] = 3.3
It returns "ERROR | Value must be numeric"
I see there have been issues like this before and i tried the solution with no luck :(

As for hosting tags on BAC0, I am unsure where to start. We have tried to add objects but have been unsuccessful at getting it to work without errors.

Again, I apologize if any of these questions/issues are simple but I am just now starting to work with BACnet.

Any help with this would be greatly appreciated!

ChristianTremblay, 2020-04-02:
When exposing point in Niagara using BACnet (via local table), if you want to be able to write to those points from something else, you must configure the user BACnet with write permission.
A little tweak, not super well documented.

Regarding tags, not sure I understand... are you talking about haystack tags ? If so, BAC0 do not support tags.
Or are you referring to PLC kinda `tags ` like "BACnet objects" that would be included inside BAC0 ?
The latter is supported and you can find an example of application here : https://pythoninthebuilding.wordpress.com/2020/02/24/a-bacnet-gateway-using-bac0-part-3/

dclark3774, 2020-04-02:
Chris,

Thank you for the quick reply :)

Where do I need to go to configure the permissions of the device on the network?

In regards to the "tags", the device is an open source PLC that we are trying to use as a gateway between a PLC network and a BACnet network. Therefore we want to use the controller to read and write tags in the BACnet network, and have tags on the PLC (as a BACnet device) for the rest of the BACnet system to be able to read and write to.

If this does not make sense let me know I may be able to explain it better.

Thanks!

ChristianTremblay, 2020-04-03:
[image]

[image]

extract from : docBACnet.pdf pages 49-50

ChristianTremblay, 2020-04-03:
Regarding your application, in BACnet terminology we talk of a BACnet Gateway. A gateway translate protocoles to BACnet objects. So tags are referred to BACnet objects.

Have a look to the link I sent. And let me know if this helps in building your PLC to BACnet Gateway.

dclark3774, 2020-04-06:
Chris,

The BAC0 blog post was able to get the information we needed (after some trials). We are now able to work on the implementation. 

Thank you for your help!

closed 2020-04-06 by dclark3774