ChristianTremblay/BAC0 #193: How to define the controller?

jubin13, 2020-04-21:
I use a Wifi Thermostat from contemporary controls(BAST-221C-BW2). I need to write a temperature value using python. I found your code suited to this task using Python. 
>>>BAC0.device(address, device_id, network)
How do I find the correct address, device_id, and network? I keep getting errors.
[image]
Kindly advice.
Thanks.

ChristianTremblay, 2020-04-22:
Start by finding the device on your network using 
```
bacnet.discover()
```

You will get the address and the device id

jubin13, 2020-04-22:
Thank you for your prompt reply. I am able to connect to the BACnet I/P thermostat using YABE and BACnet discovery tools. I used the above-mentioned code to find the device on the network. The result is shown below. 
[image]
It does not detect any network. Is there something wrong with my initial steps?

I also tried the below code as mentioned in your documentation for BAC0, but comes out with an error of unrecognized format.
[image]
[image]
I

ChristianTremblay, 2020-04-22:
Yeah, list of networks must be a list of bacnet network number... when you know them (ex. [2, 300, 1001].

For discover to work, some pieces of the puzzle must be in place.
First, basic discovery of device will use a local broadcast. So you would need to be in the same subnet than your thermostat for the request to work.

So let say your IP is 192.168.0.10 and the thermostat is 192.168.0.14, thing should go smooth.

```
import BAC0
bacnet = BAC0.lite(ip='192.168.0.10/24')
bacnet.discover()
```
You should get something here...
See what I get here : 
[image]

if not, you can try to be more violent and use a global broadcast
```
bacnet.discover(global_broadcast=True)
```
This will target 255.255.255.255 as the destination for the request, instead of 192.168.0.255.
(Here I invite you to use Wireshark, you'll see a lot of interesting things)

Based on the result, you will know what are the addresses and device instanced (device id) of the devices you have on your network.
For example, here I have a thermostat 
('303:8', 72008),

It's an MSTP device, behind a BACnet router, on MSTP network #303, MAC adress 8 and device ID 72008

I would define my thermostat like that
```
my_stat = BAC0.device('303:8', 72008, bacnet)
```

In your case, you will find an IP device. The address will be an IP address.
Let say 192.168.0.14, id 72008
You would define your thermostats like : 
```
your_stat = BAC0.device('192.168.0.14', 72008, bacnet)
```

jubin13, 2020-04-27:
Thanks a lot for the detailed explanation @ChristianTremblay . The setup works now. I am able to connect to my IP BACnet thermostat.

closed 2020-06-16 by ChristianTremblay