ChristianTremblay/BAC0 #194: Creating Virtual Environment

calgaryy, 2020-04-22:
Hi there!

I'm looking to make a virtual device using BAC0 similar to what was started in the conftest.py file. 

```
hvac = BAC0.lite(port=45501)
temp = create_AV(oid=1, name="temp", pv=12.8, pv_writable=True)
temp.units = EngineeringUnits.enumerations["degreesCelsius"]
temp.description = "AVG Temp"
hvac.this_application.add_object(temp)
ip = hvac.localIPAddr.addrTuple[0]
boid = hvac.Boid
bacnet = BAC0.lite()
test_device = BAC0.device("{}:45501".format(ip), boid, bacnet, poll=10)
print(boid)
params = namedtuple(
    "devices",
    ["bacnet", "hvac", "test_device",],
    )
params.bacnet = bacnet
params.hvac = hvac
params.test_device = test_device
params.test_device.disconnect()
params.bacnet.disconnect()
```
This is the code and it leads to the following output:
```
2020-04-22 14:10:18,780 - INFO    | Starting BAC0 version 20.02.20 (Lite)
2020-04-22 14:10:18,780 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2020-04-22 14:10:18,780 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2020-04-22 14:10:18,799 - INFO    | Using ip : 0xc0a80039b1bd
2020-04-22 14:10:18,800 - INFO    | Starting app...
2020-04-22 14:10:18,801 - INFO    | BAC0 started
2020-04-22 14:10:18,801 - INFO    | Registered as Simple BACnet/IP App
2020-04-22 14:10:18,804 - INFO    | Starting BAC0 version 20.02.20 (Lite)
2020-04-22 14:10:18,804 - INFO    | Use BAC0.log_level to adjust verbosity of the app.
2020-04-22 14:10:18,804 - INFO    | Ex. BAC0.log_level('silence') or BAC0.log_level('error')
2020-04-22 14:10:18,823 - INFO    | Using ip : 192.168.0.57
2020-04-22 14:10:18,824 - INFO    | Starting app...
2020-04-22 14:10:18,825 - INFO    | BAC0 started
2020-04-22 14:10:18,825 - INFO    | Registered as Simple BACnet/IP App
2020-04-22 14:10:18,827 - INFO    | Changing device state to DeviceDisconnected'>
2020-04-22 14:10:18,836 - INFO    | Changing device state to RPMDeviceConnected'>
2020-04-22 14:10:18,846 - INFO    | Device 3057153:[BAC0] found... building points list
2020-04-22 14:10:18,860 - INFO    | Ready!
2020-04-22 14:10:18,860 - INFO    | Device defined for normal polling with a delay of 10sec
2020-04-22 14:10:18,862 - INFO    | Polling started, values read every 10 seconds
3057153 (from the print(boid) line)
2020-04-22 14:10:18,862 - INFO    | Wait while stopping polling
2020-04-22 14:10:18,900 - INFO    | Polling stopped
2020-04-22 14:10:19,001 - INFO    | Device saved to BAC0.db
2020-04-22 14:10:19,001 - INFO    | Changing device state to DeviceFromDB'>
2020-04-22 14:10:19,001 - INFO    | Initializing DB
2020-04-22 14:10:19,004 - INFO    | Device restored from db
2020-04-22 14:10:19,004 - INFO    | You can reconnect to network using : "device.connect(network=bacnet)"
```
Is this the expected output? Further, I was wondering, is there a way to get this working with bacserv? The tool available within the toolkit provided by bacnet.org: (https://sourceforge.net/projects/bacnet/files/bacnet-tools/bacnet-tools-0.8.6/)

ChristianTremblay, 2020-04-22:
Seems correct. Not sure I understand what you want to do ? Technically, if you want to use something else to interact with your device, you would only need the first lines

```
hvac = BAC0.lite(port=45501)
temp = create_AV(oid=1, name="temp", pv=12.8, pv_writable=True)
temp.units = EngineeringUnits.enumerations["degreesCelsius"]
temp.description = "AVG Temp"
hvac.this_application.add_object(temp)
```

Let that run and any other BACnet tool will find your device.

calgaryy, 2020-04-22:
the objective is to set up a fake device (in this case a virtual HVAC) and have it poll to a separate server (on either the same or a different subnet). In terms of keeping it running, do you just recommend keeping it in an infinite loop until the program is quit?

ChristianTremblay, 2020-04-23:
Yes. In BACnet, the device needs to be "alive" all the time. This will keep the socket active.
Here you can find one way of achieving this creating a service.
https://pythoninthebuilding.wordpress.com/2020/02/24/a-bacnet-gateway-using-bac0-part-3/