ChristianTremblay/BAC0 #195: debug message

pzg250, 2020-04-24:
Hi,
I used BAC0 in twisted project. I can see a lot of debug messages when starting. 
Even if I set the level to INFO, there are still huge messages.
does someone knows how to mute the BAC0's internal debug messages ?
```
logging.basicConfig(level=logging.INFO,
    format='%(asctime)s - %(filename)s[line:%(lineno)d] - %(levelname)s: %(message)s')

BAC0.log_level('silence')

```

```
2020-04-23 21:56:02,154 - notes.py[line:229] - DEBUG: # Read Multiple
2020-04-23 21:56:02,154 - notes.py[line:239] - DEBUG: ==================================================================================================================
2020-04-23 21:56:02,154 - notes.py[line:228] - DEBUG: ###################################
2020-04-23 21:56:02,154 - notes.py[line:232] - DEBUG: ['192.168.1.173', 'binaryValue', '3000218', 'objectName', 'presentValue', 'inactiveText', 'activeText', 'description', 'binaryValue', '3000219', 'objectName', 'presentValue', 'inactiveText', 'activeText', 'description', 'binaryValue', '3000248', 'objectName', 'presentValue', 'inactiveText', 'activeText', 'description', 'binaryValue', '3000249', 'objectName', 'presentValue', 'inactiveText', 'activeText', 'description', 'binaryValue', '3000267', 'objectName', 'presentValue', 'inactiveText', 'activeText', 'description']
2020-04-23 21:56:02,154 - notes.py[line:237] - DEBUG: ==================================================================================================================
2020-04-23 21:56:02,154 - notes.py[line:238] - DEBUG: 'binaryOutput' : 3000603
2020-04-23 21:56:02,154 - Read.py[line:282] - DEBUG: ------------------------------------------------------------------------------------------------------------------
2020-04-23 21:56:02,154 - Read.py[line:329] - DEBUG: 'description'        None                 '临时占用控制'                       <class 'bacpypes.primitivedata.CharacterString'>
2020-04-23 21:56:02,154 - notes.py[line:236] - DEBUG:
2020-04-23 21:56:02,154 - notes.py[line:230] - DEBUG: ###################################
2020-04-23 21:56:02,154 - Read.py[line:279] - DEBUG: propertyIdentifier   propertyArrayIndex   value                          datatype
2020-04-23 21:56:02,154 - notes.py[line:229] - DEBUG: # Read Multiple
2020-04-23 21:56:02,154 - notes.py[line:233] - DEBUG: ###################################
2020-04-23 21:56:02,154 - notes.py[line:238] - DEBUG: 'analogValue' : 3009023
2020-04-23 21:56:02,154 - notes.py[line:239] - DEBUG: ==================================================================================================================

```

bernhard-steve, 2020-05-06:
This is how I accomplished it:
```
import BAC0
import logging

# completely disable BAC0 logging
logging.getLogger("BAC0_Root.BAC0.scripts.Base.Base").disabled = True
BAC0.log_level("silence")
```

pzg250, 2020-05-08:
> This is how I accomplished it:
> 
> ```
> import BAC0
> import logging
> 
> # completely disable BAC0 logging
> logging.getLogger("BAC0_Root.BAC0.scripts.Base.Base").disabled = True
> BAC0.log_level("silence")
> ```

Thanks very much for your reply. I will try.

closed 2020-05-08 by pzg250

ChristianTremblay, 2020-05-11:
@pzg250 woudl you suggest I add `logging.getLogger("BAC0_Root.BAC0.scripts.Base.Base").disabled = True` to the "silence" attempt ?

pzg250, 2020-05-12:
@ChristianTremblay Yes, that will be wonderful. 
It works for me.