ChristianTremblay/BAC0 #196: Device init fails with device_id = 0

bernhard-steve, 2020-05-06:
I have a BACnet device with device id of 0.  When I try to init the device in BAC0, I get the error
```
BAC0.core.io.IOExceptions.BadDeviceDefinition: Please provide address, device id and network or specify from_backup argument
```
If I update line 189 of `Device.py` to:
```
and self.properties.device_id is not None
```
I am able to init the device.

I want to point out this behavior in case it is unintentional.

Thanks for the great library!

ChristianTremblay, 2020-05-18:
How do you define the device?

`bacnet.device(“2:4”,0,bacnet)`

Should work

(2:4 being changed to the address of your device...)

ChristianTremblay, 2020-05-18:
Ho ! I see....

0 = False....

Oups :-)

Thanks for pointing that !

closed 2020-05-19 by ChristianTremblay