ChristianTremblay/BAC0 #197: Need Help on Device Communication Control and Reinitialize Device

spokale01, 2020-05-11:
Hi,
With the BAC0 and Python, I have been able to test my device multiple features. Thank you for such wonderful module BAC0 !

Further to my testing, I am going to execute the 'Remote Device Management Services' particularly   the 'Device Communication Control Services' (DCC) and 'Reinitialize Device' (RD) services. 

I would like to know more about this, but could get much information in BAC0 ( Or am I missing something in search). My interest would be in understanding how to send these two request using python script ?  Please help !

Regards,
Sandeep

spokale01, 2020-05-11:
Hi, I got the information on 'Reinitialize Device' (RD) service. Implemented in script and tested successfully. But still looking for DCC support. Please help.

ChristianTremblay, 2020-05-11:
I never implemented DCC...but it seem pretty similar to Reinitialize... 
Are you willing to serve as beta tester ?
I just pushed this feature into develop.
You will have to clone the repo and `pip install .` to get develop version on your PC.

```
# example of request
bacnet.dcc(address='address of device', duration=1, state='disable')
# state can be 'enable', 'disable', 'disableInitiation'
# duration in minutes, can be omitted for "undefined"
```

I don't have anything to test this,

spokale01, 2020-05-12:
Hi Chris,

Thank you for your support. Yes sure I will help you in testing this DCC feature..

As I am new on 'GitHub' (and also on Python) hence explaining the each steps here. just to ensure that I have not made any mistake. I am sorry about this..

1. From home page of BAC0 -> Active Branch (develop) ->BAC , By using 'Clone or Download' button, Downloaded the 'BAC0-develop.zip' folder. 
2. Copied that folder (at C:\Users\Sandeep) and then uninstalled the existing BAC0 version 20.2.20
(pip uninstall BAC0). 
3. Installed the new BAC0 version 20.2.21 ( I m using 'Anaconda' distribution package for testing, here I faced challenge in installing new version because how much ever I tried, it was installing the old version only :( )
4. After installing this new version I started getting import error 'refer to documentation or reinstall using $ "pip install BAC0"  No module named 'BAC0.core.functions.DeviceCommunicationControl'

Am I missing something in installing this new BAC0 version. Please help.

ChristianTremblay, 2020-05-12:
My preferred way is : 
```
git clone https://github.com/ChristianTremblay/BAC0.git
git checkout develop
# can validate with git status if everything is ok
# besure to be in BAC0's folder
cd BAC0
pip install .
```

You can also use
```
# inside the folder
python setup.py install 
```

develop version is "20.02.21dev"

spokale01, 2020-05-13:
[BAC0_CMD_Log.txt](https://github.com/ChristianTremblay/BAC0/files/4621587/BAC0_CMD_Log.txt)
Hi,

I tried your way, (clone and then checkout of develop ) Now I have BAC0 version 2.20.12dev on my system. 

Following are the steps I have used to install this new version of BAC0 
 
0. Clear all the previous BAC0 installation. ( pip uninstall BAC0) 
1. Install the BAC0 (pip install BAC0) - In this step, I observed, Always BAC0 V20.2.20 get installed.. 
2. Using CMD went up to new BAC0 folder and then did install BAC0 v20.05.12dev (Python setup.py install ) It gave me an import error. please find the attached CMD log. for more information.
3. After that I read the BAC0 version but it still shows 20.2.20
4. Also I checked at 'C:\Users\pokasa01\Anaconda3\Lib\site-packages\BAC0\core\functions' that DeviceCommunicationControl.py file is not there.  

Can you please help me in installing the new BAC0 or tell where I am making mistake.

ChristianTremblay, 2020-05-13:
I made a typo in a filename... I'll push another version. Sorry for that.

ChristianTremblay, 2020-05-13:
This should work now

spokale01, 2020-05-15:
Yes, I am able to 'import' the BAC0 module without any error. And also able to access the 'DeviceCommunicationCotrol.py' file.

Thank you for the help.

ChristianTremblay, 2020-05-15:
Technically, you should b able to use 

‘’’
bacnet.dcc(options...)
‘’’

spokale01, 2020-05-18:
Hello Christian, Yes I am able to use the 'bacnet.dcc' now. Although I haven't yet completed my testing. so couldnt put update here. Will post my observation on completion of my testing.
Thank you again for your support..

closed 2020-05-23 by ChristianTremblay