ChristianTremblay/BAC0 #198: Unable to test BACnet mstp device using python script.

spokale01, 2020-05-18:
Hello Christian,

I am unable to test the BACnet MSTP device using python scripts.  Actually these scripts are running successfully ( so no doubt on script) for BACnet IP device. But when did some modification ( that too in the address field of 'read' and'write' for MSTP ) it won't work. 

My setup is as below, 

1. Laptop on which running my python script ( using Spyder). It has an IP address set dynamically by MBS router. ( Dynamic IP address is 169.254.0.2).

2. MBS Router ( here router mode is set to B/IP MS/TP so that I can support both layers) And its IP address is 169.254.0.1.

3. MBS router one port used a) for communication with laptop over ethernet. and other b) for communication with IUT over MSTP with 'Network' number set to '2' and 'MAC' address set to '3'. 

4. IUT which has serial address 4, Device Identity object is also 4, Max master set to 5, Baud rate set is 38400. 

5. At the start, I have verified the communication using VTS. And I am able to get the response for 'Who-Is' request. 
Settings for VTS are, (IP) 'Destination' address is '169.254.0.1:47808' - (BVLCI) Original Unicast - (NPCI) DNET(2), DADR (04), Hop Count (255).

6. When I tried to run simple python script, I am getting APDU timeout. I have referred the script 'Read.py'. Please find the code snippet attachment. 
[image]
My script for read operation is,

import BAC0
sysIP = '169.254.0.2/16'
bacnet = BAC0.connect (sysIP)
ReadVal = bacnet.read('2:4 device 4 modelName')

But I get 'APDU Timeout' always..

Then in second attempt, Changed the number '4' in the address field (2:4) to 3 i.e. used MAC address number instead of IUT serial address. But no success.

Then in third attempt, Changed the address field '2:4' to router address '169.254.0.1' then the read request is rejected (with reason 'unidentified service' )

Can you please help me to use BAC0's 'read' and 'write' request for mstp device.

ChristianTremblay, 2020-05-18:
IP range 169.254.x.x is often symptom of bad network configuration. Are you sure your network is et up correctly ?

I would try reconfiguring to have some IP address more “common”...maybe something in the 192.168.x.x range to start with.  

Or step static IP on your devices if no DHCP is available.

spokale01, 2020-05-19:
I changing the IP address to 192.168.x.x but no success. 
Now new setup is as follow,
1. Laptop IP address is 192.168.0.12 (static)
2. MBS Router IP address is 192.168.0.10,
   [ NOTE: After changing the IP address, I am unable to access the MBS web server, Tried different  
     web browser but no success :( ]
   MSTP setting are as below Network Number is 2 and MAC address is also 2
   [image]

3. IUT Serial address 4, Device Identification Number is 3, Baud rate is 38400
[image]

4. Discovered on Yabe
    [image]

5. I have tried all the combinations for 'address field' but could get success. below is the script snippet for your reference.
   [image]

6. But after executing steps in the scripts, I am surprise to see the different 'device' identifier on Wireshark. Why this is so ?
   [image]

  For 'Read' request format should be 

BACnet.read('Nw Number: MAC add', 'Object Name', 'Object Instance', 'Property to be read')
here MAC address in 'Address Filed' is router's MAC address and not the IUT serial address..
Is this a correct understanding.

ChristianTremblay, 2020-05-19:
When using `bacnet.read`you must provide only one big string complying with bacypes format.

So something like : 
```
bacnet.read('2:4 analogInput 1 objectName')
```

https://bac0.readthedocs.io/en/latest/connect.html#read-and-write-using-the-bacnet-instance

spokale01, 2020-05-19:
I think, I have tried that way as well ( also mentioned in excample above ) But no luck. 

When I send that request, I saw on 'Who-Is-Router-To-Network' and 'Network-Number'  services on wireshark log but after that i couldn't see any request-response. 

Thanks again for your timely support.

ChristianTremblay, 2020-05-19:
Regarding point 6. 
Everytime BAC0 starts, it defines itself as a BACnet device and there is a device ID set by default generated if no one is provided. This is why you see all those I-Am.... 

It also means that you script start and stop BAC0... which is not best pratice as the `bacnet` variable, should be alive all the time. Defined once.

I also strongly suggest you play on the REPL to explore your network. Use : 
```
bacnet.discover()
```
and look what happen. You should see the devices and the networks.

Also, previously you were using this : 
```
ReadVal = bacnet.read('2:4 device 4 modelName')
```

but the device ID seems to be 3, so it should be : 
```
ReadVal = bacnet.read('2:4 device 3 modelName')
```

spokale01, 2020-05-20:
I am very sorry for confusion related to device ID , Serial Number etc.
Now I removed the confusing numbers and kept simple straight forward number as mentioned below,

**IP Network number - 1
MBS Router MAC address - 1
MBS Router Network number - 2
Device (IUT) serial address - 3
IUT's Device Identity Instance number is - 4** 

That way i send the request  **bacnet.read('2:3 analogInput 1 presentValue')**
but still get the 'Timeout' error
[image]

but if I send the request using same attributes over the VTS, I am able to get the response 
[image]

On that basis I made the conclusion that my setup is correct.
 
If I don't use the 'bacnet.disconnet' at the end of my script, then i cant run the program second time. it give me an error as metioned below. And for that i have to close the 'Console' and start with new one. that is the only reason i disconnect BAC0
[image]

I didn't get what is mean by REPL, But as per your suggestion, I used the 'bacnet.discover()'.
Couldn't get any device name or network name. see the response below (highlighted in yellow color) "set()" What is that ?
[image]

Regarding the router webpage access, I overlook the settings for 'Webserver access' mode. And it was set to 'https' and hence I couldn't get access using http. finally i made that mode to 'http and https' 
[image]

Thank you so much for the keen attention on my issue and suggestions.

ChristianTremblay, 2020-05-20:
Your subnet mask is 255.255.0.0 in the router. And based on your previous Wireshark captures, your PC use 255.255.255.0.

Set the same subnet on all devices and everything should work.

spokale01, 2020-05-22:
Oops ! I am sorry, I overlooked that important part of an IP setting.
Issue is resolved, am able to run the scripts now. 

Thank you for your through-out support and patience

closed 2020-05-23 by ChristianTremblay