ChristianTremblay/BAC0 #199: BAC0.discover() returns an empty array

Rojj, 2020-05-24:
Just starting to learn BACNET so apologies in advance if this is an obvious issue.

I cannot get the list of devices with BAC0.

Configuration
- BAC0 installed on macos
- Everything is on my local network
- bacnet stack sample server app  on Raspberry Pi 2 [(here)](https://sourceforge.net/projects/bacnet/)
- bacnet stack running on a second raspberry Pi 3

I can see the two devices from a phone app in my network, but when I use BAC0.discover() I get an empty array.

```
import BAC0
print(BAC0.version)
bacnet = BAC0.lite(ip='192.168.1.6/24')
bacnet.discover()
# bacnet.discover(global_broadcast=True) Also tried this

```
I can see the messages in wireshark, but there is no response. 

There is difference though. If I run `whois` from the raspberry pi the wireshark messages are
 (notice the destination address)
```
98	2291.729357	192.168.1.80	255.255.255.255	BACnet-APDU	54	Unconfirmed-REQ who-Is 
99	2291.729720	192.168.1.7	255.255.255.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,2055 
100	2292.139097	192.168.1.3	255.255.255.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,2049 
```

When I run `BAC0.discover()` I get
```
2020-05-24 11:21:36,490 - INFO    | Found those networks : set()
Out[12]: []
```
and the wireshark messages are (see the different destination address)
```
105	3468.377357	192.168.1.6	192.168.1.255	BACnet-NPDU	49	What-Is-Networknumber
106	3470.381466	192.168.1.6	192.168.1.255	BACnet-NPDU	49	Who-Is-Router-To-Network
```

I have a similar issue with `bacpypes`. There I can see the messages in wireshark, but the `WhoIsIam.py` sample app does not return anything. The info messages are also different.

Any idea?

ChristianTremblay, 2020-05-25:
What if you just use 
```
bacnet.whois() # with optional global_broadcast = True option if required
```
Seems `discover` is a little too dependent of "finding a network"... I'll have a look

ChristianTremblay, 2020-05-25:
Thanks for pointing me this. The develop branch have a fix for this, hope it'll work for you.

Rojj, 2020-05-25:
Thanks for looking into this. Some progress.

First I have created a router from [here](url) with the current version (20.02.20). Both `discover` and `whois` return the following

```
an error has occurred: no more packet data
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/bacpypes/core.py", line 165, in run
    fn(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/udp.py", line 301, in _response
    peer.response(pdu)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/udp.py", line 79, in response
    self.director.response(pdu)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/bvllservice.py", line 41, in confirmation
    self.multiplexer.confirmation(self, pdu)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/bvllservice.py", line 179, in confirmation
    self.annexJ.response(PDU(pdu, source=src, destination=dest))
  File "/usr/local/lib/python3.7/site-packages/bacpypes/comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/bvllservice.py", line 312, in confirmation
    self.response(rpdu)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/bvllservice.py", line 402, in confirmation
    self.response(xpdu)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/comm.py", line 320, in response
    self.serverPeer.confirmation(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/netservice.py", line 213, in confirmation
    self.adapterSAP.process_npdu(self, npdu)
  File "/usr/local/lib/python3.7/site-packages/bacpypes/netservice.py", line 587, in process_npdu
    xpdu.decode(_deepcopy(npdu))
  File "/usr/local/lib/python3.7/site-packages/bacpypes/npdu.py", line 449, in decode
    self.rmtnDNET = npdu.get_short()
  File "/usr/local/lib/python3.7/site-packages/bacpypes/comm.py", line 139, in get_short
    return struct.unpack('>H',self.get_data(2))[0]
  File "/usr/local/lib/python3.7/site-packages/bacpypes/comm.py", line 131, in get_data
    raise DecodingError("no more packet data")
bacpypes.errors.DecodingError: no more packet data
2020-05-25 06:50:55,181 - INFO    | 192.168.1.7 router to [2]
2020-05-25 06:50:57,182 - INFO    | Addr : 192.168.1.7
2020-05-25 06:50:57,188 - INFO    | 192.168.1.7 routing table
2020-05-25 06:50:57,189 - INFO    |     1 1 bytearray(b'')
2020-05-25 06:50:57,189 - INFO    |     2 2 bytearray(b'')
2020-05-25 06:50:59,187 - INFO    | Found those networks : {2}
2020-05-25 06:50:59,187 - INFO    | Discovering network 2
Out[6]: []
```

So it can see the router. Strangely the error appeared only after I installed the develop branch in another folder.

When I do discover (or whois) the output from the virtual device is `NPDU: Network Layer Message discarded!` (4 times). Not sure if this helps.

I have then removed the router, installed the development branch in another folder (BAC0.version returns 20..05.24dev) and test it again. `bacnet.discover()` returns

```
In [4]: bacnet.discover(global_broadcast=True)
2020-05-25 07:01:00,291 - INFO    | Found those networks : set()
2020-05-25 07:01:00,292 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
Out[4]: []
```
And `whois` returns
```
In [6]: bacnet.whois(global_broadcast=True)
Out[6]: []
```
The virtual device has the same output `NPDU: Network Layer Message discarded!`.

ChristianTremblay, 2020-05-25:
What is this router ? The URL points to a bad link....
In BACnet terminology, a router will allow BACnet messages to travel between two physical layers. Like BACnet/IP on one side, BACnet/MSTP on the other side. 

If you are using BACnet/IP devices, all on the same subnet, you should not need a router.

The `discover()` had, was that it was missing a condition to just "don't bother about the network number" and just make a whois when no routers where available on the network.

Rojj, 2020-05-25:
Re:router. OK. Got it.

With no router I still have an empty array. The actual output is in the second part of my message.

Just to reiterate. These are virtual devices from [bacnet-stack](https://github.com/stargieg/bacnet-stack) running on two raspberry pi. I can see them from both an iPhone based bacnet explorer and a third linux machine running the same bacnet stack. I can also read and write their inputs and outputs.

To add more details:

- I am running BAC0 on macos (I have also tried on a third Raspberry PI with the same results)
- I am using python3
- I run the test from iPython
- Everything is on local network. I am at home so quite basic

ChristianTremblay, 2020-05-25:
MacOs on the very same subnet than everything else ?
Nothing blocking port UDP 47808 ?
Do you have more than 1 network interface ? You could have to point to the interface needed for the test by specifying it :
```
import BAC0
bacnet = BAC0.lite(ip='192.168.1.2/24')
# given this is the IP of the NIC on your MacOs for example)
```

Can you see something with Wireshark on the MacOs ?

You could also make a simple test with BAC0 on one of the Raspberry Pi. You just open a Python REPL, 

```
import BAC0
bacnet = BAC0.lite()
```
You let that run and you should see a BACnet device on your network. No objects for now but you'll know if communication works.

Rojj, 2020-05-25:
Everything is on the same subnet.

Found the culprit! The interface.

The sample device was on the default interface `eth0`, but the RPi is actually using `wlan0`. If I force it to `wlan0` then it works on both macos and RPi. I can now see the device.
```
In [4]: bacnet.discover()
2020-05-25 14:15:45,878 - INFO    | Found those networks : set()
2020-05-25 14:15:45,880 - INFO    | No BACnet network found, attempting a simple whois using provided device instances limits (0 - 4194303)
Out[4]: [('192.168.1.81', 2049)]

In [5]: bacnet.devices
Out[5]: [('SimpleServer', 'BACnet Stack at SourceForge', '192.168.1.81', 2049)]
```

Just to be sure, I also tested it on `20.02.20` and I can confirm that it returns still an empty array even with the right interface.

It is interesting though how this was not affecting the iPhone app I was using. If it is useful, these are the outputs from wireshark respectively with `eth0` (wrong one) and `wlan0`

```
7	7.314170	192.168.1.81	255.255.255.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,2049 
8	91.516027	192.168.1.81	192.168.1.255	BACnet-APDU	67	Unconfirmed-REQ i-Am device,2049 
```
`192.168.1.81` being the RPi.

Finally, on macos, I had to specify the ip with `BAC0.lite('ip=192.168.1.6/24')` while on the RPi there was no need.

Thanks again. No I can get on with more experiments and learning.

ChristianTremblay, 2020-05-25:
Nice ! Have fun !

closed 2020-05-25 by ChristianTremblay

ChristianTremblay, 2020-05-25:
> Re:router. OK. Got it.
>
> With no router I still have an empty array. The actual output is in the second part of my message.
>
> Just to reiterate. These are virtual devices from [...]

The official repo for BACnet Stack in C is there : https://github.com/bacnet-stack/bacnet-stack