ChristianTremblay/BAC0 #200: long term trending on pi

bbartling, 2020-06-05:
Hi Christian,

I am attempting a long term data storage on the pi again. I updated my pi to Buster and am running the script in Python 3.7. If I am running BAC0 in a script format do I need an infinite loop to keep the script alive?

Any chance I could get a tip for how to save data to Sql lite?

This will throw an error `AttributeError: 'float' object has no attribute 'to_sql'`

If I wanted to use the built in SQL feature of BAC0 do I have to run the full version?

```
import BAC0
import time
import pandas as pd
import sqlite3

print(BAC0.version)
bacnet = BAC0.lite()

conn = sqlite3.connect('TestDB.db')
c = conn.cursor()

bacnet.whois('201:*')
controller = BAC0.device('201:26', 72026, bacnet, history_size=2000)
temp = controller['Room Temperature']

sensor = int(temp.value)

try:
    while True:

        temp.value.to_sql('temp', conn, if_exists='append')
        time.sleep(30)

except KeyboardInterrupt:
    bacnet.disconnect()
```

bbartling, 2020-06-16:
The httpbin is just a site for testing out Json requests... Out of curiosity have you ever ran BAC0 thru a CRON service? Im having some issues trying to get it to work thru CRON but the scrips runs just fine executing the code manually thru the Thonny IDE on the pi

ChristianTremblay, 2020-06-16:
I would probably try `click`
[ref : https://click.palletsprojects.com/en/7.x/]

This way, the code can be reused for anything. If you `pip instal`it (by creating a setup.py file it could also be defined as an executable available in the path... but it's no big deal to call it with python.

`python ben.py --address 303:12 --object_type analogInput --object_instance 1094`

```
#!/usr/bin/env python

import click
import time
import pandas as pd
from datetime import date, datetime
import BAC0, requests, json

@click.command()
@click.option('--address', default="201:2", help='Address of meter.')
@click.option('--object_type', default="analogValue", help='Default : analogValue')
@click.option('--object_instance', default="300", help='Default : 300')
def read_meter(address, object_type, object_instance):
    print(BAC0.version)

    bacnet = BAC0.lite()
    time.sleep(2)

    request = "{} {} {} presentValue".format(address, object_type, object_instance)

    #https://stackoverflow.com/questions/11875770/how-to-overcome-datetime-datetime-not-json-serializable
    def json_serial(obj):
        """JSON serializer for objects not serializable by default json code"""

        if isinstance(obj, (datetime, date)):
            return obj.isoformat()
        raise TypeError ("Type %s not serializable" % type(obj))

    stuff = {}
    stamp = json_serial(datetime.now())
    electricMeterReading = bacnet.read(request)
    print(f'elctricMeterReading {electricMeterReading} kW')
    stuff['Date'] = stamp
    stuff['MeterReading'] = electricMeterReading

    jsonData = json.dumps(stuff)
    response = requests.post('https://httpbin.org/post', json=jsonData)

    print("Status code: ", response.status_code)
    print("Printing Entire Post Request")
    print(response.json())

    bacnet.disconnect()

if __name__ == '__main__':
    read_meter()
```

I have no experience with CRON... but I have been able to create a service that runs on Linux here : https://pythoninthebuilding.wordpress.com/2020/02/24/a-bacnet-gateway-using-bac0-part-3/

Maybe something very simple would work...
```
while True:
    try:
        call function
    except:
        deal with exception
    sleep for some time
```

closed 2020-06-17 by bbartling